MarlinFirmware/Marlin #19144: [FR] The eccentric probe is not recommended for use in delta

frank-pian, 2020-08-25:
### Description

G33 can't to use x/y eccentric probe

I'm a delta printer user. I've always wanted to be able to use an induction probe like core X/Y.
But I found that the eccentric probe did not work properly on the G33. I did some tests, The further off center the probe is, the worse the G33 calibration result will be. A lot of people had the same problem and ended up going back to the center probe.

#12631
#7578
#10129

I thought delta printer extruder would be difficult to keep level in all positions, so it was impossible for the eccentric probe to work accurately. Should this be noted on the documentation?

ellensp, 2020-08-25:
Please attach your configuration files.

frank-pian, 2020-08-25:
> 
> 
> Please attach your configuration files.

[https://drive.google.com/file/d/1A7Ej14nweZp8WSPyOkHNxXq4dLjvvd6-/view?usp=sharing](url)
[https://drive.google.com/file/d/1xqzDN40Ux8DFAi4TsUatNmjkXUiBmzU_/view?usp=sharing](url)

frank-pian, 2020-08-25:
> 
> 
> Please attach your configuration files.

https://drive.google.com/file/d/1xqzDN40Ux8DFAi4TsUatNmjkXUiBmzU_/view?usp=sharing
https://drive.google.com/file/d/1A7Ej14nweZp8WSPyOkHNxXq4dLjvvd6-/view?usp=sharing

sjasonsmith, 2020-09-11:
I reworked the G33 code a while back so that it would not crash into towers when using an off-center probe. As part of this, I changed it to calibrate using the "probe position" rather than the "nozzle position". Was that change correct? I don't really know. My primary concern was preventing the printer from destroying itself.

The reality is that many delta printers have off-center probes, even new from the factory. This is not ideal, but it a scenario that needs to work, even if not as perfect as probing at the nozzle center.

frank-pian, 2020-09-14:
@sjasonsmith 

Thank you for your reply:
Many people cannot tell the difference between auto calibration and auto leveling. They think that if the eccentric probe can work on G29, it can also work well on G33. But the fact is not so. I think this should be clearly stated in the documentation or code comments so that others don't make the same mistakes.
The "probe position" used for automatic calibration is correct. But, G33 algorithm does not support. Is that so?

sjasonsmith, 2020-09-14:
> The "probe position" used for automatic calibration is correct. But, G33 algorithm does not support. Is that so?

It includes the probe offset when probing points on the bed. It will probe a circle of points, with the probe on the circle boundary. Whether that is correct depends on the algorithm used within G33. I do not know the details of the algorithm to know what the best compromise would be.

LVD-AC, 2021-08-28:
@sjasonsmith "I changed it to calibrate using the "probe position" rather than the "nozzle position". Was that change correct?"

The math used assumes the effector positions to be on a circle.

sjasonsmith, 2021-08-28:
> @sjasonsmith "I changed it to calibrate using the "probe position" rather than the "nozzle position". Was that change correct?"
> 
> The math used assumes the effector positions to be on a circle.

My changes were based on avoiding mechanical collisions more than mathematical correctness. Did the underlying math work correctly with an X/Y probe offset prior to my changes?

I’m not trying to prove I’m right here, there is a lot I don’t understand about G33. I was trying to solve one very real problem people were encountering and may have inadvertently worsened another.

LVD-AC, 2021-08-28:
In the calcs the ratio between printing radius and calibration radius is used. Printing radius isn't exact science and depends largely on how fans and stuff are mounted on the effector. So that ratio is not very critical and if it is **"off" by the same amount for all points**, the routine still works, it might just require more iterations to get there, and in the worst case it overshoots the solution and iterates back and forth eternally. Finding the optium ratio so the routine requires the least iterations was a bit trial and error.

When one puts the probe on a circle the calibration radius will vary from point to point. If the routine can deal with such **varying** calibration radius (to big at one side and to small at the opposite side) as well, needs to be tested to make sure. It would iterate slower at one side and faster at the other side and could end up in an endsless loop of overshooting back and forth at the fast side. In that case we need to tweak the scaling parameters taking into account the offset somehow to avoid this.

LVD-AC, 2021-08-29:
The math used to calculate and correct for the kinematic moves to be in a flat and square plane assumes **the effector** being in the centre and on a circle at the same distance off that centre. Only in these conditions the formullas take a rather simplified form that is always the same (some scaling factors aside) to all delta printers (see also #9138).

e.g. When the delta radius is a bit off this results in kinematic moves that are no longer in a plane but are bowl (or dome) shaped: when the effector is in the centre there is a minimum and probed with the effector at same distance R from the centre all the results will be the same. If you offset this by putting the probe instead of the effector in the centre and at a fixed distance you no longer measure the minimum of the bowl but slightly higher, and your probe points on the circle will no longer report all the same distance but will be reported as tilted. In Short a simple bowl shape that only needs a delta_radius correction gets reported as a **tilted** and slighly **shallower** bowl that also requires (false) endstop_correction.

[image]

The same applies for tower_angle_corrections that result in moves according a saddle shape with 2 high points on the tower axis and 2 low points on an axis perpendicular to that. By offsetting the centre this will be reported as a **tilted** saddle shape whos axis are a bit **twisted**. This will induce all kinds of false corrections.

Nevertheless with all those false tilts and twists being reported to the calibration routine it will still find a valid solution IMO since the closer it gets to a flat plane, the smaller these false tilts and twists will get as well, and with a perfect flat and square plane these would have disapeared completely. Obviously with all these false corrections that are made and that have to be removed afterwards this will require more iterations to do so; that will be the trade off by having an excentric probe. It might even make the convergance instable; but that's something we can deal with if and when it gets reported.

sjasonsmith, 2021-08-29:
@LVD-AC would't you have similar problems if you have the effector in the correct place but are measuring Z at an X/Y probe offset? Any error introdued by inaccurate settings would be accentuated (or diminished) by the probe offset.

It seems like in either case you are working with non-ideal measurements. Everybody should just probe at the nozzle then there are no problems! :)

LVD-AC, 2021-08-29:
Not realy (see update picture above) with the effector in the centre (or at any point) and the probing at an X/Y offset this would give the correct result; the bed isn't bowl shaped it is flat ans square, it is the kinematic moves that are bowl/saddle shaped or tilted.

sjasonsmith, 2021-08-29:
@LVD-AC is the effector necessarily parallel to the bed? If yes and the bed is flat, then the offset wouldn't matter. If the calibration errors cause the effector to tilt then it won't work well with an offset probe.

I just want to re-iterate that I am not arguing with you, just trying to understand the problem better. You obviously know more about this than I do!

It could certainly be switched back to putting the nozzle on a circle, it will just need more rework to avoid crashes when using runtime-configured probe offsets.

LVD-AC, 2021-08-29:
@sjasonsmith "is the effector necessarily parallel to the bed?"

The whole purpose of calibration is to make the effector move in a flat plane paralel to the bed; so if the bed is on springs and tilted towards the towers so will the calibration be. Automatic calibration by probing can not resolve bed tilt or tilted towers (all depends of which you consider fixed and which is croocked) nor for effector tilt. These are mechanical problems.

May I suggest that probing effector positions is put back as default, with an user option to select probe positions. The drawback of the first is limited space to move around and to keep the probe on the bed, so a smaller calibration radius and less accurate. The drawback of the second is all those errors introduced by offsetting the kinematic centre. Let the user choose what works best; and that will depend on the users. After all is only requires 1 bool to be changed in the probe routine G33 uses, so that is easyly implemented.

If I recollect it well G33 starts by calculating if all points are reachable by nozzle **and** probe (with the is_reachable function) and if not abandons with a warning the calibration radius is wrong. So one just needs to make sure that this is_reachable function works properly with the latest configured offsets.

sjasonsmith, 2021-08-29:
I'm fine if things are changed further to make it better. It probably won't be by me. My printers have been turned off almost all year and I have only just reappeared. It has been about 9 months since my delta has even been turned on.

LVD-AC, 2021-08-29:
G33 probes effector points as standard and G33 O for probing with the probe X/Y offsets. Makes everybody happy.

sjasonsmith, 2021-08-29:
If the current solution is incorrect than I don't think there is a need for an option.

I knew this was a risk when I made the change. I specifically called out the behavior change as being possibly incorrect. Nobody involved knew any better so it went in.
https://github.com/MarlinFirmware/Marlin/pull/15887

If the default changes back the formula to auto-calculate a probing radius would need to be modified accordingly.

LVD-AC, 2021-08-30:
Or restore the functionallity of a user selectable calibration radius. #17729

LVD-AC, 2021-09-02:
I wrote "In the calcs the ratio between printing radius and calibration radius is used. Printing radius isn't exact science and depends largely on how fans and stuff are mounted on the effector. So that ratio is not very critical..."

It's been a while! The ratio used is between **delta radius** and calibration radius; so it is (or was before #15887) much more "exact science". Included my tech notes and spreadsheet that go into more detail on how the math works.

[auto calibrate.xls](https://github.com/MarlinFirmware/Marlin/files/7098729/auto.calibrate.xls)
[G33 technical notes.pdf](https://github.com/MarlinFirmware/Marlin/files/7098737/G33.technical.notes.pdf)

LVD-AC, 2021-09-02:
Looking at my notes and spreadsheet in detail:

- Tower angle corrections are based on adding tower and opposite tower position; so if this is reported as tilted (see figure above) one point will be X to large but the opposite point will be (roughly - in linear appoxiamtion) -X to low; so this will be canceled out in the sum of both. This applies as well for displacement matrix as for the corresponding scaling factor.

- Delta radius is based on looking at the 3 towers ans opposite pairs at once; so the same conclusions can be drawn.

- End-stop corrections are calculated geometricly. On an infinite plane one can put a circle off points at a random centre, it will always report the same true tilt of that plane. The scaling factor is base on CR, the ratio of delta radius and the radius of the probed points; this ratio does not change as well if probed at nozzle or probe.

- So the only problem remaining is the false tilts that are reported with bowl/dome and saddle shapes. As stated above if tower angle corrections and delta radius are calibrated correctly those false tilts will get smaler and smaler and eventually will disappear, which seems to be the case by the use pairs of opposite points to calcultate those.

@sjasonsmith "I knew this was a risk when I made the change." Errors are made but the concept of adding opposite pairs do cancel those out for angle corrections and delta radius, and for the end-stops corrections, the linear assumption makes the formullas offsetable.

**The math was not designed for it but seems to work as well for offsetted probe points**.

- Using nozzle positions : the probe radius will be limited by the print radius + the probe running of the bed.

- Using probe positions : the probe radius will be limited by the effector crashing into the towers (not necessesary the same as for the print radius since the probe can be mounted on a favorable side where there is more of room then print radius.

IMO the user should be able to choose whatever probe radius is the largest or works best so including an option in G33 seems the way forward.

@sjasonsmith "If the default changes back the formula to auto-calculate a probing radius would need to be modified accordingly." - I leave that to the one who came up with that idea to try to calclulate this from other parameters in Bugfix 2.0.x. But I made all the necessary changes to 1.1.x bugfix (no auto calc of the probe radius) to make offset probing optional in #22698

LVD-AC, 2021-09-02:
I also commented on some minor errors in #15887 that need to be adressed in 2.0.x Bugfix

LVD-AC, 2021-09-18:
Can be closed now (#22707 was merged)

frank-pian, 2021-09-24:
I've been waiting for this problem for a year, and now someone has finally solved it.
But I haven't used the delta printer in a while and someone to test it out.

closed 2021-09-24 by frank-pian