MarlinFirmware/Marlin #19146: [FR] (NOISE_THRESHOLD for SD_DETECT_PIN )

tiwanacote, 2020-08-25:
### Description

I have been fighting during one week trying to solve freezing prints over six identical machines with the following hardware:
SKR MINI V1.1
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER with **48cm cable lenght.**
A4988

It is not the same problem as [#15337](https://github.com/MarlinFirmware/Marlin/issues/15337) or [#18358](https://github.com/MarlinFirmware/Marlin/issues/18358) or [#18117 ](https://github.com/MarlinFirmware/Marlin/issues/18117), because when printer stops ramdomly, I can navegate over screen and stop print and it go home without problems. This issue does not  kicks the processor. It stops the SPI communication.
The problem **was noise over SD_DETECT_PIN**. It was solved with #undef SD_DETECT_PIN into pins_BTT_Skr_MINI_V1_1.h file

### Feature Workflow

1. [First Action]

Add feature like ENDSTOP_NOISE_THRESHOLD for SD_DETECT_PIN read. It could be added in Configuration_adv.h file into #ifdef SDSUPPORT.

stef-ladefense, 2020-09-01:
I had a similar problem, have you tried to solder a 100nF capacitor in parallel on the contact of the filament detector?

tiwanacote, 2020-09-02:
> I had a similar problem, have you tried to solder a 100nF capacitor in parallel on the contact of the filament detector?

Hi, I dont have a problem on filament detector, just only in SD card detector pin... What you are mentioned is a possibility