MarlinFirmware/Marlin #19147: [FR] (Cooling Fan Kickstart)

Penguin-fach, 2020-08-25:
Description

Before I start I should say that at present my printer is an Ender 3 Pro with a BIGTREETECH SKR Mini E3 which is not presently supported by the main release of Marlin so if this feature is implemented it will not directly help me but I have had enough conversations with people over this that I think it might be useful to many.

When using print cooling I will often ramp the fan to a set top speed over several layers, usually no more than five, particularly as an aid to controlling stringing and blobbing with PETG but also for others from time to time. I have observed that the fan will frequently fail to start at lower speeds and if the rate of increase is low enough sometimes it will still fail to start reliably even as high as 50%.

I have tried some tests with a separate blower off the printer and found that it would operate quite consistently down to around 10% but if ramped slowly starting was not absolutely guaranteed below 55%. A brand new blower with well lubricated bearings performed slightly better operating down to 8% but not reliably starting with a slow ramp-up below 40%.

Slicer fan speed calculation takes no account of hardware limitations so it cannot be assumed that the slicer will ensure reliable fan starting and at present the only way to implement this without nurse-maiding the first few layers of a print is to manually edit the Gcode.

This problem is not unique to 3D printing. It is also a common issue with large scale real-world speed controlled ventilation systems. The solution used there is to ensure that every time a ventilation fan is started at or below a given threshold speed (below which reliable starting is not certain), the controller will ramp speed up for a few seconds (sometimes to full speed, sometimes to a lower set-point) and then drop back to the preferred set-point just to make sure that the fan is spinning. Obviously with a big fan this is to stop a stalled fan motor burning out but the outcome is the same.

Feature Workflow

I imagine that this might require five settings or controls.

1. Cooling Fan Kickstart - On/Off  (When OFF, cooling fan control would be operate as at present. ON would enable the Kickstart control).
2. Minimum Fan Speed - Percentage (fan speed requests below this level would default to this value as a minimum. This would be the lowest sustained operating speed that the blower will continue to operate reliably at - 10% would be conservative).
3. Kickstart Threshold - Percentage (Any INCREASING speed request below this level would require a kickstart i.e. if the threshold speed is, say, 50%, the fan is currently at 15% and a new Gcode requests 25% then the fan would get a kickstart but not if it was at 25% and dropped to 15%. Any speed increases above 50% in this case would not get a kickstart).
4. Kickstart Duration - 1/10 Seconds (This is the period that the fan will operate at the higher speed for, before returning to the new set-point. From my experiments 0.5s is usually enough).
4. Kickstart Maximum - Percentage (This is the kickstart speed. In some cases having the fan ramp up to 100% may chill the print so it may well be desirable to be able to choose a lower limit. The firmware would need to trap this value so that it could not be set lower than the Kickstart Threshold percentage as this would not make sense.

Note: If enabled Kickstart would also need to apply to fan speeds set manually by the user from the panel. When disabled the user would be able to use manual controls to characterise their own fan speed response to fine tune Kickstart settings.

Example

A cooling fan is off for the first layer and reaches a maximum speed of 40% at layer 5. However the fan is not starting at all, even at layer 5. The fan is a little old and the bearings are a bit sticky: it struggles to run reliably below 12% speed. The print is sensitive to chilling which reduces interlayer adhesion on this filament so high cooling fan speed is not desirable except when bridging at layer 25.

Kickstart is enabled with a minimum fan speed of 15%, Kickstart Threshold of 50%, Kickstart Duration of 0.5s and Kickstart Maximum of 65%.

Layer 1 - No fan
Layer 2 - Gcode requests 10%. Firmware Kickstart to 65% for 0.5s and then drops to 15%
Layer 3 - Gcode requests 20%. Firmware Kickstart to 65% for 0.5s and then drops to 20%
Layer 4 - Gcode requests 30%. Firmware Kickstart to 65% for 0.5s and then drops to 30%
Layer 5 - Gcode requests 40%. Firmware Kickstart to 65% for 0.5s and then drops to 40%
.
.
Layer 25 (bridging) - Gcode requests 100%. Speed is above Kickstart Threshold so fan ramps to 100%
Layer 25 (normal) - Gcode requests 40%. Speed is dropping so no Kickstart required. Fan drops to 40%

ManuelMcLure, 2020-08-25:
This is already available in current Marlin. See ``FAN_KICKSTART_TIME`` in ``Configuration_adv.h``. Also, the Mini E3 is definitely supported under current Marlin - see https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3 for example configurations using these boards.

thisiskeithb, 2020-08-25:
> I have observed that the fan will frequently fail to start at lower speeds

> I have tried some tests with a separate blower off the printer and found that it would operate quite consistently down to around 10% but if ramped slowly starting was not absolutely guaranteed below [...]

thisiskeithb, 2020-08-25:
> I have observed that the fan will frequently fail to start at lower speeds

Adjust your `FAN_MIN_PWM` setting.

> I have tried some tests with a separate blower off the printer and found that it would operate quite consistently down to around 10% but if ramped slowly starting was not absolutely guaranteed below [...]

This varies from fan to fan. I have several fans that power on at 10%, others at 20%. Adjust your `FAN_*_PWM` settings will be key here.

thisiskeithb, 2020-08-25:
Closing since Fan Kickstart/PWM adjustments are already available and the SKR Mini E3 boards are already supported.

closed 2020-08-25 by thisiskeithb