MarlinFirmware/Marlin #19150: [BUG] Bed Leveling Causing Z Increase when Tool Changes During Print

Protean-Man, 2020-08-25:
**Description:**  Tool Change causes a Z rise that only appears during a print. It does not appear when manually changing tools and checking Z height on bed. I have reported this before but it disappeared until a few days ago. After trying various changes and removing any extra Z rise from firmware. 

I have found that it happens when Bed Leveling is ON - I was using UBL - did not check other types.
When turning Bed Leveling OFF - the issue disappears. 
[Current Bugfix - Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5125879/Current.Bugfix.-.Configs.zip)

**My Configurations:**
Attached in ZIP

**Steps to Reproduce**
1 - Turn on UBL and also tool changing (I use `SWITCHING_TOOLHEAD`)
2 - Print an object that uses two or more tools
3 - Any tool outside of T0 should have a Z rise (I saw around 2mm)

**Expected behavior:** no additional Z rise when printing with T1, T2, T3, etc.

**Actual behavior:** When printing with T1+ there is a Z rise of about 2mm (in my case)

**Additional Info:** related to this which I closed because the issue disappeared. 
https://github.com/MarlinFirmware/Marlin/issues/17737

phongshader, 2020-08-30:
I'm having the same issue
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5146172/configs.zip)

Protean-Man, 2020-10-17:
Still happening. Using current bug-fix.

thinkyhead, 2020-10-17:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Protean-Man, 2020-10-18:
Attached is the log file and my configs.
[repetier.log](https://github.com/MarlinFirmware/Marlin/files/5397726/repetier.log)
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5397739/Configs.zip)

I did some more testing and what is happening is the same as I mentioned in another BUG Report that was closed by inactivity.
Running a tool changer with Switching Tool Setup - the issue is that on a Tool Change there is a Z lift that is not being removed when returning from the tool storage. 
The T1 Offset is only Z-0.2
Strangely on a second subsequent layer using the same active tool the Z lift is removed and it returns to the height that it should be for that next layer. In the log file on these lines you can see Z:0.68 which is an odd number for the Z layer to be at - then right before the layer change it returns to proper height of the first layer. 

**Sum:** Tool Changer - Changing to T1 after printing with T0 - Z lifts by a few layers in height (non layer number of 0.68)- prints the first layer with T1 - returns Z height to the proper height for second layer using the same tool (T1).

> 08:21:12.070 : X:149.69 Y:137.31 **Z:0.68** E:0.00 Count A:22711 B:-310 Z:1465
> 08:21:12.074 : ok
> 08:21:12.074 : echo:N2469 **G1 Z0.3** F7200*37
< 08:21:12.074 : N2473 G1 F3464.339*110
> 08:21:12.111 : ok
> 08:21:12.111 : echo:N2470 M140 S70*99
> 08:21:12.111 : ok
> 08:21:12.111 : echo:N2471 G1 X149.719 Y138.07*39
< 08:21:12.111 : **Printing layer 2 of 25**

**Importantly:** This issue only happens when TOOLCHANGE_NO_RETURN is active. If I turn this function off then the tool change height is exact for every layer.

I hope this helps because I have been posting for almost a year now on this topic. It would be great to have it flushed out!
Thanks for your help.

Protean-Man, 2020-11-13:
Any updates? This is something that is really frustrating and has put a total stop on my multi-tool printers. I just can't figure it out.

Protean-Man, 2020-11-30:
Today's bugfix still having the exact same issue.
After tool change - initial layer with new tool has a Z raise of a few layers (unsure the exact measurement as it is moving/printing), then if the tool prints a second layer without a tool change it lowers the Z to the proper height.

Could this be a slicer problem? I am using Prusa Slicer at the moment. I will try another slicer tomorrow and see what happens.

Today's Configs:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5613570/Configs.zip)

Protean-Man, 2020-12-12:
I made a short video showing this issue in action. It's so small but it makes printing multi-tool prints impossible since the first layer after a tool change is much too high, then the next layer using that same active tool lowers to the proper height. Having the first layer of each tool change being too high is not usable. 

The video link is here. You can see the layer adjustment on the second layer at 00:22. The first layer is much too high, by .4mm I am guessing. 
https://www.youtube.com/watch?v=FWXzhTVXepQ

If anyone could solve this you would make this printer usable again! It has been since the summer with this issue. I have tried a brand new Marlin install and nothing changes.

thinkyhead, 2021-04-17:
Good day! I am taking some time to go through issues in the backlog. If this is still a problem, please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip), adding this line to your configuration:
```cpp
#define DEBUG_TOOL_CHANGE
```
This will provide extra details in the output log about what's going on in the tool-change procedure. Copy the log output into a **.TXT file** and attach it to your next reply, and hopefully that will make clear where the problem lies.

DrumClock, 2021-04-18:
Ahoj, už vím kde je pravděpodobně problém.
Marlin ve verzi ze dne 16-7-2020 nedokončil funkci tool change kdy nevrátil osu Z o nastavenou hodnotu.
Toto při tisku zařídil G-code s výškou vrstvy.

Tato "chyba" však byla pro tisk s více extrudery lepší, protože se hlava vrátila na pozici Z až po přejezdu na
tiskovou pozici.
Tím nedocházelo k "otření" vytékajícího filamentu na jiném místě.

Navrhuji při tisku nevracet osu Z.
Tuto funkci nechat jen když se netiskne .

thinkyhead, 2021-04-18:
@DrumClock — I am confused. Are you asking me to examine all the changes from around July 16, 2020 and something will be very obvious? I mean, maybe it will be obvious…. I don't have a tool-changer running Marlin, so I will need help to confirm any change that I attempt after poking around the changes from around that date.

closed 2021-07-27 by github-actions[bot]