MarlinFirmware/Marlin #19151: [BUG] Sporadic X or Y Layer Shift using Marlin v2.0.5.3 but not v1.1.9

thomaslai, 2020-08-25:
### Bug Description
We are experiencing a sporadic layer shift while printing using Marlin v2.0.5.3 but not with v1.1.9
- We use the exact same hardware and gcode, with the only changes being the firmware
- We use an in-house custom host to run the prints for both versions of firmware
- the layer shift occurs about 1 every 10 prints
- the layer shift occurs at random layers
- layer shift happens for either x or y axis, with roughly equal frequency, but only one axis is shifted at a time.

[image]

### My Configurations
We use a modified version of BCN3D Sigma printer with the plastic extruders replaced with metal paste extruders.
We use a custom in-house host to send gcode to the printer for printing.
We originally modified Marlin to support our functionality, but after stripping everything down to bare Marlin code, we have concluded that the problem lies in Marlin (you can check the configuration files for the changes that remain)

[ConfigurationFiles.zip](https://github.com/MarlinFirmware/Marlin/files/5126758/ConfigurationFiles.zip)
GCode sent on printer host startup:
```
M115 ; Version and features check
M605 S0 ; Dual X Carriage, free mode
M107 ; turn off lamp in case its on
M114 ; report position
M220 S100 ; set feed rate 100
M221 S100 T0 ; set flow rate 100 for T0
M221 S100 T1 ; set flow rate 100 for T1
M503 ; report eeprom setting
M92 X80.00 Y80.00 Z1600.00 E510.90 ; set step per unit
M203 X200.00 Y200.00 Z12.00 E50.00 ; set max feed rate
M201 X1000 Y1000 Z500 E125 ; set max acceleration
M204 P2000.00 R2000.00 T2000.00 ; set starting acceleration
M205 Q20000 S0.00 T0.00 X15.00 Y15.00 Z0.40 E5.00 ; max jerk setting
M206 X0.00 Y0.00 Z0.00 ; home offsets
M420 S0 ; bed leveling state
M42 P11 S255 I ; red LED
M42 P12 S255 I ; green LED
M42 P13 S180 I ; blue LED
M851 X0 Y0 ;
M851 ; report T1 offset
; Nozzle plug timer, firmware will poll endstop every 16 ms
; and send Z_MAX endstop signal when it has counted 62 consecutive plugs
R735 S62 I16
; Printer Status Auto Report,
; S=temperature(s) H=position(ms) P=
M155 S2 H1000 F1
```

### Steps to Reproduce
It would be hard to reproduce without our exact hardware setup, but we are just seeking ideas on how to debug or dig into this. Thanks in advance.

**Expected behavior:**
No layer shift, ever.

**Actual behavior:**
Layer shift some of the time.

#### Additional Information
Things we have tried:
- It's not linear advance
- Its not classic jerk vs junction deviation (both resulted in layer shifts)
- it's not the host (we tried printing with Repetier Host and it also had layer shifts)
- Using a logic analyzer we have found that a weird signal was sent just before the layer shift (y axis)
[image]
[image]
CH0: x steps
CH1: x dir
CH 2: y steps
CH 3: y dir

as you can see, y dir has this weird 36us direction change without any step signals to accompany it. We highly suspect this was the reason for the layer shift, but we do not have enough domain knowledge to make the confirmation.

Any and all help and suggestions would be deeply appreciated. Thanks!

thisiskeithb, 2020-08-26:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

thomaslai, 2020-08-26:
We have cloned bugfix-2.0.x and rebased our bed dimensions ontop of it.
There's a new sanity check that warns against enabling linear advance and s-curve acceleration at the same time, so we disabled s-curve acceleration for now.
We will be continuously printing until we have statistical certainty that the bug is fixed or that the layer shift is reproduced.

By the way, we managed to map the gcodes back to the stepper signals, and we found that the 36us pulse in negative y direction is actually correct, then the printer second-guessed itself and then went in the wrong direction before deciding whoops its actually supposed to be negative. The end result is a layer shift in positive y axis which matches the wrong direction.

[image]

thomaslai, 2020-09-01:
Hi, @thisiskeithb, we have tried both bugfix-2.0.x, as well as 2.0.6.1 after the bugfixes were merged into a release shortly after this bug was posted. We have managed to reproduce the layer shift twice, all at different layers on the same gcode.

Please advise, thanks!

BomberX, 2020-09-12:
@thomaslai 
Hi. I got the same problem with an modified CR-10S (I just added a Titan extruder + BLTouch) with the latest versions of Marlin (2.0.5.3+). I injected an older 2.0.5 version (March 2020) and could not reproduce the problem so far.
Maybe you can try it and let us know.

thomaslai, 2020-09-14:
Ok we will try 2.0.5 and report back. Thanks!

thomaslai, 2020-09-17:
Hi. We have confirmed that 2.0.5 also produces layer shift on two separate printers.

BomberX, 2020-09-17:
Hi. 
Thanks for your feedback. 
Maybe you could try with one or two versions before (like 2.0.4). I'm pretty sure the glitch was introduced pretty recently. If we can narrow when, this would be great and could help the developers.
Thanks !

thomaslai, 2020-09-17:
Thanks, we will. Though trying out each version is a very time-consuming process for us as the prints are long and the shift is rare. If you were to guess, which version do you think is the most likely version to be the culprit? Perhaps we should try 2.0.0 to ascertain that v2 is actually good to begin with.

Roxy-3D, 2020-09-17:
@thomaslai   I have a few quick questions for you:

> as you can see, y dir has this weird 36us direction change without any step signals to accompany it. We highly suspect this was the reason for the layer shift, but we do not have enough domain knowledge to make the confirmation.

How did you trigger the logic analyzer to capture this trace?   How did you know a layer shift had occured?

Also...   Please be aware a number of us have been hunting this bug for a while now.   The original discussion was at: https://github.com/MarlinFirmware/Marlin/issues/12403    I personally believe we are losing stepper pulses in some corner case.    But we have never been able to find the root cause.

InsanityAutomation, 2020-09-17:
#15314 is the PR

https://github.com/MarlinFirmware/Marlin/pull/15314

thomaslai, 2020-09-17:
Hi @Roxy-3D, it's so great to hear that you all will be supporting this debug!

> How did you trigger the logic analyzer to capture this trace? How did you know a layer shift had occured?

There was no triggering mechanism: we simply recorded an entire 2-hour print.
We found a part that layer shifts within the maximum record limit of our logic analyzer (<2 hours), and just kept printing until the layer shift happened. and then we just counted the layers by hand to determine which layer the layer shift happened at. Our gcode has a very distinct behavior at each layer change that makes it obvious from the logic analyzer trace. And then we just worked backwards from the layer change gcode-by-gcode until we found this weird signal.

Obviously doing this by hand is not very scalable, so we are using an emulator to try to look out for this weird signal change.

> We could add debug code that looks for a direction change on the stepper driver chips with no corresponding step pulses before the direction is toggled back to the original direction. And when we see that condition happen, of course we could toggle a GPIO signal to help you generate a logic analyzer trace. But we also could print out (or store) what ever variables we think might be important to get this figured out.

This could be very useful. If you create such a debug build, we will test it for sure.

> Incidentally, a while ago @InsanityAutomation added a small delay when direction changes were signaled to the stepper drivers. This was done because there was a question if switching directions back and forth too quickly might be contributing to this failure mode.

ok we will try out the MINIMUM_STEPPER_POST_DIR_DELAY 20 with our steppers TMC2130 and report back. Which marlin version would you recommend us to test on? Or perhaps we should wait for your debug version?

Roxy-3D, 2020-09-17:
> ok we will try out the MINIMUM_STEPPER_POST_DIR_DELAY 20 with our steppers TMC2130 and report back. Which marlin version would you recommend us to test on? Or perhaps we should wait for your debug version?

Please use the current BugFix version.   That will have the most people using it and is the easiest to get people to help with it.   And, I'm pretty sure this problem has been in the code base for a while so most likely it doesn't matter what version you use.

Mostly, I was trying to describe the type of debug code we could add to help resolve this problem.   But if it actually helps to provide a trigger signal when the Stepper Drivers are told to switch directions and then switch directions again with no intervening step pulses...    We can certainly do that.      Just let us know.    The whole point is...   we can help with the firmware side of the debug.

AnHardt, 2020-09-17:
I would not worry about a changed direction without a step. No steps - no losses. As long the speed is zero you can switch the direction as often and as fast you want.
But the direction change right after that, in this resolution, is locking as if at high pulse-rate - what should not be.

thomaslai, 2020-09-17:
Yea, @AnHardt is right, the most probable cause for the layer shift is that the stepper direction went in the wrong direction (positive y, that matches with the print result) for about 37ms after that empty direction switch

[image]

Our theory is that the initial direction change to negative y was likely the correct one, but something happened to switch it back to positive y... only for it to change its mind again 37ms later.

If you could point us to the part of the codebase where it governs the direction change, we could add debugging code and report back.

Roxy-3D, 2020-09-18:
> If you could point us to the part of the codebase where it governs the direction change, we could add debugging code and report back.

This won't directly answer your question.   But at a higher level the Planner:: code converts the needed movements requested by the GCode into movements along a line segment.   That includes a ramp up in speed.  A constant speed section.  And a declining speed section.    These commands are put into a FIFO queue called block_buffer[].    The Planner:: code will join individual line segments together (and eliminate the ramp up and ramp down in speed) as much as possible.

Elsewhere...   in the back ground, the Stepper Interrupt Service Routine will pull moves out of the list one at a time.   This happens in the stepper.c file in a routine called Stepper::block_phase_isr().  It calls planner.get_current_block() to dequeue the blocks....

AnHardt, 2020-09-18:
Did you measure `Y_DIR_PIN` at pin 17 or 60?

Normally the dir-pins are only set when a new block is started, except for E when LIN_ADVANCE is on - all happens in the stepper interrupt.
But i can find lots of `stepper.set_directions()` in the code when `ENABLED(DUAL_X_CARRIAGE`.
All of them are locking dangerous because they do work on the currently stepped block not on that what is about to be planned. (@Roxy-3D )

AnHardt, 2020-09-18:
@thomaslai 
Can you provide the g-code from the error up to at least BLOCK_BUFFER_SIZE behind that for searching for one of the suspicious g-codes? (any using pause(), G28, M605, dual_x_carriage_unpark(), ...)

InsanityAutomation, 2020-09-18:
Im curious if we disable the condition on Stepper.cpp line 1608

```if (dm != last_direction_bits) {
                last_direction_bits = dm;
                set_directions();
              }```
and let it just always set_directions() that we may catch whatever is corrupting it faster. We could also compare the actual pin state with desired direction instead of the flag last set there.

AnHardt, 2020-09-18:
Code only used by `DIRECT_STEPPING` is not relevant here.

InsanityAutomation, 2020-09-18:
Yeah, 2164 is the corresponding non direct function to what I was thinking. My bad grabbing the wrong one!

thomaslai, 2020-09-18:
@AnHardt here is the gcode. the offending gcode is `G1 X108.120 Y105.170` at line 21075

[NPSTruncated40mmWithLampCode.zip](https://github.com/MarlinFirmware/Marlin/files/5247087/NPSTruncated40mmWithLampCode.zip)

>  But i can find lots of stepper.set_directions() in the code when ENABLED(DUAL_X_CARRIAGE.
Yes we are in fact using DUAL_X_CARRIAGE, but all the prints that layer shifted only ever used one extruder (T0)

> Did you measure Y_DIR_PIN at pin 17 or 60?
I will get back to you on this.

Roxy-3D, 2020-09-18:
@AnHardt  said:

> All of them are locking dangerous because they do work on the currently stepped block, not on that what is about to be planned. (@Roxy-3D ) =8-{

Do you mean "All of them are looking dangerous" or do you literally mean "All of them are locking dangerous" ?

Roxy-3D, 2020-09-18:
> Im curious if we disable the condition on Stepper.cpp line 1608 and let it just always set_directions() that we may catch whatever is corrupting it faster. We could also compare the actual pin state with desired direction instead of the flag last set there.

I would go so far as to say we may want to add an Configuration_adv.h option that either leaves it as is, or does as you say and forces the direction on a per block move.    It maybe that option is just temporary until we figure out this problem.   But it would be good to spend the few extra cycles and force this to happen.

InsanityAutomation, 2020-09-18:
@Roxy-3D I didn't mean it as a permanent change to the main repo but as a test to record on the logic analyzer. Though I do agree that if it proves to resolve the issues at least the temporarily it is a decent enough thing to enable for idex since it's a unique situation.

Roxy-3D, 2020-09-18:
The thing is...   If the direction gets set wrong for one block's move...   for sure you have a layer shift.     Both you and @thomaslai have GCode files that (fairly) reliably create layer shifts.    And I can duplicate the layer shifts you have with your file on my printer.

It may be worth a quick test and remove the optimizations that only set the stepper direction if it has been changed.   And instead always set the stepper direction.     We could find out fairly quickly if that helps.     If it does help, that drastically narrows where we have to hunt and search for this bug.

InsanityAutomation, 2020-09-18:
@AnHardt looking at the direct stepping vs normal logic however, am I correct in interpreting the lack of conditional to always write for the HAS_L64XX condition as a potential bug in direct stepping? Seems if it was important enough to do in the normal case for those it should be done above as well. I have never used that driver type to know.

thomaslai, 2020-09-18:
> Did you measure Y_DIR_PIN at pin 17 or 60?

From what we can tell on the schematic, the Y-DIR pin is 58. This pin is also marked PC5_(A13). Pin 60 is Y-STEP.

AnHardt, 2020-09-22:
From Your Config
```
// M736/M737 alias for M106/M107
#define RAPIDIA_LAMP_ALIAS
```
Is M106/M107 manipulated/altered in your privat code?

`#define BLOCK_BUFFER_SIZE 16`

```gcode
1 X97.848 Y110.065 E3.94052 
G1 E1.94052 F1200.00000 
G92 E0
G1 Z7.25 F4800.000 
G1 X108.120 Y105.170 ;0 !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
G1 Z6.75 ;1
G1 E2.10000 F1200.00000 ;2
G1 F2400.000 ;3
G1 X105.470 Y105.170 E3.24076 ;4
G1 X105.486 Y105.358 E3.32185 ;5
G1 X105.468 Y105.621 E3.43549 ;6
G1 X108.120 Y105.621 E4.57730 ;7
G1 E2.57730 F1200.00000 ;8
G92 E0 ;9
G1 Z7.25 F4800.000 ;10
G1 X96.979 Y110.311 ;11
G1 Z7 ;12
G1 Z12 F8400; layer 27 lamp code start;13
G1 X-43 Y14.19 F5000 ;14
M106 ;15 !!!!
G1 Y76.60 F4092 ;16
G1 Y14.19 F4092 ;17
M107 ;18 !!!!
G0 X96.979 Y110.311 Z7 F5000
G0 F4800; lamp code end
G1 E2.10000 F1200.00000 
```

M106 and M107 have about the right "distance" to `G1 X108.120 Y105.170` to be suspicious.

From `pins_MEGATRONICS_3.h`
```
...
//
// Steppers
//
#define X_STEP_PIN                            58 
#define X_DIR_PIN                             57
#define X_ENABLE_PIN                          59

#define Y_STEP_PIN                             5
#define Y_DIR_PIN                             17
#define Y_ENABLE_PIN                           4
...
#ifndef FAN_PIN
  #define FAN_PIN                              6
#endif
#define FAN1_PIN                               7
...
//
// M3/M4/M5 - Spindle/Laser Control
//
#if DISABLED(REPRAPWORLD_KEYPAD)                  // try to use the keypad connector first
  #define SPINDLE_LASER_PWM_PIN               44  // Hardware PWM
  #define SPINDLE_LASER_ENA_PIN               43  // Pullup!
  #define SPINDLE_DIR_PIN                     42
#elif EXTRUDERS <= 2
  // Hijack the last extruder so that we can get the PWM signal off the Y breakout
  // Move Y to the E2 plug. This makes dual Y steppers harder
  #undef Y_ENABLE_PIN                             //  4
  #undef Y_STEP_PIN                               //  5
  #undef Y_DIR_PIN                                // 17
  #undef E2_ENABLE_PIN                            // 23
  #undef E2_STEP_PIN                              // 22
  #undef E2_DIR_PIN                               // 60
  #define Y_ENABLE_PIN                        23
  #define Y_STEP_PIN                          22
  #define Y_DIR_PIN                           60
  #define SPINDLE_LASER_PWM_PIN                4  // Hardware PWM
  #define SPINDLE_LASER_ENA_PIN               17  // Pullup!
  #define SPINDLE_DIR_PIN                      5
#endif
```
Is your `pins_MEGATRONICS_3.h` altered?

AnHardt, 2020-09-22:
> Do you mean "All of them are looking dangerous" or do you literally mean "All of them are locking dangerous" ?

Should have said "potentiality dangerous". 
Not dangerous when done right - but what is done right in Marlin these days? At least it was worth an investigation.
Is there a `synchronize()` before? Is 'last_direction_bits' altered somewhere? Why not? What do the additional `set_directions()` at all and why? Did something change on the macro level? Is it indeed used as a part of stepper wake up and reinitialize? ...
I did not find anything interesting. Did not dig very deep into the macro-jungle because the "Fan-path" currently looks more promising.

AnHardt, 2020-09-22:
@Roxy-3D 
> And I can duplicate the layer shifts you have with your file on my printer.

Do you have similar config? Please post it.

thomaslai, 2020-09-22:
Hi @AnHardt,

We modified M106 and M107 to turn on a lamp instead of a fan. We have no need of a fan since we don't have hotends. We also did runs without the lamp code and it still layer shifted. We don't think M106/7 have any bearing on layer shift unless somehow the fan logic is tied deeply with stepper directions.

We use a pins_BCN3D.h imported almost directly from BCN3D.

Roxy-3D, 2020-09-23:
@AnHardt   I have a prototype Formbot T-Rex 2+.   I'm not at a machine with the Configuration.h files.   But the example configuration.h files for the Formbot T-Rex 2+ are kept synchronized to my machine.

If you need the actual files I'm using, I can get them for you tomorrow.   Just say you need them!

AnHardt, 2020-09-23:
@Roxy-3D 
Yes - please - Configs and if altered the pin-file.

I'm not in the mood to work with "default" files when the error can be reproduced on files being "almost default". For that i have seen to often that "almost" makes the difference.

InsanityAutomation, 2020-09-23:
@AnHardt she's running the same machine series I am, Trex2+ setup from here where I have the Trex3 toggle on.

https://github.com/InsanityAutomation/Marlin/tree/TM_Trex2+_2.0.x

AnHardt, 2020-09-24:
I can only see the coincidence of the fan commands but can't find the correlation.
The synchronizing with the moves still seems to be ok. Can't find hints for an error in the pin-files.
Checked again the usage and consequences of `stepper.set_directions()`, but can't find a problem.
No ideas left.
Giving up. :-(  -  For now ;-)

Anden78, 2020-10-03:
Hi! I am absolute not a pro at this, but I have issues with Marlin 2.0.x anyway ;)

I have a i3 with MKS gen L with TMC2100 and it prints reasonable ok with 1.1.9. I upgraded to 2.0.6.1 and I got problems with layer shift in Z-dir ( both + and - directions depending on printfile )
At the same time I started to use OctoPrint so I that it was that, but its not. So last night I downgraded to 1.1.9 again and it pints ok again. I compared configfiles .h and .adv in Notpad++ so my settings are the same in both versions. Tried with 4898 and TMC2100 and S-curve ON or OFF, but prints get layershifted after a few layer everytime even with 2.0.7. So for now I will stick with 1.1.9

Is this interresting Input?
The pictures showing the same printfile, but the closest got aborted after it was obvious it would fail again

[image]
[image]
[image]

Anden78, 2020-10-12:
So, I found what my problem was. A classic Y-bed bearing that was worn out. But one question remains, why did the fault with severe layershifting accour only when I uploaded Marlin 2.0.7. And when I shifted back to 1.9.1.1 it worked fine. And back to 2.0.7 catastrofy again? Can something deeper in the software than config.h and adv_config.h? I have compared all files between both versions and they have the exact same settings. 

But now it works great again 👍

[image]

Roxy-3D, 2020-10-15:
It could be coincidence.   But the internal timing of Marlin does change as different code is added and altered.   If you have a bearing that is sticky, it isn't hard to imagine how certain acceleration ramps could be more prone to catching and sticking than other patterns.    

Many things have shifted.   For example, just the way CPU resources are allocated to the LCD panel.   That is going to affect how the steppers get told to move.

Anden78, 2020-10-15:
Ok maybe that explains the wierd behavior when shifting back and forth between Marlin versions.

Maybe another topic, but when I print thrue Octoprint with a Rpi3b+ the printer stops in the middle of the print, with 2.0.7? 

I had to set baudrate to 115200 since I had trouble with communicate with the board. With 1.1.9 I had 250000 and worked fine

Any idea?

nb-rapidia, 2020-10-15:
#19756 may resolve the problem. It's not entirely clear exactly why this resolves the problem, so it'd be greatly appreciated if somebody with more expertise could take a look and make a guess.

closed 2020-11-22 by github-actions[bot]