MarlinFirmware/Marlin #19154: [BUG] Neopixel not turning all LEDs off

soemarko, 2020-08-26:
### Bug Description

I have 24 LEDs that I can turn on just fine, some weird rainbow coloring happened when changing colors, but I don't really mind that. But when I turn the LED off, like about 9 of them towards the end of the strip still lit. However, if I use the menu slowly turn the brightness 0, it works. All LEDs slowly dim then off when the number reach 0.

### My Configurations

- SKR v1.3, latest stable Marlin v2.0.6
- LEDs are plugged into the servo port
- LEDs have 1000uF 6.3V cap across the power and 500ohm resistor on the data pin.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5127947/config.zip) attached, but I believe the significant part is the following

~~~
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     P2_00       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 24       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 255  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif
~~~

and 

~~~
  #define LED_CONTROL_MENU
  #if ENABLED(LED_CONTROL_MENU)
    #define LED_COLOR_PRESETS                 // Enable the Preset Color menu option
    #if ENABLED(LED_COLOR_PRESETS)
      #define LED_USER_PRESET_RED        255  // User defined RED value
      #define LED_USER_PRESET_GREEN      255  // User defined GREEN value
      #define LED_USER_PRESET_BLUE       255  // User defined BLUE value
      #define LED_USER_PRESET_WHITE      255  // User defined WHITE value
      #define LED_USER_PRESET_BRIGHTNESS 255  // User defined intensity
      //#define LED_USER_PRESET_STARTUP       // Have the printer display the user preset color on startup
    #endif
  #endif
~~~

### Steps to Reproduce

1. Using the screen, Click, scroll to Light, then ON
2. Click again to turn off

**Expected behavior:** all LEDs off

**Actual behavior:** about 9 LEDs remain on

#### Additional Information

When turning it on.

[image]

When turning it off.

[image]

Again, if I use the menu slowly turn the brightness 0, it works. All LEDs slowly dim then off when the number reach 0.

ellensp, 2020-08-26:
neopixels are meant to be driven with 5v logic. The SKR 2.0 pin is 3.3v

DJDOUBLEG, 2020-08-26:
> neopixels are meant to be driven with 5v logic. The SKR 2.0 pin is 3.3v

There was no such problem with the previous version. (2.0.5.3)
My opinion is not HW realted issue.

ellensp, 2020-08-26:
I just tested with 18 neopixels on skr 1.3 pin 2.0   (using your configs, so thinks it has 25 neopixels)
Cant reproduce. all neopixels turn on and off as expected with the LCD

soemarko, 2020-08-26:
> I just tested with 18 neopixels on skr 1.3 pin 2.0 (using your configs, so thinks it has 24 neopixels)

Huh, I will just test it by setting more LEDs. I'll set it to 30 or something. And see if it works.

ellensp, 2020-08-26:
tried both current bugfix and 2.0.6, same results...  cant reproduce. 
Though current bugfix seems to have lost the Blue power on test

soemarko, 2020-08-26:
Is your LEDs also have 1000uF cap and 500 ohm resistor?

ellensp, 2020-08-26:
My LEDs have a cap and resistor each led segment. cap value unknown, resistor marked 151 and measures 150 ohms as expected.

soemarko, 2020-08-26:
> Huh, I will just test it by setting more LEDs. I'll set it to 30 or something. And see if it works.

This "fix" it. Well 30 just shifted the problem further back, 5 still left on. I don't feel like keep on flashing firmware to find the magic number so I just set it to 48 (2x my LEDs), problem solved.

boelle, 2020-08-26:
> ), problem solved.

then click the green close below

closed 2020-08-26 by soemarko