MarlinFirmware/Marlin #19155: [FR] (define DHT11 as bad temperature sensor)

aliys97, 2020-08-26:
i want define DHT11 as bad temperature sensor. would you please help me?
and also show data in repitier software.

ellensp, 2020-08-26:
not suitable for this application Temperature Range: 0°C to 50°C

closed 2020-08-26 by ellensp

aliys97, 2020-08-26:
My working temp is -10 upto 20. So it is suitable.

ellensp, 2020-08-26:
it can't do -10

aliys97, 2020-08-26:
Anyway is it possible or not?!!? 0 to 50 is fine.

ellensp, 2020-08-26:
The timing requirements for the sensor are not compatible with a 3d printer. 3d printer requires precise timing or you get blobs.
This sensor requires precise timing to read it...  one of these has to give, you can't have both.

AnHardt, 2020-08-26:
By writing "bad" instead of "bed" you hit the nail on the head.
On top of the already said the DHTs have a insufficient datarate (0.5 to 1 reading/second depending on the model). Marlin heaters need at least 4 to 5 readings/second.