MarlinFirmware/Marlin #19159: [BUG] MKS GEN L V1.0 + MKS LCS 12864A no display

Frytkownica, 2020-08-26:
### Bug Description
I tried several suggestions but still - LCD only lits up, without any display.

### My Configurations
MKS Gen L  + MKS LCD12864A V1.0
compiled in Visual Studio Code

[Settings.zip](https://github.com/MarlinFirmware/Marlin/files/5130131/Settings.zip)

### Steps to Reproduce

1. Change settings to have 
#define MOTHERBOARD BOARD_MKS_GEN_L  
#define MKS_LCD12864
2. Complie
3. Have small blue light lamp instead of printers display
[image]

**Expected behavior:** 
Graphical status display

**Actual behavior:**
Modern art installation

#### Additional Information

I tried delays changing, brightness change, different board (SKR 1.3) all with no results. Is it possible that pins are not correctly defined for this one or any other known issue?

ellensp, 2020-08-26:
comment out you shouldn't need huge delays like this on a 8 bit board. 
#define ST7920_DELAY_1 DELAY_NS(500)
#define ST7920_DELAY_2 DELAY_NS(500)
#define ST7920_DELAY_3 DELAY_NS(500)

sjasonsmith, 2020-08-26:
Try re-initializing your EEPROM either from the LCD or using M502 then M500. This screen might use software adjusted contrast, and could be using a bad default stored to EEPROM prior to defining the display.

ellensp, 2020-08-26:
Or play with setting contrast eg M250 C200

Frytkownica, 2020-08-26:
Hi guys. Thanks for feply.
i was using this board for REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, and delays were 250, 250, 0 to rok okay. I thought maybe higer ones are better. I tried with those values commented - no difference since i tried just chaging 2 values like in reproduction field, on both 2.0.x and 1.9.x marlins

i also tried M250. This is not it. Display does not change when tried 120, 200 and 255 values.

thisiskeithb, 2020-08-26:
> Is it possible that pins are not correctly defined for this one or any other known issue?

Pinout is correct since `MKS_LCD12864` is simply an alias for `MKS_MINI_12864` (which works on a MKS Gen-L), but it uses different contrast settings.

Having said that, I've run into some issues getting the B variant (for 32-bit) working on some 32-bit boards where swapping in a real MKS Mini 12864 controller would display/work fine.

I've asked Makerbase/MKS to look into the issue, but maybe they can chime in here. @makerbase-mks: Any troubleshooting tips?

Frytkownica, 2020-08-26:
I orginally tried to have it running on SKR board, but found out that A ones should be ran on 8 bit boards. This is why i try to make it run on MKS Gen L :)

borland1, 2020-08-26:
I saw a seller's listing that recommends these settings/changes for MKS LCD 12864A...

_Configuration.h:_
`#define MKS_MINI_12864`

_Marlin/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp, at line 100_
`UC1701_CONTRAST(0x32),      // set contrast from default (0x27) to 0x32`

thisiskeithb, 2020-08-26:
> I saw a seller's listing that recommends these settings/changes for MKS LCD 12864A...

That’s for Marlin `1.1.x`.

borland1, 2020-08-26:
Marlin 2.0.x still uses the outdated U8glib graphics library.  Why would there be any difference?

thisiskeithb, 2020-08-27:
> Marlin 2.0.x still uses the outdated U8glib graphics library. Why would there be any difference?

Instructions are different for Marlin `2.0.x` and those changes have already been merged. You simply have to enable `MKS_LCD12864`.

NeoMod, 2020-08-27:
If it is worth a note, the same thing happens on an SGEN L with LCD12864 B.

makerbase-mks, 2020-08-28:
@thisiskeithb @Frytkownica 
I has test it of MKS LCD12864A on 2560 motherboard and i find it's can be use; 
but i test it of MKS LCD12864B on SGEN_L or Robin E3 is cann't be use

NeoMod, 2020-08-28:
@makerbase-mks thanks for the feedback! So it seems there is some kind of problem in the latest release.

Frytkownica, 2020-08-28:
> @thisiskeithb @Frytkownica
> Marlin 2.0.6
> I has test it of MKS LCD12864A on 2560 motherboard and i find it's can be use;
> but i test it of MKS LCD12864B on SGEN_L or Robin E3 is cann't be use
> Marlin2.0.1 all can be use

But based on the boards names I'm using A version nad Gen L is 8 bit.
If LCD was incorrectly named, it should have worked on 32bit SKR board, but it also didn't. Also no way GEN L is faked because It wouldn't compile.

Please provide a way to troubleshoot the display, because it seems all is okays software wise in my opinion.

thisiskeithb, 2020-08-28:
> Marlin2.0.1 all can be use

@makerbase-mks Can you look into why it doesn't work with a current version of Marlin?

borland1, 2020-08-28:
> @thisiskeithb @Frytkownica
> Marlin 2.0.6
> I has test it of MKS LCD12864A on 2560 motherboard and i find it's can be use;
> but i test it of MKS LCD12864B on SGEN_L or Robin E3 is cann't be use
> Marlin2.0.1 all can be use

I am confused by earlier discussion of display driver ST7920, because I was assuming the correct display driver is UC1701.  Is there a Wiki for the MKS LCD12864x display that provides the specification for the correct driver?

makerbase-mks, 2020-08-29:
@thisiskeithb I doing testing it.

makerbase-mks, 2020-08-29:
MKS MINI12864 ： The driver is ST7565R
MKS LCD12864A/B ： The driver is ST7567
I just guess that it may be caused by initialization, but so far, no real reason has been found, and testing is needed.

Frytkownica, 2020-08-31:
Is there any way i could help and speed this up? : )

makerbase-mks, 2020-09-05:
@thisiskeithb @Frytkownica 
MKS has fix it: https://github.com/MarlinFirmware/Marlin/pull/19269

thisiskeithb, 2020-09-07:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

MKS' test shows that the MKS LCD12864A should work with both the Gen-L V1 and V2:
[image]

makerbase-mks, 2020-09-07:
MKS Gen-l V1 and MKS Gen-l V2.x comparison, MKS Gen-L V2.x only add TMC UART, SPI interface. The last version marlin bugfix on the MKS Gen_L V2.1 motherboard. It can be used normally all the time
[image]

Frytkownica, 2020-09-08:
<3 much love
Works like a charm now :) thanku you

PR #19271 (2020-09-06): Fix STM32F1 SPI device init, MKS_LCD12864
### Description

Some PINs file have `ENABLE_SPI2`, to select the SPI 2 as the main device. But the current code start the SPI object with the SPI 1.
If any code try to use SPI before calling `spiInit` function (that set it according to board pins), the SPI will not work.

And this is the problem with #19269 . The LCD code in u8glib don't call `spiInit`, it rely on the current SPI selected from the global SPI object.

The MKS fix call SD card mount in the init of marlin code. The SD mount code call `spiInit`, putting the SPI to the right device, so its [... cut]
closed 2020-09-08 by Frytkownica