MarlinFirmware/Marlin #19162: [FR] Adding SD firmware update support for SAM3X8E (DUE)

futuretech208, 2020-08-26:
### Description
Is it possible to have SD firmware option for SAM3X8E (DUE) based boards?
I could see from configuration_adv.h file #define SD_FIRMWARE_UPDATE command exists, but it is only for Atmega2560 based boards.
I have developed my own pinouts for DUE based board and been testing Marlin 2.0.6 on that.
If this feature is added then it would be great.
I searched a lot on the internet including Atmel's documentation on how the bootloader on SAM chip operates. It is not conventional. It does not have an Arduino bootloader basically. All chips are Factory burn bootloader. 
I came across few Github repos where it was shown if we update the bootloader with CAN Bus, then it is possible to load .bin file from SD card.
I tried uploading the bootloader, it failed.
I am ready to contribute for testing on my board if anyone developes this feature.
### Feature Workflow

1. Should be able to upload new firmware from SD card.

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-08-27:
This would require a custom bootloader. This in my view is outside the scope of Marlin.

futuretech208, 2020-08-27:
I checked today again for many files on the Microchip site. They have said that it is possible with some Dataflash memory(SPI Memory). And they have given some sample codes too. Is it possible to incorporate into Marlin? 
One thing to change while doing so is the GPNVM bit for writing firmware.bin file. 
I am just curious to know if there is any possibility to make it.

Moyster, 2020-08-31:
Well, you have part of the solution already, what you found on github (I guess a custom bootloader ?) is what you need.  

The marlin-side of this feature (the SD_FW_UPDATE option) does this :  
>  When using a bootloader that supports SD-Firmware-Flashing,
>  add a menu item to activate SD-FW-Update on the next reboot.

Now your problem is more "how do I get the bootloader to upload? Does it need fixing/edits? Which offsets to flash to (tho this might already be documented by the custom bootloader you found)"  
Depends on how it works (the custom bootloader) you might reuse and enhance the atmega only "SD_FW_UPDATE" feature to set the appropriate flag to start a fw flash on reboot. Hf ;)