MarlinFirmware/Marlin #19164: [FR] Z stepper alignment based on motor position for 2 motors

logiclrd, 2020-08-27:
### Description

My printer is a CoreXY design with two Z steppers, one on each side of the bed. The bed is 400x400mm, and I measure the stepper positions, relative to the bed origin, to be (-90, 200) and (490, 200) -- about 9 cm past the edge of the bed on each side. Obviously, I cannot probe the bed height where the motors are. I've conservatively configured the probing to go 45mm in from the bed edge, giving points (45, 200) and (355, 200).

At this point, then, it is clear that I need to supply some meaningful value for `Z_STEPPER_ALIGN_AMP`, but what isn't clear is how to compute that value. The exact calculation that `G34` is doing doesn't seem to be clearly documented anywhere.

The Configuration.h template does list an option `Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS`, which seems to allow you to simply tell Marlin where the motors are and it'll figure out how to adjust them based on the probe results. But, this is documented as requiring 3 Z motors to function. There is also a G code command `M422` that lets you tell Marlin where the motors are, but it isn't clear how that's used or if it can have any effect if you don't have 3 motors.

One way to resolve the issue would be with a clear guide for calculating the correct `Z_STEPPER_ALIGN_AMP` value, but what I would like to see is the ability to use `Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS` with only two motors.

### Feature Workflow

1. Set `NUM_Z_STEPPER_DRIVERS` to 2.
2. Enable `Z_STEPPER_AUTO_ALIGN`.
3. Set `Z_STEPPER_ALIGN_XY` to points that can actually be probed, which are different from where the actual Z axis screws are located.
4. Completely ignore `Z_STEPPER_ALIGN_AMP`.
5. Enable `Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS`.
6. Set `Z_STEPPER_ALIGN_STEPPER_XY` to the _two_ locations of the Z screws, relative to the bed origin.

Expected outcome: `G34` figures out the correct adjustment to make factoring in the probe locations vs. the Z motor locations.

#### Additional Information

[image]

sjasonsmith, 2020-09-11:
I didn't try to implement this when I added it for 3 motors, because I don't have a dual-Z printer. The mechanism used for three motors won't work at all for two motors.

If somebody wants to implement this, I think it could be done as a minor variation of the current 2-motor code. It would just need to use the motor positions to select ideal probe points, then calculate the proper value for `Z_STEPPER_ALIGN_AMP` based on those positions. That is a pretty simple math problem to solve.

logiclrd, 2020-09-11:
I was imagining manually-configured probe points, but if it could just magically pick the points I feel like that would be even better :-)

FanDjango, 2020-12-02:
I agree.

The amplification and the way it "adjusts" the correction move that is derived from the Z1 and Z2 position difference is entirely primitive and in no way geometrically correct. As a matter of fact, IIRC, the second corrective move actually is given a possibly much higher amplification than the one defined (capped at 2.0). The first one and later ones use the amplification requested by the user.

As @sjasonsmith noted, the place where this is done is a single line that applies the factor to the abs(Z1-Z2) to derive the magnitude of the corrective move.

This is the line in question:
```
            // Optimize one iteration's correction based on the first measurements
            if (z_align_abs) amplification = (iteration == 1) ? _MIN(last_z_align_move[zstepper] / z_align_abs, 2.0f) : z_auto_align_amplification;
```

This was a first step primitive try to derive the needed amp from the results of the first iterations arbitrarily derived correction and its results seen in the second iterations measurements, assuming the measurements are trustworthy.

More needed?

Maybe just forget it and use the default.

I find that, depending on the magnitude of Z1 to Z2 difference, the convergence results in an alignement (on my machine) between 1 and 2-3 iterations. I have seen 10 iterations or more when the bed is so skewed (when testing) that a beer bottle would fall over.