MarlinFirmware/Marlin #19165: Print bypasses mesh and prints above the table.

JeanCarloEM, 2020-08-27:
Hello everyone!

I'm using SKR 1.4 Turbo with BLTouch, both with Marlin 2.0.6 and Marlin Bugfix.

I made the configuration, including the offset and apparently:
1. Homing works;
2. Self-leveling also works;

After homing and leveling it is possible, in any XY position, to bring the nozzle to Z = 0, almost touching the table, leaving the perfect space for a sheet of paper. Therefore, everything indicates that homing and automatic leveling are working and the mesh is correct.

When homing is finished, this is the output:
``
X: 204.00 Y: 171.00 Z: 30.00 E: 0.00 Count X: 16320 Y: 13680 Z: 12000
``

When auto leveling is finished, the output is:
```
Bilinear Leveling Grid:
      0      1      2      3
 0 -0.075 +0.000 +0.050 +0.663
 1 -0.046 +0.016 +0.332 +0.631
 2 +0.070 +0.070 +0.089 +0.505
 3 +0.144 +0.029 +0.098 +0.467
echo:busy: processing
X:-26.00 Y:-6.00 Z:40.00 E:0.00 Count X:-2080 Y:-480 Z:15970
```
I must admit that I find it curious that, at the end of the auto leveling, there is an exit indicating `Z:15970`, however, as I said, using the command G1 Z0 in any XY, the beak almost touches the table, with room for only one leaf. `Z: 40.00` refers to _NOZZLE_PARK_POINT_.

THE PROBLEM is that when printing, apparently, the Marlin ignores the mesh, printing in the air, about **20mm** above the table.

I think there are two possible problems:
1. Something is going unnoticed in the marlin configuration, OR
2. There is a bug in the marlin.

I ask your help to be able to see the problem in the OU configuration, a solution, if it really is a bug.

Configuração:
[MarlinConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/5134205/MarlinConfigs.zip)

_PS: Sorry for the bad English._

Thanks for any help.

ellensp, 2020-08-27:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

timmeh87, 2020-08-27:
Hi I had a similar problem a long time ago and the problem was that I was not calling the "enable mesh bed levelling" in my printer init code, which I set up in cura. Without turning it on, it just simply is not applied to the print.

Also from the marlin docu:

`G28 disables bed leveling. Follow with M420 S to turn leveling on, or use RESTORE_LEVELING_AFTER_G28 to automatically keep leveling on after G28.`

JeanCarloEM, 2020-08-27:
> Olá. Eu tive um problema semelhante há muito tempo atrás e o problema era que eu não estava chamando o "habilitar nivelamento da cama de malha" no código init da minha impressora, que configurei no cura. Sem ativá-lo, ele simplesmente não é aplicado à impressão.
> 
> Também do documento marlin:
> 
> `G28 disables bed leveling. Follow with M420 S to turn leveling on, or use RESTORE_LEVELING_AFTER_G28 to automatically keep leveling on after G28.`

Sorry for the ignorance, however, LOG I do it through ESP3D and, it is not as detailed as in Printrum. However, in printum there is a limit on the number of lines on the console, which means that I am not able to copy the entire contents of a command such as G29 or M24.

Again, sorry, but how do I log everything?

JeanCarloEM, 2020-08-27:
> RESTORE_LEVELING_AFTER_G28

Thanks @ timmeh87, but the `RESTORE_LEVELING_AFTER_G28` option is enabled in the configuration.

JeanCarloEM, 2020-08-27:
Apparently I discovered the problem, by accident. The travel speed of the Z axis is too high, causing the car to spin. This makes the printer print on the air, I changed the configuration and, from there, I did a single test that worked.

I will print more and, if the error comes back I will comment again.

timmeh87, 2020-08-27:
I keep my Z axis maximum speed at about 5mm/s if that helps, it is not a
fast axis, even with dual Z steppers. I would recommend figuring out the
limits of your printer and then keeping the max speed and accel settings
programmed in the firmware so that gcode cannot command the printer to go
outside its performance envelope

JeanCarloEM, 2020-08-27:
timmeh87, the fact is that my printer is ** Anet A8 Plus and **.

It worked perfectly with Anet Board 1.7, exactly with the same configurations, except the advanced ones, like SDSUPPORT, POWER LOSS, among others (because it doesn't fit in memory). The BL touch worked and the prints came out correctly.

I haven't changed any speed, acceleration or steps from what's in Marlin's official repository for Anet A8 Plus. However, when changing the plate, for SKR 1.4, it seems that the speed / step / acceleration settings CANNOT be the same, or at least, it is something specific to my printer batch - since it seems that no one else complained the wrong settings.

Another wrong configuration (at least, for my case), is the minimum X axis. In the official configuration is `#define X_MIN_POS -26`, but this makes the nozzle car hit hard on the right aluminum, when it reaches X = 300. Therefore, `X_MIN_POS` cannot be -26, or` X_MAX_POS` cannot be 300.

I still don't have enough knowledge to change the configuration of the axis steps. But I will do research on that.

Thanks again to everyone and, as I said, I printed just one thing and it worked, I will do at least 3 more prints by the end of the week and, if the error comes back, it means that the problem is different. Comment again.

Thanks!

JeanCarloEM, 2020-08-28:
Guys, I found the problem.

In the .gco file there is a line (48), where the problem occurs:
``
G0 F6000 X25.288 Y38.085 Z0.3
``

I did the test manually. After homing, when giving this command, the printer makes a louder noise, skating and losing the position of the Z axis.

What's the problem?
See, the code says go with an acceleration of 6000mm^2. As for that, there is no problem, but look at my configuration.
The marlin should limit the speed to a maximum of [100mm^2](https://github.com/MarlinFirmware/Marlin/issues/3961) for axis Z:
``
#define DEFAULT_MAX_ACCELERATION {200, 2000, 100, 200}
``

I actually reduced it to `` #define DEFAULT_MAX_ACCELERATION {500, 500, 50, 500} '', but it didn't work.

I believe it really is a **BUG**, ​​Marlin DOES NOT make the filter limiting the maximum speed according to `` DEFAULT_MAX_ACCELERATION``. I came to that conclusion after reading the [problem here] (https://github.com/MarlinFirmware/Marlin/issues/3961) and doing several tests.

Additionally, I also tried to decrease the settings below, but without a solution:
```
#define DEFAULT_ACCELERATION 200 // X, Y, Z and E acceleration for print movements
#define DEFAULT_RETRACT_ACCELERATION 400 // E acceleration for retraction
#define DEFAULT_TRAVEL_ACCELERATION 300 // X, Y, Z acceleration for travel movements (not printable)
```

**No matter what G-CODE tells you to do, Marlin should filter the command and limit it as stated in the settings.**

I request the correction of the **BUG**. 

This is the gcode generated by the cure application (renamed to .txt for upload):
[CASE_SKR.txt](https://github.com/MarlinFirmware/Marlin/files/5139317/CASE_SKR.txt)

ManuelMcLure, 2020-08-28:
The ``F6000`` does not say to accelerate at 6000mm/s^2, it specified to move at a _speed_ of 6000mm/min (or 100mm/sec).

JeanCarloEM, 2020-08-28:
Got it! I said that, because in the configuration I did not find a configuration to limit speed (minimum or maximum). Therefore, I deduced that this is a problem of interpretation or a problem in the making of the "manual".

But no problem, so how do I limit the speed when the Z axis is involved?

Thank you!

ManuelMcLure, 2020-08-28:
You want to use ``DEFAULT_MAX_FEEDRATE`` in your configuration file, which sets the default values for ``M203``. ``M203`` (and by extension ``DEFAULT_MAX_FEEDRATE``) is exactly what you need to avoid overspeed on the Z axis. Note that if there is a movement in X, Y and Z the Z speed restriction may end up also slowing down the X and Y move since the move is planned for all the axes to reach the destination at the same time.

JeanCarloEM, 2020-08-28:
Once again, thank you very much for your collaboration.

So, in my language, using google translator, _feed rate_ does not indicate speed :).

But, because it is one of the few options that has something related, I tried it.

I changed from `` #define DEFAULT_MAX_FEEDRATE {400, 400, 50, 25} `` to `` #define DEFAULT_MAX_FEEDRATE {400, 400, 5, 25} ``, but the error persists.

ManuelMcLure, 2020-08-28:
Make sure you execute ``M502``/``M500`` after updating your firmware so the new defaults are taken into account. You should be able to type ``M503`` and look at the ``M203`` line it presents to make sure the values are correct.

ManuelMcLure, 2020-08-28:
If you keep having issues, please join the Marlin Discord to discuss them further. This bug tracker is not meant to be used for support. https://discord.gg/n5NJ59y

closed 2020-08-28 by JeanCarloEM