MarlinFirmware/Marlin #19166: [FR] No delta support for PowerLoss resuming

karabas2011, 2020-08-27:
1. when resuming delta, printer homes and moves to XY - crash.
So it needs to go down a bit, before XY moves

2. hard coded Z speed F200 too slow for delta

Foxies-CSTL, 2020-12-29:
@boelle @sjasonsmith
I confirm the problem and why put the label "Feature request" since it worked very well before 20201223.
 I validated this function on my printer and I built several firmwares for the FLSun QQS community in different combinations of stepper drivers ( A4988, TMC2208 and 2209). No user feedback since.
I also have a Delta QQS with a Hispeed aka Robin_mini clone with 2209Uart and the function has been activated since the end of September with #define POWER_LOSS_PIN -1 for the creation of the PLR ​​file on the SD card.
But since 20201223(Bump Distribution), the recovery is correct but the head crashes on the bed and its positioning movement is very slow.
Important note: if I active "#define DELTA_HOME_TO_SAFE_ZONE" in the configuration.h , the head does not crash but the recovery and the positioning are very slow !!

sjasonsmith, 2020-12-29:
@Foxies-CSTL did you have UBL enabled when it worked previously? With UBL enabled in the configuration file I think there is extreme risk of movement problems on delta right now.

Foxies-CSTL, 2020-12-30:
> 
> 
> @Foxies-CSTL did you have UBL enabled when it worked previously? With UBL enabled in the configuration file I think there is extreme risk of movement problems on delta right now.

Yes, UBL is always activated after a G28 and I put a G29Lx in the starting G code of the filament with a K = 0. Another note, when I compile in 32 step mode, the function no longer works.

thisiskeithb, 2022-02-22:
#23622 should address part of this issue.

Can you download and test the latest `bugfix-2.0.x`?

closed 2022-07-04 by thisiskeithb