MarlinFirmware/Marlin #19167: [BUG] Setting PROBING_MARGIN to 20 causes printer to hang early in print, setting back to 10 fixes the problem

kreuzhofer, 2020-08-27:
### Bug Description

Printer: Ender 3
Mainboard: 1.1.5 with silent steppers TMC2208
Environment: melzi_optimized
Marlin version: 2.0.6 release

I have set **PROBING_MARGIN** for my Ender 3 to 20. Since that change, the printer stops very early in the print. To be precise, it stops X and Y and E0 movements and just drives up Z super slowly but consistently, basically forever.

When I changed PROBING_MARGIN back to 10, the problem is gone.

### My Configurations

https://1drv.ms/u/s!At2kGv0IrYz9uZUoCnS-iRMHWl3B_A?e=8fbGUM

### Steps to Reproduce

1. Change PROBING_MARGIN to 20
2. Print something
3. Watch printer stop after beginning of print (5s literally)

**Expected behavior:** Printer should print normally

**Actual behavior:** Printer stops printing

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-08-27:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-08-27:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

kreuzhofer, 2020-08-30:
Will give it some testing in the next days

sjasonsmith, 2020-08-31:
Does it actually start printing, and stop mid-print, or does the problem occur while probing?

coolduke666, 2020-09-01:
have a similar problem ... my PROBING MARGIN = 30. printer stopps after printing 5cm of the skirt (RAMPS setup).

coolduke666, 2020-09-29:
is there a fix in 2.0.7?

boelle, 2020-09-29:
@coolduke666 or try  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) 2.0.7 is just a frozen copy of that anyway

closed 2020-11-06 by github-actions[bot]