MarlinFirmware/Marlin #19169: [BUG] UBL for delta - square too small.

karabas2011, 2020-08-27:
I set PROBE_MARGIN to 0 but  square still too small.
I need full bed calibration.
In smoothieware the bed is inside the probing square and corner points outside bed do not probe.
I try to set negative margins but got an error.
Why?

ellensp, 2020-08-27:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-27 by ellensp

thinkyhead, 2020-08-27:
> In smoothieware the bed is inside the probing square and corner points outside bed do not probe.

It works the very same way in Marlin. Your configuration must be flawed. Use the resources linked above and someone in the delta fan community will help you sort it out.