MarlinFirmware/Marlin #19170: [BUG] COLOR_UI With Mesh Bed Leveling

andreibobirica, 2020-08-28:
### Bug Description

In the procedure of mesh bed leveling, next when in Display appear "Click to Begin" ,Nothing happens if touched.
In general the touch screen works very well, only on that screen it doesn't work.
Since there are no buttons or anything, you can't even go back, and you have to take them.

### My Configurations

https://github.com/andreibobirica/Marlin/tree/2.0.x

### Steps to Reproduce

Click on Bed Leveling with Mesh Bed Leveling and COLOR_UI (no classic_ui)
Wait for homing
Wait for CLICK TO BEGIN to appear
Touch the screen, nothing is happening.

**Expected behavior:**
Step 5: Initiate a move to the next point
**Actual behavior:** 
Nothing

#### Additional Information
I tried to fix the bug but couldn't.
In menu_bed_leveling.cpp this procedure goes all the way to the stage
Step 4: Display "Click to Begin", wait for click
But no clicks are recorded afterwards.

Through octoprint and serial debugging I tried to figure out which functions were not working, and while on that screen the ui.use_click () function can be called and is working, the use_click function is executed if pressed on the screen.
However, nothing happens immediately afterwards.
I'm afraid all "click to begin" integrations are broken.

To solve this problem, I have temporarily removed the wait.
And I immediately moved on to step 5:
   //
   // Step 4: Display "Click to Begin", wait for click
   // Move to the first probe position
   //
   void _lcd_level_bed_homing_done () {
     if (ui.should_draw ()) MenuItem_static :: draw (1, GET_TEXT (MSG_LEVEL_BED_WAITING));
     / * if (ui.use_click ()) {
       manual_probe_index = 0;
       _lcd_level_goto_next_point ();
     } * /
     _lcd_level_goto_next_point ();
   }

appliedmike, 2020-08-29:
Thanks work for me, I now have MeshBL and Baby-Stepping menus :-)

andreibobirica, 2020-08-30:
I made this post to point out that the ui.use_click () function with the COLOR_UI doesn't work.
Eliminating this function is just a patch, it cannot be considered a solution.
I have not included the solution to the BUG, I have only included how to contain the problem.

rhapsodyv, 2020-09-03:
I will take a look

andreibobirica, 2020-09-05:
> I will take a look

Thanks a lot for the work.
If you have time, take a look at these two too.
I think maybe you might be interested.

https://github.com/MarlinFirmware/Marlin/issues/19051
https://github.com/MarlinFirmware/Marlin/issues/19141

rhapsodyv, 2020-09-25:
Confirmed and fixed.

PR #19514 (2020-09-26): Fix #19170 - Mesh Bed Level on Color UI
### Description

The color UI needs a control to handle clicks. 

### Benefits

Fix #19170.

### Configurations

### Related Issues

#19170
closed 2020-09-26 by thinkyhead