MarlinFirmware/Marlin #19172: [BUG] Ender 3V2 - Manual MBL does not work via the UI/Display on the

tubalainen, 2020-08-28:
### Bug Description

When selecting the "level" icon on the UI/Display the printer homes but stops and hangs. A Power cycle is needed to bring it back to life.

Doing the manual MBL via the terminal using the G29 command works like a champ - no issues.

### My Configurations
Config files are based on the templates provided on the 21 of aug - I have modded the template ones made for Ender 3 V2 to accommodate MBL.
The following changes made/un commented:
#define PROBE_MANUALLY
#define MANUAL_PROBE_START_Z 0.2
#define MESH_BED_LEVELING
#define RESTORE_LEVELING_AFTER_G28

[Config 20200821.zip](https://github.com/MarlinFirmware/Marlin/files/5140768/Config.20200821.zip)

### Steps to Reproduce

1. Turn on printer
2. Preheat PLA
3. Select "Level" in the UI/Display

**Expected behavior:**  Working MBL - not hanging

**Actual behavior:** The machine hangs and a power cycle is needed

tubalainen, 2020-08-28:
https://github.com/MarlinFirmware/Marlin/pull/18981

tubalainen, 2020-08-28:
Similar issues related to DWIN and the MBL: https://github.com/MarlinFirmware/Marlin/pull/18981

legaspikyle, 2020-08-30:
Good Day!
I'm just new to the 3d printing world and I'm still having a hard time trying to successfully compile the manual mesh bed leveling with the lcd bed leveling sub menu enabled as well. I still keep getting that error regarding the need for a programmable lcd controller :( Is there anything I can do about this? I currently have a ender 3 v2 printer thank you!

boelle, 2020-08-30:
@legaspikyle 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

tubalainen, 2020-08-30:
> Good Day!
> I'm just new to the 3d printing world and I'm still having a hard time trying to successfully compile the manual mesh bed leveling with the lcd bed leveling sub menu enabled as well in the version 2.0.6.1. I still keep getting that error regarding the need for a programmable lcd controller :( Is there anything I can do about this? I currently have a ender 3 v2 printer thank you!

Hi! Just leave out the LCD part and uncomment the four rows that I mention above and it should compile - but with the bug/issues with the UI/display.

damien6468, 2020-08-30:
Hi, I confirm the UI bug is still here with the last build (2.0.6.1). I tried the same parameters as @tubalainen

legaspikyle, 2020-08-31:
> @boelle 
> 
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.
> 
> Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

My Configurations:

Configuration files were taken from the firmware update: 2.0.6.1 files last Aug 28, 2020
The following lines are uncommented:
#define PROBE_MANUALLY
#define MANUAL_PROBE_START_Z 0.2
#define MESH_BED_LEVELING
#define RESTORE_LEVELING_AFTER_G28
#define LCD_BED_LEVELING **
#define MESH_EDIT_MENU
#define LEVEL_BED_CORNERS
#define LEVEL_CENTER_TOO

** I tried to see and compile several times to see which is the cause of the failure in the compilation and it all ended down to enabling LCD_BED_LEVELING causes an error message in the SanityCheck.h -- #error "LCD_BED_LEVELING requires a programmable LCD controller."
[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/5148587/Configuration.Files.zip)

ellensp, 2020-08-31:
The display on a ender 3 v2 does not support LCD_BED_LEVELING its that simple.
Duplicate of #19202
Duplicate of #18457

To re iterate
"You have a DWIN_CREALITY_LCD, this LCD is smart, It gets told what to display when via simple serial commands. But it has limits you cannot directly control what the pixels are doing.
The LCD_BED_LEVELING feature requires a lcd that you can control individual pixels.
This is simply not a option you can have with this display."

At least until someone rewrites the firmware in the LCD

closed 2020-08-31 by ellensp

tubalainen, 2020-08-31:
> The display on a ender 3 v2 does not support LCD_BED_LEVELING its that simple.
> Duplicate of #19202
> Duplicate of #18457
>
> To re iterate
> "You have a DWIN_CREALITY_LCD, this LCD is smart, It gets told [...]

Hi @ellensp 

@legaspikyle is not the original issue creator - I am.

The issue is not that the LCD_BED_LEVELING is not support due to that the V2 uses the DWIN.

The reason why I created the issue is that the MBL does not work via the UI/Display. The icon is there, it can be selected, the printer auto homes and then hang. This option works ok with BLTouch, so why not with MBL?

Please do not close my issue until resolved or deemed not feasible.

wikityrey, 2020-09-05:
DWIN LCD is smart?!!! It certainly is not smart if it can't do what other LCDs can do!

tubalainen, 2020-09-05:
Well, the feature has (currently only?) been removed from the menu.

https://github.com/MarlinFirmware/Marlin/commit/277042bc4e79fe89824e941605e830391ebcd3c0

legaspikyle, 2020-09-05:
I saw that there was an original issue created but it was closed as well so i thought this might bring it back up :( Correct me if I'm wrong but I think the problem is that in the dwin files, there isn't an available block of code for a sub menu for manual mesh bed leveling that's why it doesn't work? I'm guessing this is because in ABL, it can be done through a "one-click" setting unlike MBL which needs for a few other settings such as moving and setting the height manually through the z-axis through points on the bed? I'm not entirely sure tho

tubalainen, 2020-09-05:
> I saw that there was an original issue created but it was closed as well so i thought this might bring it back up :( Correct me if I'm wrong but I think the problem is that in the dwin files, there [...]

Might be, but for us (frustrated users) more info on the matter would be great so that we can understand, support and help the community, that´s asking the same question. I am guessing the same thing and by reading reddit, the issues here and some facebook posts that is my conclusion as well.