MarlinFirmware/Marlin #19173: [FR] need cold-bed for  prinnting.control?!

aliys97, 2020-08-28:
Hello. I'm trying to build bio3d printer. I need cold bed for printing bed. I use marlin code. and Peltier for cooling. I attached Peltier pins where define for heatbed pins.

The problem is i cant control temperature. Help me to control bed in clod temperature.???????????

AnHardt, 2020-08-28:
In the simplest case you could try to invert the PWM-polarity.

Maybe swapping the cases here:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L1143-L1161
can help.  Swap the expressions in every pair of occurrences where `temp_bed.soft_pwm_amount` is set. For the PWM case try 256 or 128 minus the original value.

That could work but will for sure confuse THERMAL_PROTECTION_BED.

ellensp, 2020-08-28:
You seem to be expecting us to to write you a custom marlin 1.1 (from your other posts) 
Marlin 1.1 no not being developed any more.

closed 2020-08-28 by ellensp

aliys97, 2020-08-28:
> In the simplest case you could try to invert the PWM-polarity.
>
> Maybe swapping the cases here: [...]

thanks, it worked. is it possible to do such inverting logic for PID controlling?