MarlinFirmware/Marlin #19174: Scaling of Gcode

dscvejl, 2020-08-28:
I could be nice if the firmware support on screen scaling of the g-code.

I know that this can be done in the slicer, but find it a nice feature that you could scale the model on the printer for newbies or you are too lazy to slice the figure again.

ellensp, 2020-08-28:
scaling in Z requires re slicing. Not possible in firmware

closed 2020-08-28 by ellensp

AnHardt, 2020-08-28:
If you are not scared by a lot of side effects you could try to set scaled steps/mm values. Better this would be done at the other end of the chain, when parsing g-code, like inch-mode. But since g-code macros are also used internally the side effects are also unpredictable.

dscvejl, 2020-08-28:
Well I did not expect that this request was accepted with out a fight :-)

My thought was that the firmware could take the steps/mm and multiply it with e.g. 1.1 before starting the print, for E, Y, X and Z.
Though I don't know if E is linear, because it is controlling volume.

It could be useful when two parts does not match and you just want to scale one of it by e.g. 1%.

My idea was more to make the 3d printer more like a copy machine, where you have the zoom options.

AnHardt, 2020-08-28:
>  copy machine

Better do that in the slicer.

Scaling steps/mm is far from ideal. Every Marlin internal distance is scaled with that. (bed-size, probe_offsets, mesh data, position of the cleaning station, ...)

If E has to be scaled linear or quadratic depends on volumetric mode.