MarlinFirmware/Marlin #19175: Auto center of model, when outside build plate

dscvejl, 2020-08-28:
I would be nice if the firmware gives a warning, if part of the print is outside the build plate.

A nice feature would be that the firmware ask, if you want to auto center the model.

ellensp, 2020-08-28:
Marlin processes model layer by layer. it is not possible to do this in firmware

closed 2020-08-28 by ellensp

dscvejl, 2020-08-28:
What I'm talking about is a pre-script that runs through the entire g-code before starting the printer.
I know that it adds to the time before you start printing, but most printers is waiting for the bed to warm anyway.

It would be nice with at least a warning that some of the g-code is outside the print area, before you start a multiday print.

AnHardt, 2020-08-28:
At least a message when software endstops are/have been hit during the print could be helpful in some cases to be warned for cropped parts.
Shifting/centering the part is not possible before the whole part has been seen/printed - but how much to shift, has to be known before the first line is printed.

AnHardt, 2020-08-28:
> pre-script

Yes. But please before sending to Marlin.