MarlinFirmware/Marlin #19178: [BUG] behavior change "TOOLCHANGE ZRAISE"

DrumClock, 2020-08-28:
**Previously:**
1. print T0
2. Z raise by 2 mm
3. change of tool T1
4. move to the printing position
5. Z return of 2 mm

**Now:**
1. print T0
2. Z raise by 2 mm
3. change of tool T1
4. Z return by 2 mm
5. move to the printing position

### Current behavior is bad!
### The material comes out during the crossing !!!

sjasonsmith, 2020-09-11:
Please attach your Marlin configuration files, as requested by the issue template. Marlin behavior is extremely configurable, and it is very difficult to debug any issue without knowing how the firmware is configured.

DrumClock, 2020-09-11:
Hi @sjasonsmith ,
Here are the Marlin configuration files.  [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5211022/Configuration.zip)
Among other things, these files are also in the link above ( Marlin-bugfix-2.0.x 16-7-2020.zip )

DrumClock, 2020-09-23:
Hi,
I tested the latest version of bugfix 2.0.x and it's the same behavior when changing tools.
The previous behavior was definitely lighter for the press, please undo the changes.

1. print T0
2. Z axis raise on 2 mm
3. change of tool T1
4. move to the printing position
5. Z axis return on 2 mm

DrumClock, 2020-09-28:
**Is anyone solving this?**
This GitHub system is a bit strange.
I don't know if anyone is solving it or not or if it is already solved.
**There is simply no feedback!**

thisiskeithb, 2020-09-28:
Keep in mind that all the code here is contributed by unpaid volunteers who work on it in their free time.

I've added a Help Wanted label to hopefully attract attention of someone who has the hardware, interest, and ability to fix/implement the change.

Quas7, 2020-10-07:
I am currently on bugfix e8177735a and for me on every toolchange (T0->T1, T1->T0) on my single-nozzle setup the Z axis is raised +2mm.
I searched my configs now 3 times for that parameter but after reading this threat, it seems to be hardcoded. ;)

I will try to search throuhg the change list to find the actual implemenation.

Quas7, 2020-10-07:
If someone has time to check as well, I found these in the commits for "tool" (git log --all --grep=tool)
From hottest candidates to lowest:

commit 7b2f7a94fa596e4fd24e6f7b318c68514fcf1806
Author: cbaugher <kb9ydn@gmail.com>
Date:   Fri Aug 7 18:06:25 2020 -0500
    Fix Z height after tool change (#18951)

commit 46763efb75ee92e0979adfd3988607c0a32eaf25
Author: Jason Smith <jason.inet@gmail.com>
Date:   Wed Oct 9 17:48:00 2019 -0700
    Fix tool-change move with hotend offset (#15491)

commit 0391e5bcc735d25c4618026f2656161c2631e592
Author: Giuliano Zaro <3684609+GMagician@users.noreply.github.com>
Date:   Fri Jan 10 23:12:21 2020 +0100
    Prevent Z misaligment on tool change (#16518)

commit 79810b7fcb4149896180d757e2fbb1ec7a8fc217
Author: InsanityAutomation <38436470+InsanityAutomation@users.noreply.github.com>
Date:   Tue Jan 21 03:51:23 2020 -0500
    G12 defaults per tool. Event G-code for post-toolchange (#16554)

commit 44b71103dfb7ac66847122ca2eda475beb16b4a1
Author: Scott Lahteine <github@thinkyhead.com>
Date:   Thu Apr 23 16:19:51 2020 -0500
    Patch tool_change typo

commit 28518c2352efb71e335990413f93220347410618
Author: studiodyne <42887851+studiodyne@users.noreply.github.com>
Date:   Sat Apr 25 19:20:42 2020 +0200
    Save/Restore leveling on toolchange for singlenozzle too (#17682)

DrumClock, 2020-10-07:
HI, @Quas7 
described above does not even correspond to the problem I mentioned.

**The problem is in the transfer when changing the tool, not in ZRAISE.**
**This is best understood in the videos below.**

For me, ZRAISE is absolutely fine.

Quas7, 2020-10-08:
Yes, it seems to be unrelated to your sequence issue.
I also use a delta-printer that might be the reason for this strange behavior.

DrumClock, 2020-10-31:
Hi, @thinkyhead 
**Is anyone working to fix this error?**
Even today's bugfix2.0.x didn't solve anything.

**This is a serious problem when changing tools.**
Thank you.

sjasonsmith, 2020-10-31:
One real challenge is that most people don't have multi-tool printers. One of our most active multi-tool contributors has been too busy with other things to do much this year. If one of you is willing to step up to help resolve some multi-tool issues, I'm sure the community would greatly appreciate it!

DrumClock, 2020-11-08:
Hi, please modify the "**Z raise distance for tool-change**" function as follows to return the Z axis:

1. after completing the `SWITCHING_NOZZLE` or `SWITCHING_EXTRUDER` function

2. only when the print head is in the print position for the tool being replaced
   (**as was the case in FW bugfix 2.0x from 16-7-2020**)

If you need to test adjustments before PR, I'm fully available.

Thank you very much

Protean-Man, 2020-11-14:
I have the exact same problem outlined here: #19150

It has been a problem for months now and I also have had zero success fixing it. What I think is it is related to Bed Leveling (AUTO_BED_LEVELING_UBL) and TOOLCHANGE_NO_RETURN.

What I have found is that if I disable Auto Bed Leveling and TOOLCHANGE_NO_RETURN the issue goes away. This is not my goal though because both of those make massive differences in print quality.

sjasonsmith, 2021-01-04:
Please refer to the close duplicate for newer configuration files and some videos.
#20648

DrumClock, 2021-01-24:
### Hi, can someone please address this issue?

DrumClock, 2021-01-31:
Hi, @sjasonsmith 
even today's update bugfix 2.0.x does not solve the problem of a bad tool change procedure.
Can someone please do this so that it is correct as in the release 16.7.2020?

DrumClock, 2021-03-03:
Hi @thisiskeithb

It's been here since 28 Aug 2020 and no fixes.
I'm using  old FM from 16-7-2020 which is fine.

Should I leave it open or should I close it?

Protean-Man, 2021-03-03:
I would like this to be open until it can be fixed. I have opened other issues related to this problem that have all gone stale and closed without any changes being made. I appreciate all the work the core devs do and want Marlin to push into the future but without a functioning multi-tool backend I can't see Marlin going beyond the base level of printers. If I could fix it myself I would but I am not versed enough in coding for it.

DrumClock, 2021-04-05:
Ahoj, už vím kde je pravděpodobně  problém.
Marlin ve verzi ze dne 16-7-2020 nedokončil funkci tool change kdy nevrátil osu Z o nastavenou hodnotu.
Toto při tisku zařídil G-code s výškou vrstvy.

Tato "chyba" však byla pro tisk s více extrudery lepší, protože se hlava vrátila na pozici Z až po přejezdu na 
tiskovou pozici.

Tím nedocházelo k "otření" vytékajícího filamentu na jném místě.

DrumClock, 2021-04-05:
Navrhuji při tisku nevracet osu Z.
Tuto funkci nechat jen když se netiskne .

thinkyhead, 2021-04-16:
Duplicate of #19150