MarlinFirmware/Marlin #19181: [BUG] Hybrid_Threshold breaks StealthChop

phazei, 2020-08-28:
Marlin Bugfix 2.0.x from Sunday
12v SKR 1.4, BTT TMC2209 1.2, 1.8deg 0.9A steppers
CoreXY, 256 babysteps, sensorless homing

I've been using StealthChop without hybrid just fine.
I was attempting to enable StealthChop HYBRID_THRESHOLD which I haven't used before to solve issue #18286 

As a baseline: with StealthChop and NO hybrid, everything worked fine other than the resonance issue.  I could auto home.  I could move at about F9000 horizontaly/vertically and F6000 diagonally until it made horrible squealing noises.
If I went to the TMC settings on the LCD and manually disabled StealthChop on X & Y, it had the normal slight buzz, and I could still auto home.  Also, with StealthChop off, I could then move the carriage in any direction up to at least F18000.

When turning HYBRID_THRESHOLD on and changing nothing else, it breaks movement & StealthChop.

LCD menu shows StealthChop and Hybrid are enabled, thresholds are 100mm/s (also tried 255, no difference)
First of all, if X or Y move via the LCD or when homing it sounds like it's using SpreadCycle, even though it's moving much slower than the threshold, (3000mm/min default).  Using G-code, under F2600mm/min it sounds like StealthChop, could just because it's slow though.

If I attempt to auto home, sensorless homing works perfectly for X and Y.  But when it attempts to move to the center to home Z (bltouch), it gets about 1mm and stalls.  The default 3000 feedrate moving diagonally is too much for it.

If I go to a gcode terminal:
G1 X0 Y0
G1 X100 Y100 F2000; works
G1 X200 Y200 F3000; fails, moves about 1mm in the direction and stops

G1 X0 Y0
G1 X300 F4000; works
G1 X0 F5000; fails

So it can no longer move anything like it did before HYBRID_THRESHOLD was turned on, it seems to entirely break StealthChop and doesn't even attempt to use SpreadCycle.  If I try to move at speeds well over the 100mm/s threshold that it could easily do before, it just makes bad noises and jiggles around a little.

https://github.com/phazei/Marlin/blob/SKR_1.4_FLSUN_CUBE/Marlin/Configuration.h
https://github.com/phazei/Marlin/blob/SKR_1.4_FLSUN_CUBE/Marlin/Configuration_adv.h

phazei, 2020-08-28:
When I posted I'd tested thresholds of 0, 1, 100, and 256.  It didn't seem to at SpreadCycle at all with 100 and 256.  Was just fine StealthChop at 0, and all SpreadCycle at 1.

So, if I set the threshold to 50, so 3000mm/min  and move diagonally at F2500, it's using SealthChop albeit noisily, and it gets worse and worse as it gets closer to F3000,  then if I move it at F4000, it starts using SpreadCycle and is ok.  So it is switching.

I think when the threshold was too high, it just stalled out, made noise, and skipped too much to even get to over the threshold to start using SpreadCycle.

So the issue still remains, it's a little more clear now so I updated the title.

So to be more succinct: 
**StealthChop alone works fine at up to F6000 before succumbing to resonance issues.
When Hybrid Threshold is enabled, StealthChop entirely stalls out at F3000.**

daleckystepan, 2020-08-29:
I have noticed same thing maybe a year ago on AVR 2560 and TMC2130 (Geeetech A10 with TMC2130 over SPI). I thought that it was my mistake as I was new in 3D printing. I have never tried it again with current Marlin. I ended up with disabled hybrid treshold for all motors.

phazei, 2020-09-06:
Some adjustments and more observations.
My X & Y stepper motors have been 16T GT2 pulleys.  I've just upgraded to 20T GT2 pulleys.  It's actually made a big difference, and if those are more common sizes for use on the X & Y, it might explain why more people don't come across it.

So the root bug is still there, but the speeds at which they happen have increased, enough, I think, to make StealthChop usable most of the time, so I don't need to deal with hybrid which appears broken.

So, with the larger pulley's and StealthChop only, horizontal speeds go up to 12,500mm/min before squealing, and diagonal speeds go up to 10000mm/min.

With hybrid, there are still issues.  All axis thresholds are 100mm/s.  Move Z on the LCD is set to 16*60mm/min, and is significantly louder/whooshy, that's well below the threshold, so it doesn't seems to be using StealthChop with hybrid mode on at all.  XY movements do appear to use StealthChop, but again, at speeds half of what StealthChop is capable of without hybrid threshold on.  I can move horizontally/vertically at up to 5000mm/min, at 6000m/min the motors don't have enough torque to move and instead of going horizontally/vertically, it'll end up going diagonal a very short while and then stalling.
Diagonally,  sometimes it will go at 4000mm/min, but half the time it stalls out, goes 3000mm/min just fine.  So those speeds have increased, but not quite enough to home.  HOMING_FEEDRATE_XY is 3000, so it should, but it stalls out, maybe becuase XY_PROBE_SPEED is 6000?

Either way, since the motors don't need to rotate as fast to achieve the same rates, I think I can just use StealthChop and not worry about hybrid mode for now.  I would be really nice to be able to use though, but unless StealthChop can work at least 6000mm/s diagonally using hybrid threshold (which it can without it) then there's no way to use it.

Current motors are 48oz/in, will be getting in some 72oz/in soon so maybe that'll increase my rates some more.

pediegoivan, 2020-09-20:
Ouch! I added a new issue and there was already this thread.
I have found that the vibration somehow is dependant on the hybrid threshold selected value.
And disabling stealthchop in firmware is not resulting in spreadcycle. The only way that I have found to run all the time in spreadcycle is to set a hybrid threshold value of 1 with stealthchop enabled.
In this way I can run the drivers up to 250 mm/s and more with 16mm pulleys. I will try faster speeds later.

boelle, 2020-09-20:
@pediegoivan is this the one you added`? https://github.com/MarlinFirmware/Marlin/issues/19450

pediegoivan, 2020-09-20:
> @pediegoivan is this the one you added`? #19450

Yes! I'm sorry about that.

freighttt, 2020-09-21:
> Ouch! I added a new issue and there was already this thread.
> I have found that the vibration somehow is dependant on the hybrid threshold selected value.
> And disabling stealthchop in firmware is not [...]

Are you saying that with hybrid threshold at 1(which means it’s spreadcycle always), you have less noise and resonance then when steathchop is off?

pediegoivan, 2020-09-21:
> Are you saying that with hybrid threshold at 1(which means it’s spreadcycle always), you have less noise and resonance then when steathchop is off?

I've found that disabling stealthchop in firmware does nothing. M569 still works, and if stealthchop was enabled, it will stay enabled. M913 does stop being recognized.
Firmware (hard) disabling the stalthchop is not working properly. M569 is. And to finally answer you question. Now that I have found how to properly disable stealthchop, disabling it by M569 has the same effect as M913 X1 Y1 Z1.
Oh, one additttional thing.
Each time I upload a new version of the firmware, M913 X (and only X) changes to a ridiculous value.

phazei, 2020-10-23:
This is still an issue, no clue what I can do to further assist with a fix.

pediegoivan, 2020-10-23:
Yes. I agree, it is an issue. I have no idea either. the issue should not be closed just because of that.

boelle, 2020-11-22:
@teemuatlut can you maybe shine some light on what might be going on here? you are after all the Trinamic expert

Marlor, 2020-12-26:
Should confirmed bugs like this be auto closed?

NeoMod, 2020-12-26:
> Should confirmed bugs like this be auto closed?

Definitely no, in my opinion issues like this need to stay open so that a) is clear for everyone that the issue is confirmed and is open to solution b) the community knows that the issue is recognized and knows what to do (if possible) to avoid the issue

ellensp, 2020-12-26:
Its a bot, it only comments about closing the issues if no one has commented on the issue in a month. All you need to to keep it open is comment, to  show someone still cares about it. A comment like "tried latest's bugfix, issue remains" after trying it would be good.  Post is 4 months old, has anyone actually tried  the current bugfix?

sjasonsmith, 2020-12-26:
Unfortunately like so many others the comments seem to have added two or three additional issues on top of this one.

I noticed that the original post seems to involve 256 microsteps. It is usually not practical to run with 256 microsteps. I would like to know if it is still a problem with a lower microsteps value.

Please re-test with 16 microsteps and report whether it is still a problem.

I am speaking only of the issue in the original post. Other issues should be reported elsewhere.

NeoMod, 2020-12-26:
Running with 256microsteps at 32bit I can confirm the issue is still there.

sjasonsmith, 2020-12-26:
Using 256 microsteps is usually a misconfiguration. It adds an incredible processing burden to Marlin, so maximum performance can drop considerably. I suspect that at this high step frequency the step timing is "jittery", and the TMC driver has a hard time determining whether or not the threshold has been crossed.

This is why seeing test results at 16 microsteps is so important. If it works fine there, the solution is to turn down your number of microsteps, and rely on the TMC driver's interpolation capability to provide the smoothness you want.

phazei, 2020-12-29:
I can confirm that of a week or two ago the issue still exists in bugfix.  I figured since it's a 32bit board the 256 microsteps ought to be fine.  I'll try it at a lower microstep value though and see if that works as a work around.  Even if that works, I don't believe it to be a solution.

Stealthchop works just fine by itself with 256 microsteps up to 12000mm/min with these newer 59Ncm 2A motors I've gotten, I run them at 900mA and they stay nice an cool and don't skip steps.  But if I turn on Hybrid mode, they stall at 3000mm/min.  Even if the hybrid threshold is really high.  So no change with different motors.

I've recently gotten a PI, and was going to try the setup with RRFW sometime in the next month and see if the problem exists there as well.

sjasonsmith, 2020-12-29:
I expect you will see significant improvements by going down even to 128 microsteps, but really most of us just use 16 and allow the TMC's interpolation to do the rest.

It's really not reasonable to expect software-based pulse generation to be able to maintain perfectly timed output every 6.25 microseconds, as your posted configurations would require. Even tiny periods of interruption would throw that off by a significant percentage.

phazei, 2021-01-25:
Ok, so I tried changing my micro steps from 256 down to 64.  It did improve things a bit, but it's still weird.

Stealthchop only mode, I can `G0 F40000` mm/min all over just fine, so realistically, I don't think I even need it because it can go faster than I'm possibly able to print and the frame wouldn't even stay still with movements that fast.

I turn on hybrid_threshold and set it to 200, 600, or 1000 (ones I tried) mm/s on X & Y and then I can't even do `G0 F12000` mm/min it stalls, can't go any faster than that at all.

But if I make hybrid_threshold 100 mm/s,  it does seem to work ok up to like F30000

closed 2021-03-03 by github-actions[bot]