MarlinFirmware/Marlin #19186: [BUG] Display buttons stop working while printing

TheNitek, 2020-08-29:
### Bug Description

When printing via SD card on my Anycubic Predator (SKR 1.3) the displays (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) controlls stop working. When pushing the encoder it beeps very breefly, but the menu is not shown (display stays on the info screen)

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5144913/Marlin.zip)

### Steps to Reproduce

1. Start SD print
2. push encoder "button" on the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

**Expected behavior:** Menu is shown (e.g. to stop the print)

**Actual behavior:** Info screen stays on display

#### Additional Information

* Sometimes, in very rare cases, the menu will show, but most of the time it does not work.
* When not printing the dispaly works fine
* I tried two different displays, both have the same issues

rhapsodyv, 2020-09-03:
Are you using a BTT serial TFT?

Because that BTT tft is not a real LCD. It emulates the pulses and shows the screen.
It could be a problem in the TFT firmware.

Did you test it with the last bugfix? 
Because there's a recent fix that give some conflict with some board pins too.

AnHardt, 2020-09-03:
If on a Marlin-native display this can be display-throttling. If the processor is cloth to be overloaded display updates are delayed or omitted to try to avoid stuttering of the printhead. That may feel like the buttons are not or delayed working. But usually that is the lesser evil.

TheNitek, 2020-09-03:
I also tried a "real" REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, so not the emulated one. Same issue.

Overload might be what's happening, but I would be surprised if this happens during non-complex (straight lines in the bottom layer) prints on a 32bit-SKR-Board.

hamster65, 2020-09-06:
I had this, too, for some time on my MKS S Gen L board. Increasing buffers solved it, but now it takes very long for feed rate changes to react.
#define BLOCK_BUFFER_SIZE 64
#define BUFSIZE 16

closed 2020-10-31 by github-actions[bot]