MarlinFirmware/Marlin #19188: [BUG] Bed leveling issue

KZL1992, 2020-08-29:
Hello,

I have been using my Anet A8 equipped with a BLTouch and Marlin 1.1.8 for quite some time now. I recently switched to version 2.0.6.1 (and tried 2.0.X bugfix last night)

The bed leveling does not work for some reasons. So I tried to debug afaic.
I attached the comparison of the bed leveling using 1.1.8 and 2.0.6.1

[BedLeveling.1.1.8.txt](https://github.com/MarlinFirmware/Marlin/files/5145034/BedLeveling.1.1.8.txt)
[BedLeveling.2.0.6.1.txt](https://github.com/MarlinFirmware/Marlin/files/5145035/BedLeveling.2.0.6.1.txt)

For this test I used the default Anet A8 configuration available in the repository + BLtouch activated, but I also attached my own configuration files +/- SD support to get enough space. Anyhoo this just shows that my BLtouch and returned information are working

[Configuration.1.1.8.txt](https://github.com/MarlinFirmware/Marlin/files/5145037/Configuration.1.1.8.txt)
[Configuration.2.0.6.1.txt](https://github.com/MarlinFirmware/Marlin/files/5145038/Configuration.2.0.6.1.txt)

From the log files I attached I observed an timing issue related to BLtouch and this only happens in the last versions. Can someone confirm this issue ?

danis101, 2020-09-04:
M420 L0 S1 in start -gcode after "HOME" line should do the job

boelle, 2020-09-04:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

KZL1992, 2020-09-14:
I just downloaded the last bugfix branch and the issue is the same.

daleckystepan, 2020-09-15:
You can try to experiment with #define ENDSTOP_NOISE_THRESHOLD 2, you can try 5 for example.

KZL1992, 2020-09-16:
Hi,

I tested it and it creates a weird shift along z while printing. 
The y axis get shaky while doing bed leveling but at least it starts 2 measured points.
I just want to stress that the firmware 1.9.X is working, so I doubt about an hardware issue. 

Cheers,
Marco

KZL1992, 2020-10-17:
Version 2.0.7.2 is able to run the bed leveling until the end but then the returned matrix is empty. I will try to provide some additional information once I activate the debug mode again

closed 2020-11-25 by github-actions[bot]