MarlinFirmware/Marlin #19189: SKR1.4: '1047' for PT1000 thermistor on TEMP_SENSOR_0 causes serial communication loss

Drzet, 2020-08-29:
### Bug Description

In Marlin 2.0.6.1 choosing '1047' for PT1000 thermistor on TEMP_SENSOR_0 causes loss of serial communication - both USB cable and ESP2866 

### My Configurations
SKR1.4 with x5 TMC2209 in normal mode. PT1000 probe on extruder 1, 100K probe on bed. Second extruder not connected (yet)
ESP2866 with ESP3D firmware plugged into board. No LCD screen
Config files attached.

### Steps to Reproduce

Choose 1047 (PT1000 with 4.7K pullup) as TEMP_SENSOR_0 (or any?)

**Expected behaviour:** Normal

**Actual behaviour:** Unable to connect to the board with either USB cable or through ESP3D. 
If 1 (100K thermistor) chosen for temp sensor, both USB and wireless working as expected. But this obviously gives wrong temp readings.

#### Additional Information

This is a result of countless attempts to recompile Marlin to get serial communication working. I am aware of other reports of serial communication problems which are believed to be caused by UART/serial conflicts with TMC drivers
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/5145251/config_files.zip)
. In my case I was able to pin down the issue to this seemingly unconnected setting.
Please try and reproduce. Maybe my SKR is somehow special ?

ellensp, 2020-08-29:
I suspect if you had a marlin controlled LCD you would see some form of temperature error followed by printer halted.
Try adding a "#define BOGUS_TEMPERATURE_GRACE_PERIOD    2000"  to your Configuration.h  and make sure you have a pc connected to the computer with something like proterface to catch any errors.

Drzet, 2020-08-29:
Well !
BOGUS_TEMP trick didn't work but I simply measured the probe with a multimeter and sure enough got 103KOhm
Thanks, issue closed

closed 2020-08-29 by Drzet

AnHardt, 2020-08-29:
Did you read the multi-meter correct?
1030 Ohm or 1.03 KOhm would be a quite normal resistance in between of 7 and 8 °C. With a assumed exactness of the multi-meter of not better then +-2% the temperature range would already be 3 - 16 °C - so quite normal.

Drzet, 2020-08-30:
> 103K Ohm
One hundred and three thousand Ohms

I have a few of those cartridges, there are no marking on them so I assumed that one was PT1000 (1kOhm), but it wasn't. Couldn't see temp with no LCD and no serial

AnHardt, 2020-08-31:
So more likely a 100K thermistor at a bit less than 25°C.

> there are no marking on them

> Couldn't see temp with no LCD and no serial.

Both usually a bad idea when setting up for new parts. 
Groping in the dark.