MarlinFirmware/Marlin #19190: [FR] Add Extruder Position to M114

D1plo1d, 2020-08-29:
### Description

Add the extrude actual position to M114 responses so that host software can track the extruder's movement. Eg:

`X:0.00Y:0.00Z:0.00E:0.10 Count X: 0.00Y:0.00Z:0.00E:0.07\n`

I use `M114` in conjunction with `G1` commands to move the axis continuously while the user holds the move button. This works well (lol, -ish) on X, Y, Z but the E position is not included in M114 responses so it needs a work around.

Sending an E position would allow me to remove this work around (albeit in the very distant never when everyone has upgraded their firmwares - so maybe not the top priority, still it would be nice).

### Feature Workflow

1. Host software sends `M114`
2. Marlin responds with `X:0.00Y:0.00Z:0.00E:0.00 Count X: 0.00Y:0.00Z:0.00E:0.00\nok\n`

thisiskeithb, 2020-08-30:
Enable `M114_DETAIL` to get the extruder position. Also note that some slicers will reset this value on every layer.

See https://github.com/MarlinFirmware/Marlin/pull/16026 for more information/comments.

closed 2020-08-30 by thisiskeithb

D1plo1d, 2020-08-30:
Ah, that makes sense. Thanks!

AnHardt, 2020-08-31:
> Also note that some slicers will reset this value on every layer

... or may use 'extruder-relative-mode'.