MarlinFirmware/Marlin #19194: Firmware JGAurora A5 please me!!

servitecjc, 2020-08-30:
Hola buenas, tengo una JGAurora A5 y hace mas de 2 meses mi placa mks_gen_L v1.0 dejo de funcionar, compre una placa MKS SGEN L v1.0 y no consigo ningun firmware que haga que mi ordenador la haga funcionar ya que la pantalla tampoco hace que responda. He intentado mil maneras de hacer el firmware, siendo la mas eficaz usar vscode con platformio pero despues de mas de un mes aprendiendo, lo mas que consigo con un firmware compilado exitosamente, es que me parpadeen los leds azules sin parar... alguien podria pasarme un firmware para esta placa y mi maquina?? Tengo puesta la TFT28 4.0 que me venia de fabrica. Mil gracias

swilkens, 2020-08-30:
Makerbase provide firmware sources, try those first;

https://github.com/makerbase-mks/SGEN_L

closed 2020-08-30 by ellensp

ellensp, 2020-08-30:
Marlin already has config examples for your machine https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/JGAurora/A5 
You just need to make a few changes for your new mks_sgen_l controller.
I have made them for you. all but one change  here
[JGAurora_A5-mks_sgen_l.zip](https://github.com/MarlinFirmware/Marlin/files/5146625/JGAurora_A5-mks_sgen_l.zip)
You need to set the pin you have the filament run out sensor plugged into.
In Configuration_h is the line " #error #define FIL_RUNOUT_PIN {set this} "
you need to remove the  "#error" and replace {set this} with the pin your using eg P1_31   so the line becomes "#define FIL_RUNOUT_PIN P1_31" or what ever pin you used.

servitecjc, 2020-08-31:
> Marlin ya tiene ejemplos de configuración para su máquina https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/JGAurora/A5
> sólo tiene que hacer algunos cambios para su nuevo controlador de mks_sgen_l.
> Las he hecho para ti. todos menos un
> cambio aquí [JGAurora_A5-mks_sgen_l.zip](https://github.com/MarlinFirmware/Marlin/files/5146625/JGAurora_A5-mks_sgen_l.zip)
> Necesitas ajustar el pin en el que tienes el sensor de salida del filamento.
> En Configuration_h es la línea " #error #define FIL_RUNOUT_PIN "establecer
> esto" es necesario eliminar el "#error" y reemplazar el pin con el pin que usa, por ejemplo, P1_31 para que la línea se convierta en "#define FIL_RUNOUT_PIN P1_31" o lo que haya utilizado.

Mil gracias, conseguí compilar el archivo, me ayudaron por discord en Marlin y finalmente le puse el pin, P1.23. Ahora tengo otro problemilla, la placa con ese firmware cuando carga parpadea el led azul pero se queda fijo y aunque pronterface me la lee, se desconecta al cabo de unos segundos. Me di cuenta que no había indicado que tengo puestos los tmc2130 así que los cambie y compile de nuevo y ahora no se me queda el led azul fijo, solo se queda el rojo y no me la reconoce el pronterface. No se si tengo los TMC2130 bien colocados, no les he quitado ningún pincho en su parte inferior y le coloque el pincho en la punta
[image]

ellensp, 2020-08-31:
You have physical endstop, you need to remove DIAG jumpers or they interfear with endstops.
[image]
```
And you need in Configuration.h

#define X_DRIVER_TYPE  TMC2130
#define Y_DRIVER_TYPE  TMC2130
#define Z_DRIVER_TYPE  TMC2130
#define E0_DRIVER_TYPE TMC2130
#define E1_DRIVER_TYPE TMC2130
```

servitecjc, 2020-08-31:
> You have physical endstop, you need to remove DIAG jumpers or they interfear with endstops.
> [image]
> 
> ```
> And you need in Configuration.h
> 
> #define X_DRIVER_TYPE  TMC2130
> #define Y_DRIVER_TYPE  TMC2130
> #define Z_DRIVER_TYPE  TMC2130
> #define E0_DRIVER_TYPE TMC2130
> #define E1_DRIVER_TYPE TMC2130
> ```

Eso ya lo he modificado, has visto el driver? sobra la patilla que le he puesto?

servitecjc, 2020-08-31:
Everything is almost ready friend, with pronterface I get everything to move except the hotend that I can't get hot. He sends me an error and disconnects. And the MKS TFT28 4.0 series screen that comes with the machine does not control anything.

servitecjc, 2020-08-31:
Connecting...
Printer is now online.
Setting hotend temperature to 115.000000 degrees Celsius.
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed (WindowsError(22, 'El dispositivo no reconoce el comando.'))