MarlinFirmware/Marlin #19201: [BUG] MINTEMP triggered with low temp incorrectly

gabrieled07, 2020-08-31:
### Bug Description

Error: MINTEMP triggered, system stopped! Heater_ID: bed - Printer halted. kill() called! 
when reading: 
Recv:  T:11.72 /0.00 B:13.30 /90.00 @:0 B@:0
and:
'#define BED_MINTEMP 5

temp values are true, all hardware works fine.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5148257/Configuration.zip)

### Steps to Reproduce

1. [First Step]
 Use printer in cold environment
2. [Second Step]
 Try to heat bed
3. [and so on...]
 Error triggered
4. Heat slightly above 10°c and it works

**Expected behavior:** [What you expect to happen]
Marlin to trigger this error when temperature actually is equal or below set limit

**Actual behavior:** [What actually happens]
Marlin triggers error when temp is low, but above set limit.

#### Additional Information

I suspect bed_hysteresis, temp might be on the limit.
Ambient temp was checked, and the software reading was close enough (.5 ° difference)
Not life or death issue, but I control the printer remotely from my office and when its very cold I cannot start it (i just go and heat the bed with something and its all) but, it might be a bug.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-08-31:
I'm using current bugfix and your configs on a ramps 1.4 but using variable resistors instead of thermistors, setting bed and hotends to 10C, no issues, 9C, no issues, 8C.. no issues
Cannot replicate this issue.  
Can only suspect a hardware issue, if your thermistors get momentarily disconnected (ie a broken wire) this would cause this issue.

gabrieled07, 2020-08-31:
Hi, 
Thank you for your fast reply
I will try to replicate the issue with your method using a pot.
It is not a momentary disconnect, but knowing the software is not an issue, it might be electrical noise triggering the failure mode.
I will try to confirm this as soon as possible and update the ticket.
Thank you again

boelle, 2020-08-31:
> It is not a momentary disconnec

check if the wire is broken under the plastic, that would count as a disconnect too, 

you know many bends will break metal in the end but not the plastic

sjasonsmith, 2020-08-31:
> It is not a momentary disconnect

I have certainly had thermistors with broken wiring in the middle of the run. They would only fail when the bed was in a particular position. In that case I suggest replacing the entire wire. Once it is broken in one place it is probably nearly broken in other places as well.

If that is your issue, you might be able to reproduce by just watching your LCD as you slowly move the bed by hand. If you see it flash to MINTEMP (usually about -15, I think), then you have bad wiring for sure.

gabrieled07, 2020-09-09:
Hi everyone, a little update.
I've tried simulating the thermistor with a variable resistor and the issue was fully repeatable, but it gave me another data, the cutoff temp wasn't constant, so it seemed to confirm a non bug thing

I've checked without too much disassembly the ramps board but there was no obvious cause

Then, I went to the noise side, but the issue was still present without significant power being drawn

A while ago I've modified the ramps board to control an atx psu (ps_on) with the logic powered from the standby line, so I could have everything 'powered off' and with remote commands I could start it remotely, very lazy
I don't have a scope to confirm here, but I believe in some point the ripple on that line was too high, causing the already noisy atmega adc to be much worst, and in situations when you are near limits this behavior occurred, 
What I did to indirectly prove this is to power the logic from the USB (octoprint server) and with about 50hs of prints starting early in the morning with near limit temps this problem never appeared again. 
Next I will have to replace or repair the PSU, but it was not a bug, although at the time I was almost sure about it.
Thank you for testing and taking the time to look for my issue.
You guys do an amazing job with Marlin

closed 2020-10-16 by github-actions[bot]