MarlinFirmware/Marlin #19204: [BUG] Color UI Blank Screen when trying to Move Z Manually

ilakuli, 2020-08-31:
### Bug Description

I compiled Version 2.0.6.1 for my Sapphire Plus (MKS Robin Nano V1.2 mit TFT35) using the new Color UI. Everything works except when I navigate to Motion --> Move Axis --> Move Z it creates a blank Screen. I then need to power cycle the printer to get out. The Move Menus for X, Y and E work just as expected.

### My Configurations

[Marlin-Sapphire_Plus_Stable.zip](https://github.com/MarlinFirmware/Marlin/files/5149036/Marlin-Sapphire_Plus_Stable.zip)

andreibobirica, 2020-09-01:
I tried to replicate the bug with my Longer3d broad and color ui but from me does not make any problem moving the Z.

The only problem there is, is the z babystepping, in addition to the options 10,1,0.1 mm the smaller one of babystepping does not appear the number.

Miraculix200, 2020-09-01:
Do you maybe have problems with not enough free memory? What amount of flash/memory space does it show after compiling?

ilakuli, 2020-09-01:
@andreibobirica ok I'll try to disable baby stepping since I'm not really using the manual Mesh bed levelling. I just left it enabled in case I need it.

@Miraculix200 I don't think it's a memory problem. The bugfix version I compiled with the ColorUI worked normal. I just wanted to switch back to the stable build since they added the ColorUI there. The settings however are the same in my Bugfix and my Stable Firmware.

ilakuli, 2020-09-01:
Ok so weird stuff is happening. Disabling babystepping did not fix the problem.
So I just wanted to reflash my Bugfix Build (which worked normally and has not been changed, just recompiled) and now this has the same problem...

Miraculix200, 2020-09-01:
As you can see here I had a similar problem: https://github.com/MarlinFirmware/Marlin/issues/18967

ilakuli, 2020-09-02:
Ok what did you change exactly to solve your issue (Which file and what lines)? Maybe it will fix mine, too. Thanks

Miraculix200, 2020-09-02:
It'll probably not fix your problem, because you didn't increase buffer sizes. I reduced mine, which were very high:
https://github.com/Miraculix200/Marlin-SapphirePro-2209-UART/blob/b2fca4a92690f5bed312686039d1d977ab7744d8/Marlin/Configuration_adv.h#L1862

ilakuli, 2020-09-02:
Yeah mine are actually set exactly the same as in your config file. Im kinda out of ideas. I used the custom user menus in marlin to implement 4 buttons for Z up and down 10 or 1mm to work around it. I'm gonna try to download a fresh copy of marlin 2.0.6.1 and configure it freshly. Thanks for your help, much appreciated.

ilakuli, 2020-09-02:
Actually you were exactly right! I compared my Config to the Base Marlin Configuration and my buffers were really big. I must have done this while troubleshooting another issue (Screen stayed completely off after flashing). Turning back BLOCK_BUFFER_SIZE from 64 to 16 and commenting out RX_BUFFER_SIZE did the trick. Thanks a lot for helping me out, much appreciated!

closed 2020-09-02 by ilakuli

andreibobirica, 2020-09-04:
Yesterday I encountered the same problem, I need to enlarge the buffer to try to solve another problem.
I set it to 64 but with the COLOR UI it doesn't work, even though 10% of memory remains free.

The other problem I encountered concerns octoprint printing, by chance the machine stops, the screen is blocked, and the machine heats up a lot.
I absolutely have to solve this problem.
I risk a fire from Marlin any moment.
I will do an ISSUE for my problem, but for now I would like to understand if with the COLOR UI there is a way to use 64 of buffer size

Miraculix200, 2020-09-04:
> I will do an ISSUE for my problem, but for now I would like to understand if with the COLOR UI there is a way to use 64 of buffer size

As you can see here I do use 64 buffer size: https://github.com/Miraculix200/Marlin-SapphirePro-2209-UART/blob/b2fca4a92690f5bed312686039d1d977ab7744d8/Marlin/Configuration_adv.h#L1862

You may have to reduce memory usage another way to make it work. E.g. by disabling EEPROM features.

andreibobirica, 2020-09-07:
I won't to disable the EEPROM .
It is strange, compiling it it say 90% flash, 87% rom, something like that, in any case lower than the maximum limit.
I don't know why it needs more memory.
In the first config i has a very big Buffer for the TFT, now i have 3200 and it works too (screen_width x 10 i think).
Maybe now it works with a BLOCK_BUFFER_SIZE of 64.

I need you to ask something.
I see that you have BLOCK_BUFFER_SIZE 64, BUFSIZE 64 and TX_BUFFER_SIZE 0 , RX_BUFFER_SIZE 0

I have activeted advanced_ok too and in respons receving Marlin sends ok N4997 P0 B31
i  have a bufsize of 32, if bigger can improve something?
It happens that the P number, planar moves i think, it increase sometimes.
With a Bigger Bufsize like your may i have less problem?

You have also a slowdone factor of 32.
#define SLOWDOWN
#if ENABLED(SLOWDOWN)
  #define SLOWDOWN_DIVISOR 32
//  #define SLOWDOWN_DIVISOR 2
#endif

Can you teach me about your configuration?

I have a bad problem with octoprint, and i try to solve that.
With bigger GCODE and Longer one, the firmware blocks and disconect .
The display is blocked and the printer is unusable.
Only with Octoprint.

AnHardt, 2020-09-07:
> I don't know why it needs more memory.

The processor needs some stack, on the RAM, to run.

`BLOCK_BUFFER_SIZE` above 32 are mostly useless but decrease the available RAM.
That is usually better invested in some larger RX/TX buffers.
SLOWDOWN is mostly useless when ADAPTIV_STEP_SMOTHING is active. All the saved time from SLOWNOWN is happily consumed by ADAPTIV_STEP_SMOTHING.
With a  `BLOCK_BUFFER_SIZE 64` and a `SLOWDOWN_DIVISOR 32` SLOWDOWN is almost shut off.
`BUFSIZE` above 4 is almost useless when using ping-pong-ok-protocol like from octoprint. But it occupies `(BUFSIZE * MAX_CMD_SIZE)` Bytes of RAM.

andreibobirica, 2020-09-07:
Console LOG
https://pastebin.com/vR32JUAG
Video Demostration of the problem
https://streamable.com/06bh9t

I have this annoying problem that I can't solve and I guess it's due to a bad Buffer configuration.

With printing from SD it does not happen.
The printer apparently hangs at a random Layer when printing from Octoprint.

I'm almost sure it's not the fault of the graphics changes I made to the firmware, I still had this problem from version 2.0.5.3 stock, it happens on multiple raspberry and multiple printers.
it's the firmware's fault.

I currently have set
BLOCK_BUFFER_SIZE 64
BUFSIZE 32
SLOWDOWN_DIVISOR 2
TX_BUFFER_SIZE 0

i had the same problem with
BLOCK_BUFFER_SIZE 32
BUFSIZE 8

I don't really know where to go to solve the problem.
it is true that I made some profound changes to the Marlin code, to add features that were not existing.
But these are independent:
This is the comparison with the latest Bugfix:
https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...andreibobirica:worktest

What do you recommend me to try to test?
What should I do?
what BUFFER configurations should I set to try to solve the problem.

Of course it happens only and only with Octoprint, printing from SD has no problem.
And only with large and very durable SD cards.

rhapsodyv, 2020-09-07:
You can try `#define TFT_BUFFER_SIZE` to smaller sizes to. Try something like 2400, or even 1200.

andreibobirica, 2020-09-07:
> You can try `#define TFT_BUFFER_SIZE` to smaller sizes to. Try something like 2400, or even 1200.

thanks, i will try to save space by lowering the TFT buffer to save space with COLOR_UI and maybe ABL active
it was stock 14400 like your chitu board then i set to 3200 like TPRUVOT says, i will lower it more.

The problem is that this problem occurs indistinctly both on the CLASSIC UI and on the COLOR_UI, it happens both with RAM at 90% and with RAM at 40%, both with FLASH at 80% and with FLASH at 30%

Let's say it is not dependent on available memory.
I don't know where to look to fix the problem.

Miraculix200, 2020-09-07:
You can try installing the Arc Welder plugin for Octoprint. It transforms G0/G1 moves into G2/G3 moves, which can significantly reduce file size on prints with many arcs/circles.

andreibobirica, 2020-09-08:
> You can try installing the Arc Welder plugin for Octoprint. It transforms G0/G1 moves into G2/G3 moves, which can significantly reduce file size on prints with many arcs/circles.

I just installed it, reduced a file from 30MB to 6MB, I set #define MM_PER_ARC_SEGMENT 0.1
and all other stock and standard settings.
I believe it can help significantly.

I have read that it is not fully optimized and printing defects may appear at times however, plus it does not get along so well with the ABL and MBL.
I will try it fully, to see if the problem goes away, but I still don't think it is possible to call it a solution to my problem.
The problem is the firmware, not the GCODE, being big is its only fault hahahaha

I'm trying other firmware, other branches, and for now I just know that it's not just my firmware that has that annoying problem.
Some TPRUVOT configurations have it too, maybe is an Hardware problem of Longer Board or of STM32F1 microcontroller...
Now ALL my branches and version have this problem, color ui, classic ui... big and lower Buffer, alls.
Many of the branchs of TPRUVOT have this problem, for example even his ultimate dev-colorui bugfix-updated branch have this problem (but only with overclock at 128 Hz), i did not the overclock on my versions, but problem still appeas on stock 72Hz

I will do more in-depth analysis and open an ISSUE ticket for my problem.

andreibobirica, 2020-09-08:
> You can try installing the Arc Welder plugin for Octoprint. It transforms G0/G1 moves into G2/G3 moves, which can significantly reduce file size on prints with many arcs/circles.

Nothing, i tried with the Ark:

``[...]
Recv: ok N82603 P0 B31
Send: N82604 M204 S500*107
Recv: ok N82604 P0 B31
Send: N82605 G1 X100.788 Y122.446 F7800.000*86
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
```

It still block in an random layer, the printer is unusable and requires restart

rhapsodyv, 2020-09-08:
@andreibobirica can you do a test with this PR: https://github.com/MarlinFirmware/Marlin/pull/19301 

It fix a few problems of serial communication, that can freeze some machines while printing and using serial.

andreibobirica, 2020-09-09:
> @andreibobirica can you do a test with this PR: #19301
> 
> It fix a few problems of serial communication, that can freeze some machines while printing and using serial.

I'll do a test.
The test I do to see if the printer freezes is this:
https://file.io/OaABYNb9isN4
A 10 hour Gcode (no filament), sometimes freezes at 20%, sometimes 80%, it depends.
I can only do 2/3 tests in a day.

I have tried various firmware for my Longer3d.

By TPRUVOT
https://github.com/tpruvot/Marlin/tree/master
72 Mhz It seems to have no problems
128 Mhz Freezes
https://github.com/tpruvot/Marlin/tree/dev
https://github.com/tpruvot/Marlin/tree/dev-colorui
They get stuck

MINE
https://github.com/andreibobirica/Marlin
72 Mhz Freezes
Both Classic UI and Color UI

I have changed various Buffsize parameters, etc .... but I think it is a problem due to the Serial communication.

The interesting thing is that I noticed that with TPRUVOT's 128 Mhz Overcled BUILD it freezes after 20 minutes, it freezes immediately.
While with mine, the officers, or yours at 72 Mhz it stops even after hours.

Currently the branch https://github.com/andreibobirica/Marlin/tree/worktest
is the one I merged with your changes you listed.
I'll do some tests.
I had already updated everything to the latest Bugfix, in fact I noticed the Emergency Stop support for the STM32F1 and I'm having a lot of fun with the M108 command finally

rhapsodyv, 2020-09-09:
> The test I do to see if the printer freezes is this:
> https://file.io/OaABYNb9isN4

I get "File not found"

andreibobirica, 2020-09-09:
Here are the Test:
https://drive.google.com/file/d/15Ffk4LN898Y7TyYJKCW2kY8EwhU_hdRZ/view?usp=sharing

I merged the pull request that you recommended.
I did a test with the GCODE that I attached to you.
You have to take into account that I never managed to terminal that GCODE (with my firmware), it always stopped, either at 20%, or at 50%, sometimes at 80%.

This time for the first time, I managed to carry it to the end.
It seems miraculous.

Of course I will do other tests, maybe it is a coincidence, but it seems that there are improvements.
Thanks for the work

rhapsodyv, 2020-09-09:
> Here are the Test:
> https://drive.google.com/file/d/15Ffk4LN898Y7TyYJKCW2kY8EwhU_hdRZ/view?usp=sharing

Thanks!

> I merged the pull request that you recommended.
> I did a test with the GCODE that I attached to you.
> You have to take into account that I never managed to terminal that GCODE (with my firmware), it [...]

Good to know! Fantastic!

> 
> Of course I will do other tests, maybe it is a coincidence, but it seems that there are improvements.

Just to give you a little explanation of what @sjasonsmith and I did in that PR.

The current Marlin code (without that PR), don't handle any serial error. For example: if there's any serial communication error, the old code simply ignore it, giving a lot of unexpected behaviour in that situations. 

In that PR, we added some validations and checks for common serial errors, like overrun, parity and framing errors. So, we expect that the serial became more stable than before.

Yes, it could be coincidence that your test was successful. But even if it was, it give us more and more confidence about the stability of that PR. 

> Thanks for the work

Thanks for your test!! :beers:

andreibobirica, 2020-09-10:
> Thanks for your test!! 🍻

I did further tests, another 20 hours of printing.
There are no more problems, I saw that the changes have also been introduced in the BUGFIX, there are no more problems now with the Serial communication.
Octoprint printing works very well and the printer is no longer stuck.

It seems really strange to me that before Marlin did not take into account communication errors, but it is good that he does now, it is a relief that this problem has been solved, I had tried in every way to ask for help, even with POST on FB on the Marlin's official page, but the fact that no one else had this problem made me feel crazy.

It probably happens much more often on our STM32F1 processor, I have no idea.

Now whatever firmware version i try works very well.

The only thing that appears to be unsupported is TPRUVOT's overclocked firmware.
I tried to print via 128 Mhz octoprint but it freezes after a few minutes, to tell the truth even the graphic interface (CLASSIC_UI) is not optimized for an overclocked firmware because all the action clocks are accelerated, as well as an effective sliding effect everything is more accelerated, EDIT screens, and modification of values.
if you go from 72Mhz to 128Mhz, double the actions are performed at the same time, with disastrous effects on the UI.

In any case, this function of TPRUVOT will have to remain an experiment, Marlin already works very well at 72 Mhz and everything is fine like that.

In any case, I repeat, it seems that the problem is solved, Nice work.

rhapsodyv, 2020-09-10:
> I did further tests, another 20 hours of printing.
> There are no more problems, I saw that the changes have also been introduced in the BUGFIX, there are no more problems now with the Serial communication.
> Octoprint printing works very well and the printer is no longer stuck.
> 

Fantastic!!! Thanks for the test!! 

> It seems really strange to me that before Marlin did not take into account communication errors, but it is good that he does now, it is a relief that this problem has been solved, I had tried in [...]

Marlin was relying on the maple framework. The problem is (as we just discovered), the maple framework wasn't handling errors conditions. 

With the development of emergency parser for STM32F1, now Marlin handles UART directly. This bring us the possibility to do a more robust IRQ handler. 

Thanks again!