MarlinFirmware/Marlin #19209: [FR] Change GRID_MAX_POINTS_X from configuration bed leveling menu

jsimmonstx, 2020-08-31:
Description: 

Instead of having to recompile the firmware to specify a different value for GRID_MAX_POINTS_X, provide a configuration menu item to let us set it there.

AnHardt, 2020-08-31:
That's difficult.
Currently Marlin is compiled with that number fixed. Making that changeable on the fly requests for the use of dynamic memory (malloc(), free()). Especially free() is not wanted because of fragmenting the heap. Alternatively one could reserve a maximum number for GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y and use only a part of that. How large this maximum numbers can be without making Marlin to large for the targeted board is depending on the other constant and the rest of the configured options - so is not known before Marlin has been compiled.

jsimmonstx, 2020-08-31:
So, when the printer is turned on, it creates an array on nxn elements on the heap, where n is the value assigned to the compiler definition?

I suppose you make a valid point, re: fragmented memory, but I wasn't aware that memory was so tight on 32-bit boards.

AnHardt, 2020-08-31:
The STM32F103 with only 256KB Flash seem to be already limited. You can't simply configure 'all in' at once.  I'm currently not aware of RAM limitations with the provided configurations - but with really large buffers in RAM you can break them (Try GRID_MAX_POINTS_X 100).
All kinds off AVRs have been brought to all kinds of their limits (flash, RAM, EEPROM, speed) by me - some only purposely.

> it creates an array on nxn elements on the heap

Currently in the area for global and static variables. On top of that is the heap. Stack comes from top and grows down in direction to the heap (AVR). If made dynamic it would be reside on the heap.

mymanjgodfrey, 2021-01-19:
Would be a very handy feature to have.

thisiskeithb, 2021-11-02:
Duplicate of #10547

closed 2021-11-02 by thisiskeithb