MarlinFirmware/Marlin #19212: [BUG] Z-Offset readings off with BLTouch which can change between reboots

GUICenter, 2020-08-31:
### Bug Description
The origination of this is strange and I do not know if this is firmware or hardware. Installed a new mount for Ender 3 with a single 5015 fan and single duct (Hero Me v5) but decided to switch to single 5015 with dual ducts. I printed 3 or 4 prints including the new duct and then installed it. Prior I had no issues with z-offset reading wrong and the BLTouch. Since then, I have only been able to get it to read correctly a couple of times and can't replicate. I thought it might be interference but cannot isolate after isolating each of the new parts added (fans, wire joins, etc.).

The Z-offset is off by a specific amount but that amount can change between reboot. I have measured the offset numerous times and it should be -3.124 at the current hight, but when it's set to that and run G29 the readings are significantly lower. I have had it happen a couple of times when the readings are opposite and much higher (during G29 for a print), but I don't have log for that only an [image of the mesh](https://i.imgur.com/7pYRyBf.png). The initial Bilinear Leveling Grid is closer to how it should read but that was forcing z-offset to -2.48 based on the corrected z-offset post G29. However, that is not true offset and not a fix permanently.

This is with both 2.0.6 and now with 2.0.6.1

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5151283/Config.zip)
ZIP file containing `Configuration.h`, `Configuration_adv.h`, and output log files.

### Steps to Reproduce

1. send the G28 command 
2. send the G29 command 
3. reboot or start print (sometimes)
4. repeat 1 and 2

**Expected behavior:** z-offset of -3.124 
**Actual behavior:** z-offset is incorrect and various (mostly lower) offsets are being read by G29

#### Additional Information

GUICenter, 2020-09-09:
Thought this might be EMI so I replaced the BLtouch cables to a single twisted cable which also eliminated the extension connections. I also added ferrites to multiple cables. Initially I thought it had fixed the issue of offset grid readings changing as I got consistent, but still incorrect offset, readings between reboots. However, when I went to print and had G29 in the start gcode of the print, the grid changed and was not consistent with the multiple previous tests.

non-print G29:
Bilinear Leveling Grid:
     0      1      2      3      4
0 -0.023 -0.067 -0.082 -0.062 +0.027
1 -0.241 -0.243 -0.251 -0.248 -0.184
2 -0.305 -0.308 -0.323 -0.323 -0.258
3 -0.204 -0.201 -0.233 -0.248 -0.196
4 +0.161 +0.104 +0.059 +0.029 +0.059
G29 uncorrected Z:13.12
corrected Z:13.06

Print gcode G29:
Bilinear Leveling Grid:
    0      1      2      3      4
0 -0.077 -0.092 -0.104 -0.104 -0.047
1 -0.243 -0.228 -0.226 -0.246 -0.199
2 -0.268 -0.258 -0.275 -0.293 -0.253
3 -0.174 -0.157 -0.196 -0.211 -0.184
4 +0.146 +0.119 +0.081 +0.034 +0.037
G29 uncorrected Z:13.12
corrected Z:13.09

Z-offset is 3.12 for both, however, looking at the code the top grid above starts off with this:
>>> G29  X0.00 Y-5.00 Z0.00
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X-43.00 Y-18.00 Z-3.12 (Left-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)

and the bottom is this:
>>> G29  X0.00 Y-5.00 Z0.02
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X-43.00 Y-18.00 Z-3.12 (Left-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR (enabled)
ABL Adjustment Z-0.023

Why does the top grid has (disabled) next to Auto Bed Leveling: BILINEAR? Would that be why I am seeing variations?

[Log file for data above](https://github.com/MarlinFirmware/Marlin/files/5194985/serial.1.log)

boelle, 2020-09-10:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

GUICenter, 2020-09-20:
Sorry, will test bugfix this week. Also have a new board coming, so can do before/after test as well.

GUICenter, 2020-10-07:
Finally got around to testing the bugfix version. Same issues happening.

I first ran tests at non-heated bed and as you can see in this file, the first 5 test have a corrected Z ranging from 3.34 to 3.40 with uncorrected at 3.12, which is the offset I set. On the last test you can see it changes to a corrected z of 2.99 with same uncorrected. 
[Bugfix non-heated bed](https://github.com/MarlinFirmware/Marlin/files/5337753/serial.3.log)

Example:
3.34
Bilinear Leveling Grid:
     0      1      2      3      4
0 -0.352 -0.369 -0.379 -0.352 -0.254
1 -0.557 -0.544 -0.529 -0.524 -0.449
2 -0.594 -0.592 -0.597 -0.602 -0.527
3 -0.502 -0.494 -0.514 -0.527 -0.474
4 -0.157 -0.192 -0.222 -0.254 -0.217

2.99:
Bilinear Leveling Grid:
     0      1      2      3      4
0 -0.012 -0.032 -0.037 -0.007 +0.091
 1 -0.214 -0.202 -0.184 -0.177 -0.102
2 -0.249 -0.247 -0.249 -0.249 -0.179
3 -0.157 -0.147 -0.169 -0.179 -0.127
 4 +0.191 +0.153 +0.123 +0.093 +0.131

After a few days I retested with a heated bed and got similar results:
[Bugfix heated bed](https://github.com/MarlinFirmware/Marlin/files/5337754/serial.4.log)

2.98 (corrected) and 3.12 uncorrected:
Bilinear Leveling Grid:
    0      1      2      3      4
0 -0.002 +0.008 +0.023 +0.048 +0.126
1 -0.169 -0.137 -0.114 -0.117 -0.062
2 -0.224 -0.197 -0.194 -0.209 -0.149
3 -0.119 -0.099 -0.117 -0.144 -0.102
4 +0.186 +0.178 +0.153 +0.121 +0.146

The next test came to:
3.34 corrected:
Bilinear Leveling Grid:
     0      1      2      3      4
0 -0.432 -0.427 -0.417 -0.399 -0.329
1 -0.577 -0.552 -0.534 -0.537 -0.487
2 -0.584 -0.572 -0.572 -0.582 -0.527
3 -0.482 -0.452 -0.487 -0.492 -0.454
4 -0.152 -0.172 -0.202 -0.234 -0.212

This is still on the original board as I have not had time to replace it yet, but as you can see, there is some strange variability going on which seem to be constant now compared to the original posting, but still apparent regardless of heated or unheated bed.

LarrxX, 2020-10-11:
Is this the same as [issue #18769](https://github.com/MarlinFirmware/Marlin/issues/18769)?

schleir, 2020-10-11:
LarrxX, sounds similar to me.

closed 2020-11-18 by github-actions[bot]