MarlinFirmware/Marlin #19213: [BUG] (short description)

katherinehackworth, 2020-08-31:
### Bug Description
- Ender 3 w/ SKR Mini E3 v1.2 and BLTouch
- Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is enabled
- AUTO_BED_LEVELING_BILINEAR is enabled
- Z_SAFE_HOMING is also enabled

- All settings unchanged from Marlin 2.0.5.3 copied over from working config into 2.0.6.1. 2.0.5.3 homes to (117, 117), as expected, but 2.0.6.1 homes to (58, 117).

Already checked for errors in bed size definition, probing settings, probe offset, etc. Unsure if issue is related to change from 2.0.5.3 -> 2.0.6.1 of Z_SAFE_HOMING_N_POINT from (N_BED_SIZE) / 2) to N_CENTER. Manually setting these coordinates with a constant (in this case, 117), still results in the unexpected behavior.

### My Configurations
[Available here](https://gist.github.com/katherinehackworth/d4c34cf8b5b0a7fc380725217bead139)

### Steps to Reproduce
1. Use the display to select Auto Home.

**Expected behavior:**
X and Y are homed normally, then the probe moves to the center of the bed and homes the Z axis.

**Actual behavior:**
X and Y are homed normally, then the probe moves to the center of the Y axis (Y_MAX / 2) but not the center of the X axis (roughly X_MAX / 4), and homes the Z axis there.

thisiskeithb, 2020-08-31:
Did you Initialize EEPROM through the menu or using `M502` followed by `M500` after flashing?

katherinehackworth, 2020-08-31:
Yep

katherinehackworth, 2020-10-01:
The problem persisted on 2.0.6.x versions, but appears to be fixed on 2.0.7. Not sure what changed because I don't see anything in the changelogs related, but it's fixed nonetheless.

closed 2020-10-01 by katherinehackworth