MarlinFirmware/Marlin #19214: [FR] smart filament jam detector

DrumClock, 2020-08-31:
Hi,
I've been testing a smart filament jam detector for over half a year.
It is on the Arduino NANO board as an external component.
Would it be possible to use the principle and integrate it into FW Marlin?

**Video:** https://www.youtube.com/watch?v=kDGh5_Zjrdg

**Principle description:**
- A counter with a default value of eg 20 is defined.
- The extruder encoder decreases the counter value by one regardless of the direction of rotation (20 pulses per revolution).
   Here you could use the driver information of all extruders E-STEP.
- Filament encoder (1 or more) resets the counter to the default state.

If the counter status is "0", a filament jam is detected or has run out. It also works for grits and short retractions.

Thank you so much for your reply.

**Foto:**

[image]

[image]

thisiskeithb, 2020-08-31:
Marlin supports encoder-based filament runout sensors, but I’m not sure if it supports *yours*.

Enable `FILAMENT_MOTION_SENSOR`/look at the code surrounding this feature.

DrumClock, 2020-08-31:
I believe that the functions `FILAMENT_RUNOUT_DISTANCE_MM` and `FILAMENT_MOTION_SENSOR` 
work on the **principle of measuring** the length of the unwound filament.

**My suggestion is to compare unwound and extruded filament - more accurate and most importantly more reliable.**

In Marlin, this feature is definitely not, so I'm asking for the option to add.

thisiskeithb, 2020-08-31:
> I believe that the functions FILAMENT_RUNOUT_DISTANCE_MM and FILAMENT_MOTION_SENSOR
work on the principle of measuring the length of the unwound filament.

`FILAMENT_MOTION_SENSOR` doesn't measure the filament length, it checks for motion and toggles the runout pin as the filament moves.

See [runout.cpp](https://github.com/MarlinFirmware/Marlin/blob/8d76d8c0810dc14ce1180ff1d7e9d6169199de41/Marlin/src/feature/runout.cpp) & [runout.h](https://github.com/MarlinFirmware/Marlin/blob/8d76d8c0810dc14ce1180ff1d7e9d6169199de41/Marlin/src/feature/runout.h) for more details.

DrumClock, 2020-08-31:
@thisiskeithb 
I'm sorry, I'm not a programmer and I don't know much about the code, or I eat very little.

DrumClock, 2020-09-09:
**I would have an idea about this setting by the user ....**

```
#define SMART_JAM_DETECTOR
  #if ENABLED(SMART_JAM_DETECTOR)
    #define ENC_FIL1_PIN         64 // Pin to detect encoder pulses for filament 1
    #define ENC_STEP1            20 // number of encoder pulses per revolution for filament 1
    #define ENC_FIL2_PIN         65 // Pin to detect encoder pulses for filament 2
    #define ENC_STEP2            20 // number of encoder pulses per revolution for filament 2
    #define E_STEP               20 // number of extruder pulses per revolution (for all extruders)
    #define COUNTER_STEP         14 //default value of the pulse counter
  #endif
```

DrumClock, 2020-09-28:
Hi,
**Is anyone solving this?**
This GitHub system is a bit strange.
I don't know if anyone is solving it or not or if it is already solved.
**There is simply no feedback!**

thisiskeithb, 2020-09-28:
Keep in mind that all the code here is contributed by unpaid volunteers who work on it in their free time.

I've added a Help Wanted label to hopefully attract attention of someone who has the hardware, interest, and ability to fix/implement the change.

DrumClock, 2021-03-07:
Hi
I have been testing this system of mine for a year now and it is completely reliable.
Just integrate it into Marlin.

yet-another-average-joe, 2021-03-15:
Why do you want to integrate code into Marlin ?

As I understand this sensor, a Nano interrupts on pulses from two optos hooked to pins 2 and 3, counts them, substracts the result, and triggers an alarm if the result is too far from zero. With some sort of integration over time. Pretty simple.

You hooked a free Nano output to a free motherboard input and set this input as the runout sensor input. Fine.

What is the rotary encoder for ? Sensivity I presume. Good.

Such a device can live 100% standalone (and has to be IMHO). Can be adjusted on the fly, without menus and submenus, and will not act as a "pin hog". Worse : not all pins are interruptable on all motherboards, meaning that it could not work with every board (depending on the interruptable pins used by Marlin) (I doubt  you are polling the optos...)

Replace the large TM1637 display with a tiny 0.91" or 0.96" OLED, replace the huge rotary encoder with a tiny decimal or BCD encoder (or even a DIP-switch block), print a small enclosure, and you get a really nice device. A few wires to the optos, 3 wires to the printer controller, done !

There's nothing related to your project on your Git, so I could be missing something...

DrumClock, 2021-03-15:
Hi @yet-another-average-joe Thank you very much for your response 

> Why do you want to integrate code into Marlin ?

Because I wanted to disable the encoder on the extruder motor and use the information about steps E0, E1, E2
 .... directly from FW  Marlin 

> As I understand this sensor, a Nano interrupts on pulses from two optos hooked to pins 2 and 3, counts them, substracts the result, and triggers an alarm if the result is too far from zero. With some sort of integration over time. Pretty simple.

No, the value stored in NANO is eg "30"
The extruder encoder subtracts 1 from this value with each pulse.
Encoders on the filament reset the value to the original state "30"

If the filament jams or runs out, the value "30" is gradually reduced by the encoder on the extruder to "0".
This indicates a fault and starts the FILAMENT_RUNOUT_SENSOR function in Marlin. 

> Replace the large TM1637 display with a tiny 0.91" or 0.96" OLED, replace the huge rotary encoder with a tiny decimal or BCD encoder (or even a DIP-switch block, or even a pot on an analog input). Print a small enclosure, and you get a really nice device. A few wires to the optos, 3 wires to the printer controller, done !

Yes I can replace the HW, but I can't program it.
SW was made by a programmer for me for a lot of money. 

> There's nothing related to your project on your Git, so I could be missing something...

Yes, I didn't put anything on my Git ..... if there is interest I can create something

yet-another-average-joe, 2021-03-15:
I undersstand better now !

You need Marlin sending a pulse to an output pin when it sends a STEP or microstep command to the motor driver. DIR signal (for extrusion or retraction) is not needed : if it moves, it's OK. This seems easy to do... but needs a in depth knowledge of Marlin sources : probably just a few lines of code to add to stepper.cpp (but where ?) and in the pins definitions. 

It would also allow the detection of the extruder missing steps, or a clogged hotend, if you keep the encoder ! But it is not doable with a Nano (2 interrupts only). A bluePill (cheaper !) would be perfect.

This is an interesting project. It's been around for a while, with mouse wheel encoders. It seems to me that SmoothieWare and even Klipper can do it (?). Why not Marlin ?

There's an easy hardware workaround : getting the signals directly from the drivers pins (assuming the extruder drivers are not soldered to the board, and that they do not use a serial protocol. Just some perfboard + single row PC104 pin sockets (long pins...) = a beakout board for the STEP and DIR signals, the drivers being inserted on top. Then, the external microcontroller can do the job. At least you could start whith the experimentations. Plus a 5$ Saleae knockoff logic analyzer in order to see what happens.

Seriously, you payed a lot ? 99% of the Arduino code can be done with C++ tutorials, a few well known and free libraries, and some help from the Arduino forums. And it is an excellent project for someone who wants to learn coding.

DrumClock, 2021-03-15:
> Seriously, you payed a lot ? 99% of the Arduino code can be done with C++ tutorials, a few well known and free libraries, and some help from the Arduino forums. And it is an excellent project for someone who wants to learn coding.

Hi @yet-another-average-joe 
I tried to learn programming, but I can't. I paid about 100 USD for that SW (that's a lot for me).
I have no problem with electronics or mechanics but programming it hinders me terribly in the development of a 3D printer ....

yet-another-average-joe, 2021-03-15:
100$ is a lot for anybody, but a professionnal have to eat and pay taxes... After all, it's probably not that much. 

No problems with electronics ? Great ! Why wouldn't you try to "read" the motor wires themselves, directly ? I never hooked the scope to them ; they should be squares or sines (more or less depending on microstepping). A filtered, differential input circuit followed with a Schmitt trigger -> a square signal you can "count" with the Arduino. Would be 100% independant from the hardware and firmware, and would work with any firmware/printer.

You should probably find help from other makers on RepRap forums. Did you try ?
Such projects exist, or existed in the past.

DrumClock, 2021-03-15:
Hi, yes everything can be solved externally and independently of the FW of the 3D printer. 
I've tried it for over a year and it's perfect.
I also thought of using the STEP pins of the TMC driver and doing everything on Arduino ..... but why if it could go with Marlin?

yet-another-average-joe, 2021-03-15:
Because of DOTADIW.
Because doing this externally allows you to hook your sensor to ANY printer, from the past, from now and in the future.
Because it is firmware independant.
Because it needs no additional work.
Because it can also work with OctoPrint (RasPi GPIO) with no additionnal plugin (coding OctoPrint plugins is not easy).
Because it is simpler, and simpler is better.
Because there's no risks of bugs in Marlin.
Because you have full control over your device.
Because it will support any number of direct drive extruders (think of the E3D Tool Changer). Just add more sensors and some hardware logic (gates).

Etc...

I see only cons for firmware support other than a pin triggering M600, and only pros for a standalone device. Can't find any pro.

DrumClock, 2021-03-16:
Yes you're right
I will leave it as an external module with output for the function `FILAMENT_RUNOUT_SENSOR` 
However, it would help a lot to define a PIN in Marlin
s where the information pulses from all `E_STEP` will be.
This eliminates the need to complete taps with the driver for E motors. 

**Would anyone be able to do it?**

closed 2021-03-29 by DrumClock