MarlinFirmware/Marlin #19216: [BUG] compile failure in LCD DAC edit menu (board w/MCP4728 DAC)

fiveangle, 2020-09-01:
Printrboard RevF (AT90USB) compile failure:

```
In file included from Marlin/src/lcd/menu/menu_advanced.cpp:31:0:
Marlin/src/lcd/menu/menu_advanced.cpp: In function 'void menu_dac()':
Marlin/src/lcd/menu/../../inc/../core/language.h:319:15: error: expected unqualified-id before string constant
 #define MSG_X "X"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:48: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:66: note: in expansion of macro 'GET_TEXT'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_X'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:69:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(X);
     ^
Marlin/src/lcd/menu/../../inc/../core/language.h:320:15: error: expected unqualified-id before string constant
 #define MSG_Y "Y"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:48: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:66: note: in expansion of macro 'GET_TEXT'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_Y'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:70:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(Y);
     ^
Marlin/src/lcd/menu/../../inc/../core/language.h:321:15: error: expected unqualified-id before string constant
 #define MSG_Z "Z"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:48: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:66: note: in expansion of macro 'GET_TEXT'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_Z'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:71:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(Z);
     ^
Marlin/src/lcd/menu/../../inc/../core/language.h:322:15: error: expected unqualified-id before string constant
 #define MSG_E "E"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:48: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:66: note: in expansion of macro 'GET_TEXT'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_E'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:72:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(E);
     ^
*** [.pio/build/at90usb1286_dfu/src/src/lcd/menu/menu_advanced.cpp.o] Error 1
========================= [FAILED] Took 13.46 seconds =========================
```

### My Configurations

https://github.com/fiveangle/Marlin/tree/bf2-2020-08-31/Marlin
https://github.com/fiveangle/Marlin/blob/bf2-2020-08-31/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/bf2-2020-08-31/Marlin/Configuration_adv.h

ellensp, 2020-09-01:
Configs are version 2.0  we are currently at 2.0.6.1  please use a current Marlin

fiveangle, 2020-09-01:
> Configs are version 2.0 we are currently at 2.0.6.1 please use a current Marlin

Thanks for the oversight of a second set of eyes.  Blew caches away from IDE and refreshed d/l's from scratch and indeed there was an issue with my IDE.

Current compile failure appears as it may be related to same issue as previous cached version:

```
[avr-g++ -o .pio/build/at90usb1286_dfu/firmware.elf -fmerge-all-constants -Os -Wl,--gc-sections,--relax -mmcu=at90usb1286 .pio/build/at90usb1286_dfu/src/src/HAL/AVR/HAL.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/HAL_SPI.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/MarlinSerial.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/Servo.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/eeprom.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/fast_pwm.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/fastio.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/AVR/watchdog.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/HAL_spi_L6470.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/backtrace.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwarm.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwarm_arm.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwarm_thumb.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwarmbytab.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwarmmem.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwinder.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/backtrace/unwmemaccess.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/eeprom_api.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/eeprom_if_i2c.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/eeprom_if_spi.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/esp_wifi.cpp.o .pio/build/at90usb1286_dfu/src/src/HAL/shared/servo.cpp.o .pio/build/at90usb1286_dfu/src/src/MarlinCore.cpp.o .pio/build/at90usb1286_dfu/src/src/core/serial.cpp.o .pio/build/at90usb1286_dfu/src/src/core/utility.cpp.o .pio/build/at90usb1286_dfu/src/src/feature/babystep.cpp.o .pio/build/at90usb1286_dfu/src/src/feature/bedlevel/bedlevel.cpp.o .pio/build/at90usb1286_dfu/src/src/feature/bedlevel/mbl/mesh_bed_leveling.cpp.o .pio/build/at90usb1286_dfu/src/src/feature/caselight.cpp.o .pio/build/at90usb1286_dfu/src/src/feature/leds/neopixel.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/bedlevel/G42.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/bedlevel/M420.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/bedlevel/mbl/G29.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/bedlevel/mbl/M421.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/calibrate/G28.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M200-M205.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M220.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M221.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M301.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M304.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M575.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M672.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/config/M92.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M108_M112_M410.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M111.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M120_M121.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M17_M18_M84.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M211.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M226.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M280.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M350_M351.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M380_M381.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M400.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M42.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M80_M81.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M85.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M993_M994.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M997.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/M999.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/control/T.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/eeprom/M500-M504.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/feature/advance/M900.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/feature/digipot/M907-M910.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/feature/pause/G27.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/gcode.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/geometry/G92.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/host/M110.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/host/M113.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/host/M114.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/host/M115.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/host/M118.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/host/M119.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/lcd/M0_M1.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/lcd/M117.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/lcd/M145.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/lcd/M250.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/lcd/M300.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/motion/G0_G1.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/motion/G4.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/parser.cpp.o .pio/build/at90usb1286_dfu/src/src/gcode/queue.c [... cut ...]
.pio/build/at90usb1286_dfu/src/src/lcd/menu/menu_advanced.cpp.o: In function `menu_dac()':
menu_advanced.cpp:(.text._Z8menu_dacv+0x12): undefined reference to `dac_current_get_percent(AxisEnum)'
menu_advanced.cpp:(.text._Z8menu_dacv+0x226): undefined reference to `dac_commit_eeprom()'
collect2: error: ld returned 1 exit status
*** [.pio/build/at90usb1286_dfu/firmware.elf] Error 1
========================== [FAILED] Took 3.93 seconds ==========================]
```

Current configs pushed to above repo.

ellensp, 2020-09-01:
try adding #define DAC_STEPPER_CURRENT to your configuration.h

fiveangle, 2020-10-02:
> try adding #define DAC_STEPPER_CURRENT to your configuration.h

This is a red herring. DAC_STEPPER_CURRENT has always been set it the pins file for the Printrboard RevF and still is:

https://github.com/MarlinFirmware/Marlin/blob/348570fd683b56ebb930da9405aaf282c1165c0a/Marlin/src/pins/teensy2/pins_PRINTRBOARD_REVF.h#L145-L146

Currently as of https://github.com/MarlinFirmware/Marlin/commit/348570fd683b56ebb930da9405aaf282c1165c0a it's gone from the frying pan and into the fire with what looks like a problem with the Teensy2 library, as USBCON is typically set there:

```
Processing at90usb1286_dfu (platform: teensy; extends: env:at90usb1286_cdc; board: at90usb1286; board_build.f_cpu: 16000000L; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>, -<src/lcd/HD44780> -<src/lcd/TFTGLCD> -<src/lcd/dwin> -<src/lcd/dogm> -<src/lcd/tft>, -<src/lcd/menu>, -<src/lcd/menu/game/game.cpp> -<src/lcd/menu/game/brickout.cpp> -<src/lcd/menu/game/invaders.cpp>, -<src/lcd/menu/game/maze.cpp> -<src/lcd/menu/game/snake.cpp>, -<src/lcd/menu/menu_backlash.cpp>, -<src/lcd/menu/menu_bed_corners.cpp>, -<src/lcd/menu/menu_bed_leveling.cpp>, -<src/lcd/menu/menu_cancelobject.cpp>, -<src/lcd/menu/menu_delta_calibrate.cpp>, -<src/lcd/menu/menu_filament.cpp>, -<src/lcd/menu/menu_info.cpp>, -<src/lcd/menu/menu_job_recovery.cpp>, -<src/lcd/menu/menu_led.cpp>, -<src/lcd/menu/menu_media.cpp>, -<src/lcd/menu/menu_mmu2.cpp>, -<src/lcd/menu/menu_password.cpp>, -<src/lcd/menu/menu_power_monitor.cpp>, -<src/lcd/menu/menu_spindle_laser.cpp>, -<src/lcd/menu/menu_temperature.cpp>, -<src/lcd/menu/menu_tmc.cpp>, -<src/lcd/menu/menu_touch_screen.cpp>, -<src/lcd/menu/menu_ubl.cpp>, -<src/lcd/extui/lib/mks_ui>, -<src/lcd/extui/lib/dgus> -<src/lcd/extui/dgus_lcd.cpp>, -<src/lcd/extui/example.cpp>, -<src/lcd/extui/malyan_lcd.cpp>, -<src/lcd/extui/lib/ftdi_eve_touch_ui>, -<src/lcd/extui/anycubic_chiron_lcd.cpp>, -<src/lcd/extui/anycubic_i3mega_lcd.cpp> -<src/lcd/extui/lib/anycubic_i3mega>, -<src/lcd/lcdprint.cpp>, -<src/sd/usb_flashdrive>, -<src/feature/backlash.cpp>, -<src/feature/baricuda.cpp> -<src/gcode/feature/baricuda>, -<src/feature/bedlevel/abl> -<src/gcode/bedlevel/abl>, -<src/feature/bedlevel/mbl> -<src/gcode/bedlevel/mbl>, -<src/feature/bedlevel/ubl> -<src/gcode/bedlevel/ubl>, -<src/feature/binary_stream.cpp> -<src/libs/heatshrink>, -<src/feature/bltouch.cpp>, -<src/feature/cancel_object.cpp> -<src/gcode/feature/cancel>, -<src/feature/caselight> -<src/gcode/feature/caselight>, -<src/feature/closedloop.cpp>, -<src/feature/controllerfan.cpp> -<src/gcode/feature/controllerfan>, -<src/feature/dac> -<src/feature/digipot>, -<src/feature/direct_stepping.cpp> -<src/gcode/motion/G6.cpp>, -<src/feature/e_parser.cpp>, -<src/feature/encoder_i2c.cpp>, -<src/feature/fanmux.cpp>, -<src/feature/filwidth.cpp> -<src/gcode/feature/filwidth>, -<src/feature/fwretract.cpp> -<src/gcode/feature/fwretract>, -<src/feature/host_actions.cpp>, -<src/feature/hotend_idle.cpp>, -<src/feature/joystick.cpp>, -<src/feature/leds/blinkm.cpp>, -<src/feature/leds/leds.cpp>, -<src/feature/leds/pca9533.cpp>, -<src/feature/leds/pca9632.cpp>, -<src/feature/leds/printer_event_leds.cpp>, -<src/feature/leds/tempstat.cpp>, -<src/feature/max7219.cpp>, -<src/feature/mixing.cpp>, -<src/feature/mmu2> -<src/gcode/feature/prusa_MMU2>, -<src/feature/password> -<src/gcode/feature/password>, -<src/feature/pause.cpp>, -<src/feature/power.cpp>, -<src/feature/power_monitor.cpp> -<src/gcode/feature/power_monitor>, -<src/feature/powerloss.cpp> -<src/gcode/feature/powerloss>, -<src/feature/probe_temp_comp.cpp>, -<src/feature/runout.cpp> -<src/gcode/feature/runout>, -<src/feature/snmm.cpp>, -<src/feature/solenoid.cpp>, -<src/feature/spindle_laser.cpp> -<src/gcode/control/M3-M5.cpp>, -<src/feature/tmc_util.cpp> -<src/module/stepper/trinamic.cpp>, -<src/feature/twibus.cpp>, -<src/feature/z_stepper_align.cpp>, -<src/gcode/bedlevel/G26.cpp>, -<src/gcode/bedlevel/G35.cpp>, -<src/gcode/bedlevel/G42.cpp>, -<src/gcode/bedlevel/M420.cpp>, -<src/gcode/calibrate/G33.cpp>, -<src/gcode/calibrate/G34_M422.cpp>, -<src/gcode/calibrate/G76_M871.cpp>, -<src/gcode/calibrate/G425.cpp>, -<src/gcode/calibrate/M12.cpp>, -<src/gcode/calibrate/M48.cpp>, -<src/gcode/calibrate/M100.cpp>, -<src/gcode/calibrate/M425.cpp>, -<src/gcode/calibrate/M666.cpp>, -<src/gcode/calibrate/M852.cpp>, -<src/gcode/config/M43.cpp>, -<src/gcode/config/M217.cpp>, -<src/gcode/config/M218.cpp>, -<src/gcode/config/M221.cpp>, -<src/gcode/config/M281.cpp>, -<src/gcode/config/M302.cpp>, -<src/gcode/config/M305.cpp>, -<src/gcode/config/M540.cpp>, -<src/gcode/control/M7-M9.cpp>, -<src/gcode/control/M211.cpp>, -<src/gcode/control/M605.cpp>, -<src/gcode/feature/advance>, -<src/gcode/feature/camera>, -<src/gcode/feature/i2c>, -<src/gcode/feature/L6470>, -<src/gcode/feature/leds/M150.cpp>, -<src/gcode/feature/leds/M7219.cpp>, -<src/gcode/feature/macro>, -<src/gcode/feature/mixing/M163-M165.cpp>, -<src/gcode/feature/mixing/M166.cpp>, -<src/gcode/feature/pause/G27.cpp>, -<src/gcode/feature/pause/G60.cpp>, -<src/gcode/feature/pause/G61.cpp>, -<src/gcode/feature/pause/M125.cpp>, -<src/gcode/feature/pause/M600.cpp>, -<src/gcode/feature/pause/M603.cpp>, -<src/gcode/feature/pause/M701_M702.cpp>, -<src/gcode/feature/trinamic/M122.cpp>, -<src/gcode/feature/trinamic/M569.cpp>, -<src/gcode/feature/trinamic/M906.cpp>, -<src/gcode/feature/trinamic/M911-M914.cpp>, -<src/gcode/geometry/G17-G19.cpp>, -<src/gcode/geometry/G53-G59.cpp>, -<src/gcode/geometry/M2 [... cut ...]
esp32                     IGNORED

teensy31                  IGNORED

teensy35                  IGNORED

teensy36                  IGNORED

teensy41                  IGNORED

linux_native              IGNORED

include_tree              IGNORED

==================== 1 failed, 0 succeeded in 00:00:01.001 ====================

[10/02/20 10:23:58]
```
 
Although, there has been changes to the pins file as of 4 days ago, so it looks like this is just getting further and further away from unwinding the original issue. 🤷‍♂️

fiveangle, 2020-10-02:
#19592 opened to track new regression above introduced with #19468 

Original compile failure remains if rolled immediately prior to 90bc1993b6811623dd9d0de00251e0c63415cb49

fiveangle, 2020-10-14:
Applying PR #19773, compile is now back to the first failure mode I reported above:

sjasonsmith, 2020-10-26:
I can confirm the issue still exists. Here are updated config files for current bugfix.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5437020/Configuration.zip)

fiveangle, 2020-10-26:
Updated configs to 8bdde8d7a0b944bdf78421000c30d7eae68778d1

sjasonsmith, 2020-10-28:
@fiveangle this has now merged with my change you tested.

I’m going to go ahead and close this. Please let us know if it breaks again.

closed 2020-10-28 by sjasonsmith