MarlinFirmware/Marlin #19218: MKS Robin pro with discount full graphics

prahjister, 2020-09-01:
Any idea where I can start looking at to find issue with MKS Robin pro and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

The encoder is defined but does not work. Fyi had to fix missing pins to fix sd card issue that was missing.

sjasonsmith, 2020-09-11:
This issue ignored the issue template and did not provide the requested information.

@prahjister, I am going to close this issue. It sounds like you may have a legitimate problem to report, but it is important that you include all the requested information in the issue template. That information helps developers and users participate in your issue, and without it many will simply ignore it.

Please feel free to report a new issue with all the requested information, or to update the description in this Issue accordingly.

sjasonsmith, 2020-09-11:
If you want to discuss the issue with others, there are several community forums available. I myself frequent the Discord server, but there are experienced users in all these locations.

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-11 by sjasonsmith