MarlinFirmware/Marlin #19219: 2.0.6.1 TFT_LVGL_UI_FSMC can't display pictures

diver197, 2020-09-01:
can't display pictures, only text，Maybe the picture was not read successfully
[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5155481/Configuration.zip)

rhapsodyv, 2020-09-03:
Did you put the `assets` folder in the SD root?
The `assets` folder is generated in the build folder together with the firmware.

rhapsodyv, 2020-09-03:
Do you board have SPI Flash?

Chip W25Qxx 

LVGL requires spi flash.

thisiskeithb, 2020-09-03:
What is `BOARD_YL3V_V1`? That board doesn't exist in the main Marlin codebase.

diver197, 2020-09-03:
Yes, the files in SD card on the screen are also written into flash

diver197, 2020-09-03:
> What is `BOARD_YL3V_V1`? That board doesn't exist in the main Marlin codebase.
If you draw but directly on the screen, you can display it,
[image]

diver197, 2020-09-03:
[image]

rhapsodyv, 2020-09-03:
Which flash?

There’s the prog flash. And the spi flash. That is a separated chip.

Seems your board don’t have a spi flash.

thisiskeithb, 2020-09-03:
Which motherboard are you using? `BOARD_YL3V_V1` doesn't exist in Marlin.

diver197, 2020-09-03:
#define HAS_SPI_FLASH                          1
#define SPI_FLASH_SIZE                 0x4000000  // 16MB
#if HAS_SPI_FLASH
  #define W25QXX_CS_PIN                     PA4
  #define W25QXX_MOSI_PIN                   PA6
  #define W25QXX_MISO_PIN                   PA7
  #define W25QXX_SCK_PIN                    PA5
#endif
[image]

diver197, 2020-09-03:
[image]
[image]

rhapsodyv, 2020-09-03:
I think is because the current code fixes the spi flash to spi 2.

But your spi flash pins seems the spi 1.

We don’t have any other board similar to yours right now.

thisiskeithb, 2020-09-03:
Closing since `BOARD_YL3V_V1` is not a valid/supported board.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-03 by thisiskeithb