MarlinFirmware/Marlin #19222: print stopped aber only 30 secs

coolduke666, 2020-09-01:
### Bug Description

after upating from 2.0.6 to 2.0.6.1 print stopps within seconds after printing 10cm of the skirt
with 2.0.6 everything works fine!

only happens with larger Gcode files ... very small STL prints fine!

marlin 2.0.6 config:
[MyMarlin_2.0.6.zip](https://github.com/MarlinFirmware/Marlin/files/5157196/MyMarlin_2.0.6.zip)

### My Configurations
Marlin 2.0.6.1 config
[MyMarlin_2.0.6.1.zip](https://github.com/MarlinFirmware/Marlin/files/5157215/MyMarlin_2.0.6.1.zip)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

simply print

**Actual behavior:** [What actually happens]

stopps after staring print

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

daleckystepan, 2020-09-01:
What do you mean by "stop printing"? Any error on display or serial? Which?
Can you provide gcode file?

coolduke666, 2020-09-01:
simply stopped printing .... no error nothing....  only able to hard reset it.

coolduke666, 2020-09-01:
I think it has to do with the open issue regarding the probing margin... small prints work fine - large print don't (maybe because the interfere with the probing margin - if that makes sense...)

can provide the gcode tommorrow

coolduke666, 2020-09-02:
[GCode.zip](https://github.com/MarlinFirmware/Marlin/files/5160452/GCode.zip)

attached the 2 GCodes... one that worked and the larger one, which didnt work. again simply stopped the print after a few centimeters of the skirt.

zonalimitatore, 2020-09-13:
Me too same problem....But it's happen time.One after 1h printing,and last one tonight,stop print after 3mm of brim.No error,no sound,lcd not seems dont receive command.Only hardreset
skr 1.4T

boelle, 2020-09-13:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

zonalimitatore, 2020-09-13:
27 July

coolduke666, 2020-09-29:
Is there a fix in 2.0.7?

ellensp, 2020-09-29:
there are several hundred fixes in 2.0.7. but since your issue is not replicable by most of us.. you will have to try it and see.

boelle, 2020-09-29:
@coolduke666 or try  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) 2.0.7 is just a frozen copy of that anyway

closed 2020-11-06 by github-actions[bot]