MarlinFirmware/Marlin #19229: [FR] BLTouch unpower during print

Tampa, 2020-09-01:
A setting in Marlin mode to entirely remove the voltage input from the sensor during printing to prevent bad sensor deployment and to remove the LED illumination.

AnHardt, 2020-09-02:
To switch BL-Touch on and of it's power has to be switche via a MOSFET or Relais. Switching one of this needs to switch a not otherwise used pin. A pin can already be set with M42. Remembering a pin-number is not more difficult than remembering a M- or G-code number. There is nothing what needs to be done in Marlin to make it work. You just have to build the hardware and find a free pin.

closed 2020-09-02 by ellensp

Tampa, 2020-09-05:
The point is to prevent BLTouch from faulting and extending while the print is running potentially hitting the print even snapping BLTouch clean off.

I currently have BLTouch Servo 5V and GND on a plug to unplug it during print as it keeps going into fault condition. While this is an issue with BLTouch most likely, the point is it doesn't even need to be on during the print. It's a safety measure to prevent the print from failing due to shifting or breaking BLTouch.

If this can be done via Gcode that's fine, I guess I can attempt to find the correct pin for the 5V input and turn it off after leveling, but it still makes sense, to me at least, to have the option to disable BLTouch when it isn't being used, especially since there are already options to manually extend or stow the probe.