MarlinFirmware/Marlin #19230: [BUG] z homing only goes positive

thetechy2, 2020-09-01:
### Bug Description
z homing only moves in the positive direction, i thought i blew a dir pin but was and is working in both directions 

### My Configurations

**Required:** my ssd just died :( so i dont have them

### Steps to Reproduce

1.-- 3+ prints in a row started and stopped in software and not restarted (slicer setting calibration)
2-- insert new sd card with new gcode and preheat and start print
3--homes x and y(sensorless homing) and then when trying to home z with pinda 2 it just keeps going +Z instead of up down up its goes up up up

**Expected behavior:** [What you expect to happen]
z sensor homing
**Actual behavior:** [What actually happens]
up up up
#### Additional Information
skr 1.4 turbo, tmc 2209, marlin 2.0.x, hypercube style corexy, was working perfectly fine the other prints before-hand, i go to the other room and slice a new file and come back and this happens : (, the sensor isnt triggered accidentally, it triggers when i put metal to it
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-09-02:
Did you cut/remove the diag pin on the Z driver?

Without configs, the rest of troubleshooting will just be guessing at what might be the issue.

- - -

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-02 by thisiskeithb

thetechy2, 2020-09-02:
ok well i tested 3 files in a row, got up walked to the other room and came back and the homing didnt work anymore. It does move both directions with the move commands just the homing acts as if the sensor is permanently triggered, when it is not.

thetechy2, 2020-09-02:
no i didnt cut a pin on my driver

thisiskeithb, 2020-09-02:
> no i didnt cut a pin on my driver

That’s your issue. Cut or desolder the diag pin.

> It does move both directions with the move commands just the homing acts as if the sensor is permanently triggered, when it is not.

Endstops are only enabled while homing, which is why you can still move Z up and down.

Again, this is not a Marlin bug, so use the resources linked above for further configuration troubleshooting.

thetechy2, 2020-09-02:
so if im supposed to cut this pin then how have i been printing previously?

thetechy2, 2020-09-02:
> > no i didnt cut a pin on my driver
> 
> That’s your issue. Cut or desolder the diag pin.
 just did it and same thing, why did you just tell me to mangle my driver for no reason?

thetechy2, 2020-09-02:
> Again, this is not a Marlin bug, so use the resources linked above for further configuration troubleshooting.

so its supposed to home and print fine, then when i insert a different sd card and try to print a different file the homing fails?

nicobits46, 2020-09-15:
hi i´m having the same problem with a skr 1.4 turbo.
x and y won´t home, i cut off the diag pin because i want to use the endstops but i can{t get them to home.
the endstops work perfect doing an M119 . it´s making me crazy thank you!!!