MarlinFirmware/Marlin #19233: [BUG] 2.0.6.1, TFT, undefined reference to functions

Sergey1560, 2020-09-02:
Marlin 2.0.6.1, branch 2.0.x

Robin Nano35, with tft 480*320, FSMC

Configuration.h:

```
// TFT display with optional touch screen
// Color Marlin UI with standard menu system
#define TFT_480x320
```

Build successful, but there are warnings about undefined reference:
```
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `_lcd_babystep(AxisEnum, char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Marlin2.0/Marlin/src/lcd/menu/menu_tune.cpp:74: warning: undefined reference to `lcd_moveto(unsigned char, unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `lcd_put_u8str_P(char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Marlin2.0/Marlin/src/lcd/menu/../lcdprint.h:149: warning: undefined reference to `lcd_put_u8str_max_P(char const*, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `lcd_put_wchar(unsigned long)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Marlin2.0/Marlin/src/lcd/menu/../lcdprint.h:167: warning: undefined reference to `lcd_put_wchar_max(unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `lcd_put_u8str(char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Marlin2.0/Marlin/src/lcd/menu/../lcdprint.h:161: warning: undefined reference to `lcd_put_u8str_max(char const*, unsigned short)'
```
When you try to open the tune->babystepping menu while printing, a hard faults occurs.
[image]

Sergey1560, 2020-09-03:
> Did you test it with the last marlin bugfix-2.0.x?

I haven't tested this on hardware, but the build warnings are the same.
```
Linking .pio/build/mks_robin_nano35/firmware.elf
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `_lcd_babystep(AxisEnum, char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Official_marlin/Marlin/Marlin/src/lcd/menu/menu_tune.cpp:74: warning: undefined reference to `lcd_moveto(unsigned char, unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `lcd_put_u8str_P(char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Official_marlin/Marlin/Marlin/src/lcd/menu/../lcdprint.h:149: warning: undefined reference to `lcd_put_u8str_max_P(char const*, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `lcd_put_wchar(unsigned long)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Official_marlin/Marlin/Marlin/src/lcd/menu/../lcdprint.h:167: warning: undefined reference to `lcd_put_wchar_max(unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_tune.cpp.o: In function `lcd_put_u8str(char const*)':
/home/sergey/Projects/3D_Printer/FirmWare/Marlin/Official_marlin/Marlin/Marlin/src/lcd/menu/../lcdprint.h:161: warning: undefined reference to `lcd_put_u8str_max(char const*, unsigned short)'
```

sjasonsmith, 2020-09-11:
I am presuming this has been fixed and am closing this issue. Please reply if you are still having issues with the latest bugfix branch.

closed 2020-09-11 by sjasonsmith

anteyru, 2020-09-23:
Поставил скомпилированный последний bin-файл all 2208. После окончания печати и отработки end g-code голова пытается уехать на Y 361. При нагреве hotend в конце цикла температура может упасть на 10 градусов, а во время печати скачет плюс/минус 2 градуса.
[image]

thisiskeithb, 2020-09-23:
> Поставил скомпилированный последний bin-файл all 2208. После окончания печати и отработки end g-code голова пытается уехать на Y 361. При нагреве hotend в конце цикла температура может упасть на 10 градусов, а во время печати скачет плюс/минус 2 градуса. Нет настройки LA.

@anteyru: This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y