MarlinFirmware/Marlin #19248: [BUG] Not Compile Marlin 2.0.6.1 for BTT V1.4 Turbo

Kris83is, 2020-09-03:
Hello, someone can help me to solve this problem below?
I am trying to compile Marlin for BTT SKR V1.4 Turbo on my pc with windows 7 pro.

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.1.3 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.4
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <LiquidCrystal> 1.0.0
Building in release mode

Unable to find destination disk (Command 'fsutil fsinfo drives' returned non-zero exit status 1.)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy
the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_I2C_routines.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\LCD_pin_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\usb_serial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\watchdog.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwinder.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\LPC1769\src\src\MarlinCore.cpp.o
Compiling .pio\build\LPC1769\src\src\core\serial.cpp.o
Compiling .pio\build\LPC1769\src\src\core\utility.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\ubl\ubl.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o
[... 296 lines cut ...]
mks_robin_lite3           IGNORED
jgaurora_a5s_a1           IGNORED
STM32F103CB_malyan        IGNORED
STM32F070RB_malyan        IGNORED
malyan_M300               IGNORED
chitu_f103                IGNORED
chitu_v5_gpio_init        IGNORED
STM32F103RET6_creality    IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
LERDGEX                   IGNORED
LERDGES                   IGNORED
LERDGEK                   IGNORED
rumba32                   IGNORED
esp32                     IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
linux_native              IGNORED
include_tree              IGNORED
============================================================================================= 1 failed, 0 succeeded in 00:03:37.388 =============================================================================================

ellensp, 2020-09-03:
Please attach your configuration files. As was requested in the template.

ellensp, 2020-09-03:
My guess is you have enabled SDIO_SUPPORT, but SDIO_SUPPORT is not supported on LPC1768/9 its only for STM32 based controllers

closed 2020-09-03 by ellensp

Kris83is, 2020-09-03:
> Please attach your configuration files. As was requested in the template.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/5171047/Configuration.h.zip)

Thanks for the help.

YHRen, 2020-09-23:
I got a similar problem. I compiled to the wrong board, and didn't "clean all". After "clean all", the re-build works. 
This might help others who have similar problem.