MarlinFirmware/Marlin #19249: [FR] Define Sensorless for individual Axis

hackebike, 2020-09-03:
Currently sensorless can be enabled only for both X and Y axis at the same time.
It would be good to be able to define which axis to enable/disable.
This allows for example to have sensorless enabled for a driver that a sensitivity value is found to be working, and to disable sensorless for another driver that cannot find a good value for sensitivity.

For example in the configuration file we could have:
**#define sensorless_x true
#define sensorless_y true**

teemuatlut, 2020-09-04:
Comment out the sensitivity setting for the axis you don't want.
```cpp
    #define X_STALL_SENSITIVITY  8
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    //#define Y_STALL_SENSITIVITY  8
    #define Y2_STALL_SENSITIVITY Y_STALL_SENSITIVITY
``

hackebike, 2020-09-04:
oh ok, did not know that was the intent ! great, thx

thisiskeithb, 2020-09-04:
This is more of a documentation issue since the feature already exists.

closed 2020-09-04 by thisiskeithb