MarlinFirmware/Marlin #19252: ASSISTED_TRAMMING and MAGNETIC_SWITCHING_TOOLHEAD : Compilation error in G35.cpp

yopla31, 2020-09-03:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5170441/Marlin.zip)
Printer : Ender 5 Pro modified !
Board  : MKS SGEN-L
I use 2 hotends with magnetic attachment (#define MAGNETIC_SWITCHING_TOOLHEAD)
See at https://www.youtube.com/watch?v=mZzq3hM-eWg
If I set #define ASSISTED_TRAMMING, I get a compilation error : 

**> _Marlin\src\gcode\bedlevel\G35.cpp: In static member function 'static void GcodeSuite::G35()':
> Marlin\src\gcode\bedlevel\G35.cpp:106:5: error: 'tool_change' was not declared in this scope
>   106 |     tool_change(0, true);
>       |     ^~~~~~~~~~~
> *** [.pio\build\LPC1768\src\src\gcode\bedlevel\G35.cpp.o] Error 1_**
 
Same error with bugfix-2.0.x #19250 (Bump distribution date (2020-09-03) 768182c )
If I disable ASSISTED_TRAMMING, the compilation is OK
I use a BL-Touch.

ellensp, 2020-09-05:
add to G35.cpp around line 32..  after includes but before define DEBUG_OUT
```
#if HAS_MULTI_HOTEND
  #include "../../module/tool_change.h"
#endif
```
I cannot test this... but it compiles.

yopla31, 2020-09-05:
Yes, it compiles, but a big print is printing.
I shall try tomorrow this feature.

Thx !!

thisiskeithb, 2020-09-15:
Did @ellensp's fix work for you @yopla31?

yopla31, 2020-09-15:
Yes, it's OK.🙂

JBL

thisiskeithb, 2020-09-15:
> Yes, it's OK.

Great!

@ellensp: I'm hoping your suggestion will also fix the issue in https://github.com/MarlinFirmware/Marlin/issues/19402, but I'm waiting for more feedback (hopefully some error output over serial will give me more info), but feel free to put in a PR.

thisiskeithb, 2020-09-16:
I've put in a PR that will allow `G35` to work on multi hotend setups: https://github.com/MarlinFirmware/Marlin/pull/19418 (thanks @ellensp for the fix!)

closed 2020-09-18 by thisiskeithb