MarlinFirmware/Marlin #19253: [FR] (Additional Realtime commands for Marlin)

fedetony, 2020-09-03:
### Description

### Feature Workflow

#### Additional Information

Grbl behaves in this way with the commands:
 '?' -> sends position and state information immediately to the host
 '!' -> Holds feed immediately
 '~' ->Resumes Feed
'^X' -> Soft reset

Is very well described in Grbl documentation:
A realtime command:
Will execute within tens of milliseconds.
Is a single character that may be sent at any time.
Does not require a line feed or carriage return after them.
Is not considered a part of the streaming protocol.
Are intercepted when they are received and never placed in a buffer to be parsed.
Will ignore multiple commands until it has executed the first received command.
May be tied to an input pin and may be operated with a button or switch.
Actions depends on state or what Machine_state is doing. It may not do anything.

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-09-03:
There's an open PR with this feature: https://github.com/MarlinFirmware/Marlin/pull/17462

Also from the comments:

> Instant stop from 60mm/s to zero is going to lose steps, and instant start at that speed will also lose steps. Both can be potentially damaging to hardware. TMC driver models that have StandStill detection will go into shutdown and must be reset. It is not likely in most cases that a job could be resumed after stopping with this method.

GMagician, 2020-09-03:
@thisiskeithb I think that, as for all industrial machines, hold is meant to stop with ramp, not abruptly

thisiskeithb, 2020-09-03:
There is a comment in that PR about a ramp down, but it might get traction again if you leave a comment 🙂

fedetony, 2020-09-03:
I have been thinking how to implement it.My idea is to store the command issued, buffer and current position in memory. Then a stop command can be issued(M410) or quickstop_stepper(); which works just fine, just it clears the buffer. When resumed the buffer will be restated and the command/process restated. If stopped then buffer clears.
Added this function to motion.cpp to report while moving. I call it from emergency parser and works fine:
```
/**
 * Output the current position (processed) to serial while moving
 */
void report_current_position_moving() {

  get_cartesian_from_steppers();
  const xyz_pos_t lpos = cartes.asLogical();
  SERIAL_ECHOPAIR("X:", lpos.x, " Y:", lpos.y, " Z:", lpos.z, " E:", current_position.e);

  stepper.report_positions();
  #if IS_SCARA
    scara_report_positions();
  #endif
  report_current_grblstate_moving(); 
}
/**
 * Output the current grbl compatible state to serial while moving
 */
void report_current_grblstate_moving() {
  #if ENABLED(FULL_REPORT_TO_HOST_FEATURE)
    SERIAL_ECHOPAIR("S_XYZ:", marlin_state);
    SERIAL_EOL();
  #endif  
}

```

fedetony, 2020-09-03:
Ok took me less time that I imagined, take in count please I've only seen Marlin for 1 week, yet is so well structured that is easy to find the programming.
[Marlin_ModifiedPauseresume.zip](https://github.com/MarlinFirmware/Marlin/files/5171751/Marlin_ModifiedPauseresume.zip)

I modified all the files enclosed in the Zip file  and It pause and resumes without much issue. I did not change the Parser.cpp so after returning from the Command it shows "unknown command" messages for all stuff. The emergency parser commands are: 
P to quick pause, R to quick resume and S to print status while moving or processing. Modified the keep_alive routine to constantly give the position information and not only a busy message. Is activated by FULL_REPORT_TO_HOST_FEATURE I've added to the configuation.h.

Hope it helps :)

AnHardt, 2020-09-03:
Pleas try:
```
M117 SHOWSTOPPER
```
somewhere in a g-code script. Any reaction of your new additions would be a showstopper.

Please use git or ad least provide diffs. Reviewing your code from the .zip is a pain.

fedetony, 2020-09-04:
> Pleas try:
> 
> ```
> M117 SHOWSTOPPER
> ```
> 
> somewhere in a g-code script. Any reaction of your new additions would be a showstopper.
> 
> Please use git or at least provide diffs. Reviewing your code from the .zip is a pain.

Sorry, I'm new to this. Please let me know how  (or give me a link where I can find how to) add to git? You can compare the files with a compare tool as tortoise. 

The M117 issue is that I used S, R and P for codes, since I was just trying out that the functions would work.  
Same as  
`M117 LCR Print M112 out :)`

It needs to be modified to an actual code as M000 or something it will not interfere with normal text and commands. Attached adapted the e_parser.h for P000 for quick pause R000 for quick Resume and S000 to show  position moving.

[e_parser.zip](https://github.com/MarlinFirmware/Marlin/files/5171976/e_parser.zip)

fedetony, 2020-09-05:
@AnHardt So Downloaded the repository again, made a `git init` then replaced my folder with all changes, staged and commit them adding the correspondent comments on my local directory. 
Here git diff. using `git diff 9913333c33d234264a9072630e8b5ffbf07bad7e f4933bff4c1b30b1d69c35277ed1591e261f32db` last commit hash on repository on github.
 I have not `git push` since I do not know if it will rewrite the Marlin repository and I do not want to damage the files (I'm very new to this, first time using git). I am not even a software developer... so please be a bit patient with me :)

```diff

 #
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 8cd2faf7f..beb62ed7c 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -71,7 +71,7 @@
 // @section homing

@@ -1493,9 +1494,10 @@
 // every couple of seconds when it can't accept commands.
 //
 #define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages
-#define DEFAULT_KEEPALIVE_INTERVAL 2  // Number of seconds between "busy" messages. Set with M113.
+#define DEFAULT_KEEPALIVE_INTERVAL 1  // Number of seconds between "busy" messages. Set with M113.
+#define KEEPALIVE_INTERVAL_DIVIDER 3 // Divide the KEEPALIVE_INTERVAL for faster reporting
 #define BUSY_WHILE_HEATING            // Some hosts require "busy" messages even during heating
-
+#define FULL_REPORT_TO_HOST_FEATURE        // Enable this to send Machine status reports while moving and status reports GRBL style
 
 // Host Receive Buffer Size
 // Without XON/XOFF flow control (see SERIAL_XON_XOFF below) 32 bytes should be enough.
@@ -1908,7 +1908,7 @@
  * Currently handles M108, M112, M410, M876
  * NOTE: Not yet implemented for all platforms.
  */
-//#define EMERGENCY_PARSER
+#define EMERGENCY_PARSER

 // Some clients will have this feature soon. This could make the NO_TIMEOUTS unnecessary.
-//#define ADVANCED_OK
+#define ADVANCED_OK

 // Printrun may have trouble receiving long strings all at once.
 // This option inserts short delays between lines of serial output.
 #define SERIAL_OVERRUN_PROTECTION

 // @section extras
 /**
  * G-code Macros
@@ -3224,9 +3224,9 @@
  * Host Prompt Support enables Marlin to use the host for user prompts so
  * filament runout and other processes can be managed from the host side.
  */
-//#define HOST_ACTION_COMMANDS
+#define HOST_ACTION_COMMANDS
 #if ENABLED(HOST_ACTION_COMMANDS)
-  //#define HOST_PROMPT_SUPPORT
+  #define HOST_PROMPT_SUPPORT
 #endif

 /**
@@ -3376,9 +3376,9 @@
  * string to enable synchronization with DLP projector exposure. This change will allow to use
  * [[WaitForDoneMessage]] instead of populating your gcode with M400 commands
  */
-//#define NANODLP_Z_SYNC
+#define NANODLP_Z_SYNC
 #if ENABLED(NANODLP_Z_SYNC)
-  //#define NANODLP_ALL_AXIS  // Enables "Z_move_comp" output on any axis move.
+  #define NANODLP_ALL_AXIS  // Enables "Z_move_comp" output on any axis move.
                               // Default behavior is limited to Z axis only.
 #endif

diff --git a/Marlin/src/MarlinCore.cpp b/Marlin/src/MarlinCore.cpp
[... 563 lines cut ...]
+  report_current_grblstate_moving();
+  stepper.wake_up();
+
+}
+

 void Planner::endstop_triggered(const AxisEnum axis) {
   // Record stepper position and discard the current block
diff --git a/Marlin/src/module/planner.h b/Marlin/src/module/planner.h
index f3a3a0e0f..3ef5e2d91 100644
--- a/Marlin/src/module/planner.h
+++ b/Marlin/src/module/planner.h
@@ -828,6 +828,10 @@ class Planner {
     // Called to force a quick stop of the machine (for example, when
     // a Full Shutdown is required, or when endstops are hit)
     static void quick_stop();
+    // Called to force a quick pause of the machine (for example, when
+    // a pause is required on the middle of movement)
+    static void quick_pause();
+    static void quick_resume();

     // Called when an endstop is triggered. Causes the machine to stop inmediately
     static void endstop_triggered(const AxisEnum axis);

```

fedetony, 2020-09-05:
> Pleas try:
> 
> ```
> M117 SHOWSTOPPER
> ```
> 
> somewhere in a g-code script. Any reaction of your new additions would be a showstopper.
> 
> Please use git or at least provide diffs. Reviewing your code from the .zip is a pain.

```
M117 SHOWSTOPPER

//action:notification SHOWSTOPPER

ok P15 B3

```

fedetony, 2020-09-05:
#18427 is quite similar to what is achieved here.

mron, 2020-09-08:
I will contribute time to make this happen. I want to see it done reliably. No lost steps. No lost commands. Perfect command sync between CNC controller and Marlin.

IvanBayan, 2020-09-08:
@fedetony First of all, thank you for your attempt to make real time pause/resume.
I would like to suggest you to make a fork of that repo, and create your own branch from 2.0.x branch.
You will be able to push code changes into your repository and use github's pull request functionality. I hope that guide will help you: https://gist.github.com/MarcDiethelm/7303312
BTW, M114 already has realtime report ability, check for `M114_REALTIME`

mron, 2020-09-09:
I followed the guide and cloned Marlin. I'm struggling with VSC and plattformio. I'm not sure how to keep two projects separate. I have the Marlin that works on my CNC and the one I want to use to work on the Pause/Resume. Create a branch?

@fedetony I'd like to work on this as well. I think I understand your approach. Won't a quick_stop() loose steps?

fedetony, 2020-09-09:
@IvanBayan  Thank you for your comment. I've created the fork, pushed my changes, and made my first pull request. I'm not sure if I made a mistake since : All checks have failed. lol 🍡 
@mron In my machine it does not loose any, yet it does not move too fast and has lead screws. This is to be tested in other machines. But is possible to recover: 
```
P000 You pause abruptly
S000 Show actual position  (for checking)
M410 then you Stop the machine, 
R000 then resume (wakes the steppers), then 
M60 save position, 
G28 home sequence
M61 go back to position
```

mron, 2020-09-09:
I'm pretty sure if you turn off the steppers, you've lost the true count. With lead screws, you're right, probably won't notice it, unless you do it a lot. I'm going to look at how GRBL implemented M0.
@IvanBayan I set M114_REALTIME in my Configuration_adv.h. I get immediate response, but the XYZ values don't change. They show the destination values. I'm looking for current. What @fedetony  implemented.

mron, 2020-09-09:
Not meaningless, sorry. Not accurate.

mron, 2020-09-09:
Looks like most of GRBL "feed hold" stuff is in stepper.c

IvanBayan, 2020-09-12:
@mron If hare any chance, that you compiled M114 real time support, but issued that gcode without R flag? It is not in documentation, but to get realtime report you need to issue 'M114 R' gcode

IvanBayan, 2020-09-12:
@fedetony my bad, I suggested you to use wrong branch for your patch.

My cent about real time stop/resume command. It definitely should use ramp or s-curve for pause/resume. Immediate start/stop without ramp will lead to steps losses and vibration, it's not so noticeable on machines with lead screws but it still present.
Moreover under the load or on high speeds step motors may not start motion at all. Without a doubt that functionality need much more time investment than using quickstop_stepper().

mron, 2020-09-12:
I’m working on a smooth start/resume. I think I need to modify the planner and stepper code. I want to be very careful with that. I’m open to suggestions on how to proceed.

mron, 2020-09-12:
as as M114 R. I tried this. I did not get current position. I got the position for the end of the current move. Maybe I needed other flags, of I’m using the wrong branch?

mron, 2020-09-12:
I just tried it again. Works! I must have disabled it in the Config file at some point. It’s what I need to work on feed hold.

AnHardt, 2020-09-12:
A while ago i wrote down a [concept for a deaccelerated fast stop](https://github.com/AnHardt/Marlin/issues/87). Maybe it can help.

mron, 2020-09-13:
Thanks for responding. I read your concept. It’s along the lines of my thinking. One of the things that worried me were end conditions, trying to stop too close to the end of a line. There is a number of steps needed to stop and restart from the current velocity. If this is less than the remaining number of steps in a move, then let the move finish with a stop. Otherwise inject a stop.

I don’t understand the code well enough yet. I’m not a fast programmer and tend to be methodical. I’m currently testing the planner and stepper code and watching how it works. I hope to be injecting a stop and start today.

AnHardt, 2020-09-13:
> There is a number of steps needed to stop and restart from the current velocity. If this is less than the remaining number of steps in a move, then let the move finish with a stop.

Right. 
When a bufferline (aka segment) is running and its already behind the deacceleration point it will not stop in this segment when the following junction is not at zero (jerk) speed. But then there is always a next segment. For the last segment it's granted to end at zero speed.

But if there is a buffer full of short segments, what all together run only a few ms, that's not the case we worry about. We worry about the situation the buffer is full of long moves lasting together several seconds.

So if the break flag is set, simply ignore the deacceleration-point and always deaccelerate as much as possible (the info about how much is already in the buffer). If the segment ends just begin to run the next - immediately with the deaccelaration phase active. Until zero is reached. Then store the states.

mron, 2020-09-13:
If I understand correctly, the deceleration point may have been calculated with another moves velocity in mind. Continuing to zero from that point might overshoot the correct endpoint. Recalculating a new deceleration point might cause more problems. Maybe if we’re decelerating, don’t insert. Let the next move start then drive it to zero.

AnHardt, 2020-09-13:
Sorry. I'm currently not in the mood for holding the "beginners planner basics" course.

mron, 2020-09-13:
No problem.

closed 2020-11-07 by fedetony

intuity-hans, 2020-11-18:
@fedetony, are there any news on this subject?

fedetony, 2020-11-18:
@intuity-hans  I personally do not see an issue on my machine. When moving on the fastest speed and I use the pause it does not loose steps at all. I tried to modify the stepper function adding deceleration but it didn't work, and since I do not have a way of debugging it, I just left it there. The deceleration routines are implemented inside the stepper .C file I couldn't trigger them yet I only tried once.
The fork is functional and updated, I can recheck it again if you specify exactly what you would like to have.. 👍

intuity-hans, 2020-11-19:
Thanks for the quick reply! Just so I understand what you did – it is now possible to pause the program while it is executing a movement command. But, instead of an emergency stop (which clears the planner buffer) it is able to restart , right?

fedetony, 2020-11-19:
it is now possible to pause the program while it is executing a movement command?
Yes, the command P000 will pause Marlin while moving (probing,homing,or any positional movement). Will not do anything if not moving. For that you need M0 and M108 to resume

But, instead of an emergency stop (which clears the planner buffer) it is able to restart , right?
Yes, When in P000 state, you can:
Resume S000 and will continue where it was.
stop (M410) (which clears the planner buffer) 
or kill M112 (you need to reset by HW after)

intuity-hans, 2020-11-30:
okay thanks for clarifying. I will take a closer look at this. Seems fitting for my needs =)