MarlinFirmware/Marlin #19255: [BUG] Can't finish bed leveling  with BLTouch

danis101, 2020-09-04:
### Bug Description

When Im doing bed leveling it can't finish. It stops after last point of second line of measure points

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5171945/Marlin.zip)

### Steps to Reproduce

1. Set bltouch on cr10-s5 with skr1.3
2. try to level bed with bltouch

**Expected behavior:** [What you expect to happen]
I want it to be finished
**Actual behavior:** [What actually happens]
It can't be finished  since it fininsh one line, than go for second line and in the end of second line it stops
#### Additional Information
Here is ending of console report

> 
> 00:33:48.141 : current_position= X106.25 Y34.00 Z1.40 : sync_plan_position
> 00:33:48.142 : current_position= X106.25 Y34.00 Z1.40 : <<< Probe::probe_down_to_z
> 00:33:48.142 : current_position= X106.25 Y34.00 Z1.40 : <<< run_z_probe
> 00:33:48.142 : >>> do_blocking_move_to X106.25 Y34.00 Z5.40
> 00:33:48.142 : <<< do_blocking_move_to
> 00:33:49.177 : <<< Probe::probe_at_point
**> 00:33:49.178 : >>> Probe::probe_at_point(151.25, -51.25, raise, 0, probe_relative)
> 00:33:49.178 : current_position= X106.25 Y34.00 Z5.40 :
> 00:33:49.178 : Position Not Reachable**
> 00:33:49.178 : current_position= X106.25 Y34.00 Z5.40 : Probe::set_deployed
> 00:33:49.178 : deploy: 0
> 00:33:49.178 : Probe::move_z(4.00)
> 00:33:49.178 : >>> do_blocking_move_to X106.25 Y34.00 Z5.40
> 00:33:49.178 : <<< do_blocking_move_to
> 00:33:49.178 : current_position= X106.25 Y34.00 Z5.40 : > probing complete
> 00:33:49.178 : >>> do_blocking_move_to X106.25 Y34.00 Z4.00
> 00:33:49.178 : <<< do_blocking_move_to
> 00:33:49.677 : X:106.25 Y:34.00 Z:4.00 E:0.00 Count X:17000 Y:5440 Z:3200
> 00:33:49.678 : <<< G29
> 

By following those lines I noticed that marlin want my printer to move bed to y=-51.25 position with is lolgicaly not reachable
But what is going on, I did not anything that could point marlin to do that.
Same thing goes when I disable Y_FIRST - Enabled that to make sure its not a cable connection problem, just to make sure that during leveling cables do not move as much as with X axis moving all the time, but it have same effect just in different points (without Y_FIRST it wants X axis to move out of area)

Any suggestions?

ellensp, 2020-09-04:
From your Config #define CONFIGURATION_H_VERSION 020005, this is not the most recent marlin.

Please use a current marlin.

closed 2020-09-04 by ellensp

danis101, 2020-09-04:
seems like New marlin works with BLtouch, finally I measured whole bed - For future who will have same issue - Update your marlin