MarlinFirmware/Marlin #19256: Compilation Error with STM32 Boards when not selecting a Display

XBrav, 2020-09-04:
Version Tested: 2.0.6.1
Platforms: STM32F103RC_bigtree, STM32F103RE_bigtree.

Issue: When compiling without a defined display, the Dependency Graph indicates U8glib is required, instead of U8glib. Several errors prevent the compilation as per the below build:

https://pastebin.com/PhMmWp4R

For the test, I ensured PlatformIO was free and clear of any cached libraries. .pio directory was deleted, along with all PlatformIO master libraries.

Once compiled with a display enabled (compiling for a BigTreeTech E3 DIP, uncommented CR10_DISPLAY), the errors no longer appeared, and the compile completed successfully.

thisiskeithb, 2020-09-04:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

thisiskeithb, 2020-09-04:
I just tested several of my SKR Mini E3 configs with both `STM32F103RC_bigtree` & `STM32F103RE_bigtree` environments and all compile fine with the latest `bugfix-2.0.x` and no LCD controller enabled.

ellensp, 2020-09-04:
Closing till more information is provided, like example configs that trigger this issue.

closed 2020-09-04 by ellensp

XBrav, 2020-09-04:
Sorry, I was unavailable for a few hours. Please see the attached configs.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5172704/Marlin.zip)

Ticket details being updated.

thisiskeithb, 2020-09-04:
There's likely something broken in your VSCode/PlatformIO setup because your config builds fine in all the supported environments:

[image]