MarlinFirmware/Marlin #19260: [BUG] Part cooling fan only working when pressing on screen

bpivk, 2020-09-04:
### Bug Description

Part cooling fan only works is you click on the screen.

### My Configurations

https://drive.google.com/file/d/1KKNgMa3MVhhmtfPdePK_tJhBfsKEgI1j/view?usp=sharing

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** Fan spins

**Actual behavior:** Bupkis

#### Additional Information
I have a SKR e3 2.0 board plugged into an Ender 5 Plus with TFT32 screen. Latest (ish) 2.0.6 version of Marlin.
I have noticed that my part cooling settings only start if I click on the rotary button. If I don't the settings do change on the screen but nothing happens and the fan doesn't spin.
As soon as I press on a button (any screen) the fan wakes up and starts working. It used to work in the previous versions and this is the first time I have noticed it but it seems that I'm not alone based on this post: https://www.reddit.com/r/BIGTREETECH/comments/idyfuq/skr_mini_e3_v20_fan_problem/

bpivk, 2020-09-04:
Reporting back...

The latest 2.0.6.1 with the same config does the same. But I did notice that this happens only when printing from SD card. Printing the same gcode from Octoprint works fine.

thisiskeithb, 2020-09-04:
I can't say I've had this issue with my SKR Mini E3 V2 with the `CR10_STOCKDISPLAY` or a BigTreeTech TFT35.

Have you tried using a different LCD controller/TFT or use the TFT outside of its case (to make sure nothing is pressing against it)? Tried different cables?

> It used to work in the previous versions

Please track down which commit works for you and report back. I doubt this is software related, but the extra datapoint will help.

bpivk, 2020-09-04:
Yes I have tried different TFTs. I have three more E3 displays. All have their cables.
I can't really track down which commit worked as I have only noticed this issue. and the problem comes and goes. I have just reflashed with speaker off  (used to be on to have the speaker functional) and I'll see if it's a timer issue (maybe?).

I haven't noticed this issue on 2.0.5 though but I could be wrong as I don't really look at the part cooling fan so often. The post I have linked to suggests that it's not an isolated issue (as in not just my device) and that it spans across multiple commits although I can't say that I have noticed that.

J0hnMatrix, 2020-09-05:
Already happened to me but because I had forgot to activate:
- FAN_SOFT_PWM (line 2295)
- SOFT_PWM_SCALE 0 (line 2302)
- SOFT_PWM_DITHER (line 2308)

boelle, 2020-09-05:
@bpivk tried what @J0hnMatrix suggested?

bpivk, 2020-09-05:
I have compiled the firmware but I have yet to install it. I'm printing something right now and the fan is working.
The issue is really hard to repeat because it seems that sometime it works and sometime it doesn't. Or maybe the fact that I have disabled the speaker helped as I haven't noticed the issue since then. But I did only make two prints and I haven't looked at one of them so I don't really know if the fan worked.

closed 2020-10-13 by github-actions[bot]