MarlinFirmware/Marlin #19261: [FR] (feature request title)

malicmt, 2020-09-04:
### Description
Hello,

Recently I had an issue with my custom made printer that the Solid State Relay which is used to control the heated bed got faulty and it acted like a closed switch which made the heater to keep heating. Even though all the thermal protection was ON in the firmware, and the display was showing the printer is halted (and it stopped the printing of course), the heater was still ON.
Luckily, my wife noticed the burning smell and melting plastic on the bed and she switched off the power immediately.
In order to avoid such a situation, I would like to have another relay which will turn off whenever the printer is halted due to any reason.
I believe it would be good if I can configure a servo pin or any other available pin for this purpose.

### Feature Workflow

A hardware pin configurable in the firmware which will go high when the printer is ON (or when starting the print job) and should go low whenever the printer gets halted due to an error/user interrupt.

AnHardt, 2020-09-04:
For that request you are at least 5 yeas to late.
Lock up `PS_ON_PIN` and `PSU_CONTROL` in 'Configuration.h'.
This pin is switching the power-supply to off when the machine is killed.

closed 2020-09-04 by thisiskeithb

malicmt, 2020-09-05:
My bad....
My thought didn't go that way
Thanks buddy - for the direction