MarlinFirmware/Marlin #19264: [BUG] Marlin (CLI) command UnicodeEncodeError

Nikongen, 2020-09-04:
### Bug Description

Wrong (unicode) character in https://marlinfw.org/docs/basics/install_platformio_cli.html leads to UnicodeEncodeError when copy-paste commands with -target argument

e.g.: platformio run –target upload -e target_env 
should be platformio run -target upload -e target_env
-> "-" instead of "–" (U+2013) in front of target
### Steps to Reproduce

1. Octopi 0.17.0 or Debian Buster (Kernel: Linux 4.19.0-9-amd64) with platformio core installed with pip
2. git clone Marlin files
3. follow 'Installing Marlin (CLI)' web instruction
4. Copy&Paste Build command -> works fine
5. Copy&Paste Upload command -> UnicodeEncodeError

**Expected behavior:**
successful upload

**Actual behavior:** 
UnicodeEncodeError

#### Additional Information
`platformio run –target upload -e mega2560`

leads to

```
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1257, in invoke
    sub_ctx = cmd.make_context(cmd_name, args, parent=ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in make_context
    self.parse_args(ctx, args)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1053, in parse_args
    "s" if len(args) != 1 else "", " ".join(map(make_str, args))
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2013' in position 0: ordinal not in range(128)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  pip install -U platformio command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
```

thisiskeithb, 2020-09-04:
I can't reproduce this on macOS copying text out of Chrome. I even tried to change the character in the [Marlin Documentations repo / install_platformio_cli.md](https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_basics/install_platformio_cli.md) and git doesn't see any difference, so there's nothing to commit.

Have you tried a different browser?

Does copying the text directly from the `install_platformio_cli.md` linked above work for you?

Nikongen, 2020-09-04:
I did not copy the command from the git documentation but from the marlinfw.org website linked above. 
In the git documentation it's 
...--target...
On marlinfw.org it's 
... –target... (Unicode U+2013)

Checked this with chrome on win10 and safari on iOS

thisiskeithb, 2020-09-05:
Oh, I see what's happening now.

The source of Marlinfw.org's content is the MarlinDocumentation repo, so there must be some kind of character conversion happening when the content is pushed.

@thinkyhead: Do you know why double en dashes are being converted to an em dash?

Nikongen, 2020-09-06:
> ~~I can't reproduce this on macOS copying text out of Chrome. I even tried to change the character in the [MarlinDocumentation repo / install_platformio_cli.md](https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_basics/install_platformio_cli.md) and git doesn't see any difference, so there's nothing to commit.~~  I was testing the wrong line.
> 
> 
> 
> Have you tried a different browser?
> 
> 
> 
> Does copying the text directly from the `install_platformio_cli.md` linked above work for you?

Yes,Copying from *.md document works for me.

thisiskeithb, 2020-09-06:
There's a [fix posted to the MarlinDocumentation repo](https://github.com/MarlinFirmware/MarlinDocumentation/pull/333) for that specific page, but I'm guessing there's still a bigger issue related to conversion when the site is published that is hopefully easy to spot by thinkyhead.

closed 2020-09-10 by thinkyhead