MarlinFirmware/Marlin #19265: [FR] (Add instructions on how to compile for the X5SA example)

tuskiomi, 2020-09-05:
### Description

It's as simple as it says, on the X5SA example (https://github.com/MarlinFirmware/Configurations/blob/release-2.0.6.1/config/examples/Tronxy/X5SA/HOWTO-INSTALL.md) add instructions for compilation. The compilation, as per the [install instructions](https://marlinfw.org/docs/basics/install.html), being an ARM32 based board, should be compiled using Platform IO, but Platform IO only returns a 'Error - Invalid Board' when compiling.

### Feature Workflow

No workflow, just a few sentences are needed to clarify.

#### Additional Information

I'm pretty sure this isn't a bug, and there just needs to be some clarification, the chitu board is not available to the consumer, so its build process may be different from others.

thisiskeithb, 2020-09-05:
The Tronxy X5SA config builds fine for both supported environments with VSCode+PlatformIO.

<p align="center">[image]</p>

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-05 by thisiskeithb

tuskiomi, 2020-09-05:
@thisiskeithb  evidently not. the V5 board is not supported at all in the master branch.

tuskiomi, 2020-09-05:
@thisiskeithb  Also, if you didn't see the tag (which evidently, you did not) this is a feature request. meaning it is a request to add a feature. in this case the feature is information.

rhapsodyv, 2020-09-05:
> @thisiskeithb evidently not. the V5 board is not supported at all in the master branch.

I compile Marlin almost every single day with the V5 board: `BOARD_CHITU3D_V5`. It always work. I wrote that tutorial too.

Seem your marlin version is outdated.

thisiskeithb, 2020-09-05:
> the V5 board is not supported at all in the master branch

The board is already supported in both `2.0.x` and `bugfix-2.0.x` branches. Make sure you are using the most recent version of either of those branches.

This is not a Marlin bug.

Use the resources linked in my first reply for further config troubleshooting.