MarlinFirmware/Marlin #19270: [BUG] With MAGNETIC_SWITCHING_TOOLHEAD, compilation error

yopla31, 2020-09-05:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5178487/Marlin.zip)

Printer : Ender 5 Pro modified !
Board : MKS SGEN-L
I use 2 hotends with magnetic attachment (#define MAGNETIC_SWITCHING_TOOLHEAD)
See at https://www.youtube.com/watch?v=mZzq3hM-eWg
If I set #define MAGNETIC_SWITCHING_TOOLHEAD in Configuration.h , I get a compilation error :

_In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1085:6: error: #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
 1085 |     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1085:6: error: #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
 1085 |     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1085:6: error: #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
 1085 |     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
MCompiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
arlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1085:6: error: #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
 1085 |     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1085:6: error: #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
 1085 |     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:41,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1085:6: error: #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
 1085 |     #error "(ELECTRO)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
      |      ^~~~~
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\include\i2c_util.c.o
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1_

**My SOLUTION :** 
I added the line : 
_#define SWITCHING_TOOLHEAD_Z_HOP          10_
just after : 
 _#elif ENABLED(MAGNETIC_SWITCHING_TOOLHEAD)
    #define SWITCHING_TOOLHEAD_Y_RELEASE      -10         // (mm) Security distance Y axis
    #define SWITCHING_TOOLHEAD_X_SECURITY   { 58, 183 } // (mm) Security distance X axis (T0,T1)
    **#define SWITCHING_TOOLHEAD_Z_HOP          10         // (mm) Z raise for switching**_

and the compilation is OK, and the changetool is ok.

boelle, 2020-09-05:
SWITCHING_TOOLHEAD_Z_HOP is allready there in configuration.h at line 291 so closing this one as not a bug

boelle, 2020-09-05:
[image]

closed 2020-09-05 by boelle

yopla31, 2020-09-05:
<br>
Hi Bo,<br>
<br>
Sorry, but if define MAGNETIC_SWITCHING_TOOLHEAD , the line is never executed !<br>
In my attachment, in the file Configuration.h, I added the line #288 to solve this issue.<br>
<br>
Regards.<br>
<br>
JBL
<div class="gl_quote" style="margin-top: 20px; padding-top: 5px;">De : "Bo Herrmannsen"<br>
A : "MarlinFirmware/Marlin" ,"yopla31" ,"Author"<br>
Envoyé: samedi 5 septembre 2020 20:59<br>
Objet : Re: [MarlinFirmware/Marlin] [BUG] With MAGNETIC_SWITCHING_TOOLHEAD, compilation error (#19270)<br>
 
<div class="gl_quoted">
<p> </p>

<p>SWITCHING_TOOLHEAD_Z_HOP is allready there in configuration.h at line 291 so closing this one as not a bug</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
</div>

<div class="gl_quoted"> </div>
</div>

yopla31, 2020-09-07:
Any comment ??
I define MAGNETIC_SWITCHING_TOOLHEAD and no **ELECTRO**MAGNETIC_SWITCHING_TOOLHEAD.
So the error message is explicite : 
#error "(**ELECTRO**)MAGNETIC_SWITCHING_TOOLHEAD requires SWITCHING_TOOLHEAD_Z_HOP."
but for MAGNETIC_SWITCHING_TOOLHEAD, SWITCHING_TOOLHEAD_Z_HOP is never define !